openwrt/openwrt #17411: MMC card reader does not function

BorreSanders, 2024-12-28:
### Describe the bug

[    7.509414] mtk-msdc 10130000.mmc: no support for card's volts
[    7.521080] mmc0: error -22 whilst initialising SDIO card
[    7.532469] init: - preinit -
[    8.960184] mmc0: Card stuck being busy! __mmc_poll_for_busy
[    9.010318] mtk-msdc 10130000.mmc: no support for card's volts
[    9.021991] mmc0: error -22 whilst initialising SDIO card
[   10.199853] mmc0: Card stuck being busy! __mmc_poll_for_busy
[   10.244659] mtk-msdc 10130000.mmc: no support for card's volts

### OpenWrt version

24.10.0-rc4

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

ramips-mt76x8

### Device

RAVPower RP-WD009

### Image kind

Official downloaded image

### Steps to reproduce

1. Install 24.10.0-rc4
2. Observe MMC errors in kernel and system logs.

### Actual behaviour

Builtin MMC card reader does not function

### Expected behaviour

Builtin MMC card reader does function

### Additional info

I was hoping that the 24.x release/kernel would fix the issue with certain SD cards in the builtin MMC module that I reported with the 23.x release. However with rc4 of the new release, the MMC module does not function at all.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

BorreSanders, 2024-12-29:
I fixed version and target reports after initial bugreport submission.

hauke, 2024-12-31:
Please try a snapshot build. #15896 is part of the snapshot build, but it was not backported to 24.10 because someone reported that it did not fix his problem with the SD card. Maybe you have a different problem.

BorreSanders, 2025-01-01:
Thanks all for your follow up, and promising fixes! I'm no software development expert, but I am willing to test a snapshot build. Can you please suggest which build I should try?

DragonBluep, 2025-01-01:
Try this [ravpower_rp-wd009.tar.zip](https://github.com/user-attachments/files/18285273/ravpower_rp-wd009.tar.zip)

BorreSanders, 2025-01-01:
Just tried, and the result looks very promising (see fragment from syslog):
Wed Jan  1 13:08:07 2025 user.info kernel: [    4.557397] init: - preinit -
Wed Jan  1 13:08:07 2025 kern.info kernel: [    5.164071] mmc0: new high speed SDXC card at address 59b4
Wed Jan  1 13:08:07 2025 kern.info kernel: [    5.213595] mmcblk0: mmc0:59b4 LX1TB 928 GiB
Wed Jan  1 13:08:07 2025 kern.info kernel: [    5.251786]  mmcblk0: p1
Seems that the MMC card reader now correctly picks up on the 1TB micro-SD card.
Hope this helps to make a decision on how to proceed. Let me know if you need further input from me, or a test.

DragonBluep, 2025-01-01:
If you want, maybe you can do some file copying && checksum tests to see if it is reliable. That firmware has already integrated the basic ext4 and exfat filesystem support.

BorreSanders, 2025-01-01:
Successfully copied a >9GB file from the SD card with exfat file system to another location on the same card, and compared them. According to cmp on the source and destination files the copy operation went fine.

closed 2025-01-04 by openwrt-bot

BorreSanders, 2025-01-14:
Problem is not solved in rc5, see https://github.com/openwrt/openwrt/pull/17446#issuecomment-2579407635