openwrt/openwrt #17416: apk: sysupgrade -k --create-backup still referencing opkg

wberrier, 2024-12-29:
### Describe the bug

It looks like `sysupgrade --create-backup -k` hasn't been updated to populate `/etc/backup/installed_packages.txt` from the apk database, and is still referencing the opkg database.

### OpenWrt version

r28410-a85059438f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

`sysupgrade --create-backup -k`

### Actual behaviour

```
find: /usr/lib/opkg/info: No such file or directory
etc/uci-defaults/10_disable_services
etc/backup/installed_packages.txt
```

### Expected behaviour

It's expected that all the config files, including `/etc/backup/installed_packages.txt` get included in the tarball.

### Additional info

My image was created from the imagebuilder, using the following command:

```
make image PROFILE=bananapi_bpi-r4 'PACKAGES=wpad-mesh-mbedtls ddns-scripts qos-scripts jq curl ip ipset iptables-mod-conntrack-extra iptables-mod-ipopt luci-app-ddns luci-app-qos luci-proto-wireguard wireguard-tools luci luci-ssl iwinfo kmod-mt7921u iptraf-ng iftop ncat nmap tcpdump ethtool fping procps-ng-watch luci-app-wol diffutils kmod-usb-serial-ftdi picocom tmux speedtest-netperf iperf iperf3 usbip usbip-client usbip-server usbutils lm-sensors kmod-mtd-rw stress -wpad-basic-wolfssl -wpad-basic-mbedtls -odhcpd-ipv6only' EXTRA_IMAGE_NAME=wberrier-router-4 'DISABLED_SERVICES=odhcpd dnsmasq firewall' FILES=/home/wberrier/wa/git/wberrier-home/computer-configs/router/router-4
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-12-29:
Its already been added in:
https://github.com/openwrt/openwrt/commit/cee18af7049c32713a5ada5196d8ad238ee837fb

wberrier, 2024-12-29:
Ah, great, sorry for the noise

closed 2024-12-29 by wberrier