openwrt/openwrt #17417: Wifi stopped working on latest git

grosjo, 2024-12-29:
### Describe the bug

Upgrade a device to latest git. Clients can not connect anymore to the wifi

### OpenWrt version

r28433-15b21c474e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

YunCore/KuWfi CPE830(D)

### Image kind

Self-built image

### Steps to reproduce

Just upgrade to latest git

### Actual behaviour

No client connecting to Wifi setup (Master, no security, no mac filter)

### Expected behaviour

Clients connecting to Wifi setup (Master, no security, no mac filter)

### Additional info

Wifi has no security (all open)

### Diffconfig

```text
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_yuncore_cpe830=y
# CONFIG_ATH9K_HWRNG is not set
CONFIG_ATH9K_SUPPORT_PCOEM=y
# CONFIG_ATH9K_UBNTHSR is not set
# CONFIG_ATH_USER_REGD is not set
CONFIG_CLEAN_IPKG=y
# CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set
# CONFIG_DROPBEAR_AGENTFORWARD is not set
# CONFIG_DROPBEAR_CHACHA20POLY1305 is not set
# CONFIG_DROPBEAR_CURVE25519 is not set
# CONFIG_DROPBEAR_DBCLIENT_AGENTFORWARD is not set
# CONFIG_DROPBEAR_ED25519 is not set
# CONFIG_DROPBEAR_U2F is not set
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
CONFIG_KERNEL_DEVKMEM=y
CONFIG_KERNEL_DEVMEM=y
# CONFIG_KERNEL_DEVPTS_MULTIPLE_INSTANCES is not set
CONFIG_KERNEL_DEVTMPFS=y
[... 191 lines cut ...]
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-lua=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_urandom-seed is not set
# CONFIG_PACKAGE_urngd is not set
# CONFIG_PACKAGE_usign is not set
CONFIG_PACKAGE_wpad-basic=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zoneinfo-asia=y
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PROCD_SHOW_BOOT=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_MKLIBS=y
# CONFIG_WPA_MBO_SUPPORT is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-29:
Please correct diffconfig output, looks like you pasted output back to terminal.

grosjo, 2024-12-31:
> Please correct diffconfig output, looks like you pasted output back to terminal.

@brada4 done

hauke, 2024-12-31:
Is the AP sending beacons?

With which version was it working the last time?

Do you get any error messages? Please post the last 200 lines from logread after the AP was created and a client wants to connect.

grosjo, 2025-01-01:
24.10-RCx works very well - latest git NOT

No error message, just client devices dont see the wifi

nbd168, 2025-01-01:
Please try reverting commit fdc2c6ab9901ad3e64ec1f93d5ccdc38990243b4, it looks broken to me

grosjo, 2025-01-01:
It seems @robimarko  reverted all his commits
[image]

robimarko, 2025-01-01:
Yes, they are clearly broken currently so I reverted them

closed 2025-01-03 by grosjo