openwrt/openwrt #17421: GL-inet Brume 2 (MT2500) slow networking

roschelle63, 2024-12-29:
### Describe the bug

OpenWRT installs fine, but network speed tops out around 35 mbps. Before and after this install, I used the manufacturer's forked release and it was able to tap the full speed of my connection at 310 mbps. Happy to test this device in more detail for you, if that would be helpful. I had tried an OpenWrt 23 snapshot with the same slow result as the 24 RC candidate.

### OpenWrt version

OpenWrt 24.10.0-rc4

### OpenWrt release

OpenWrt 24.10.0-rc4

### OpenWrt target/subtarget

mediatek/mt7981

### Device

GL.iNet GL-MT2500

### Image kind

Official downloaded image

### Steps to reproduce

install; run Speedtest

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-29:
You are not using OpenWRT apparently.
Info missing can be added from putput of
```
ubus call system board
```

roschelle63, 2024-12-29:
I have already switched back to the gl-inet version, which is why you are saying its "invalid." It would be a painstaking processs to re-install rc4 just to grab the debugging information -- but I can assure you that it was slow on the most recent release downloaded from the firmware locator. Is there a way that we can easily make clear that what I was having troubles with is this: https://firmware-selector.openwrt.org/?version=24.10.0-rc4&target=mediatek%2Ffilogic&id=glinet_gl-mt2500

And I'll try to update the above based on what it says on that page.

brada4, 2024-12-29:
Reassesment is mñnual, in the meantime check porting thread in forum?

roschelle63, 2024-12-29:
I looked carefully in the forum and didn't find anything about this device. Jeremy

nxhack, 2024-12-30:
I am using gl-mt2500a as an edge router. I am also using SQM. (HW & SW Flow offloading is stopped) (openwrt main snapshot)

I have the following settings.
```bash
echo 2 > /proc/irq/75/smp_affinity
echo 1 > /proc/irq/76/smp_affinity
taskset -p -c 0 `pgrep mtk_eth-5`
taskset -p -c 1 `pgrep mtk_eth-6`
for q in /sys/class/net/eth0/queues/rx-*/rps_cpus; do echo 3 > $q; done
for q in /sys/class/net/eth1/queues/rx-*/rps_cpus; do echo 3 > $q; done
```

With these settings, I can handle 300Mbps, but anything more than that will use up all the CPU.
I plan to replace it with nanppi r6c.

roschelle63, 2024-12-30:
@brada4 please see if the details now look correct.

roschelle63, 2026-02-17:
> dangowrt
>  left a comment
> (openwrt/openwrt#17421)
> Please check EEE (Energy Efficient Ethernet) is active, and if so disable and try if it helps.
>
> for e in eth0 eth1; do
>   ethtool --show-eee $e
>   ethtool --set-eee $e eee off tx-lpi off
> done
> If that makes any difference it would be important to know
>
> which version of the MT2500 are you using (MaxLinear PHY or Airoha PHY)?
> if you install and run iperf3 on the device itself and test eg. with proof.ovh.net, do you also see the same reduced speed?
> if the above is true, please swap LAN and WAN ports in /etc/config/network and see if performance of iperf3 running on the device itself is also impacted when using the other Ethernet port as WAN.

I had some more testing time available. Iperf3 is very slow at all times for me (in Washington DC); it’s not a useful test. As before, it runs fast for 6-8 hours and then slows down.

It is maxlinear.

Both when the router was fast and slow, it’s not in energy-efficient mode:

EEE Settings for eth0:
        EEE status: disabled
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  Not reported
tx-lpi unmodified, ignoring
eee unmodified, ignoring
EEE Settings for eth1:
        EEE status: disabled
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  100baseT/Full
                                                 1000baseT/Full
tx-lpi unmodified, ignoring
eee unmodified, ignoring

Here is the sensors command, taken when it is running slow. It doesn’t actually seem that it’s hot.
cpu_thermal-virtual-0
Adapter: Virtual device
temp1:        +37.6°C

mdio_bus:05-mdio-5
Adapter: MDIO adapter
temp1:        +39.9°C

After a reboot (from the slow state), it was still slow. Sensors readout was approx the same after a reboot.

Below is the dmesg and the part that is after the slow down is the last few lines. Doesn’t seem to say much to me, but maybe you see something notable.

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.12.66 ***@***.***) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 14.3.0 r32747-7c26668217) 14.3.0, GNU ld (GNU Binutils) 2.44) #0 SMP Sat Jan 24 00:44:39 2026
[    0.000000] Machine model: GL.iNet GL-MT2500
[    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ***@***.***
[    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable ***@***.***
[    0.000000] OF: reserved mem: 0x0000000047c80000..0x0000000047d7ffff (1024 KiB) nomap non-reusable ***@***.***
[    0.000000] OF: reserved mem: 0x0000000047d80000..0x0000000047dbffff (256 KiB) nomap non-reusable ***@***.***
[    0.000000] OF: reserved mem: 0x0000000047dc0000..0x0000000047ffffff (2304 KiB) nomap non-reusable ***@***.***
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x0000000047c7ffff]
[    0.000000]   node   0: [mem 0x0000000047c80000-0x0000000047ffffff]
[    0.000000]   node   0: [mem 0x0000000048000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 20 pages/cpu s42584 r8192 d31144 u81920
[    0.000000] pcpu-alloc: s42584 r8192 d31144 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[... 174 lines cut ...]
[    8.776101] PPP generic driver version 2.4.2
[    8.782407] NET: Registered PF_PPPOX protocol family
[    8.800100] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[    8.807966] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld ***@***.***>. All Rights Reserved.
[    8.821214] kmodloader: done loading kernel modules from /etc/modules.d/*
[    8.868984] urngd: v1.0.2 started.
[   12.353168] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[   12.366700] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/gmii link mode
[   12.375857] br-lan: port 1(eth1) entered blocking state
[   12.381158] br-lan: port 1(eth1) entered disabled state
[   12.386420] mtk_soc_eth 15100000.ethernet eth1: entered allmulticast mode
[   12.393491] mtk_soc_eth 15100000.ethernet eth1: entered promiscuous mode
[   12.410443] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:05] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
[   12.420686] mtk_soc_eth 15100000.ethernet eth0: configuring for phy/2500base-x link mode
[   19.699184] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx
[   31.128950] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[   31.128984] br-lan: port 1(eth1) entered blocking state
[   31.142614] br-lan: port 1(eth1) entered forwarding state

It was fast here, then the part below is when it was slow

[  309.748450] Ethtool ioctl interface doesn't support passing EEE linkmodes beyond bit 32
[  309.758775] Ethtool ioctl interface doesn't support passing EEE linkmodes beyond bit 32
[  309.798441] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[  313.958023] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx