openwrt/openwrt #17441: Write message to /dev/console failed on 24.10

jjm2473, 2024-12-31:
### Describe the bug

```
root@OpenWrt:~# echo hello > /dev/console 
ash: write error: I/O error
root@OpenWrt:~# 
```

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

x86/64

### Device

innotek GmbH VirtualBox

### Image kind

Official downloaded image

### Additional info

When upgrading the firmware of the previous version (22.03) of openwrt, the terminal would print information, but now it is gone.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-31:
What is in `cat /proc/cmdline` and `ls -l /dev/console`

jjm2473, 2024-12-31:
@brada4 

```
root@OpenWrt:~# cat /proc/cmdline ; ls -l /dev/console 
BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=c08eae04-04d7-3b08-205e-fa50fb5f3d02 rootwait console=tty1 console=ttyS0,115200n8 noinitrd
crw-------    1 root     root        5,   1 Dec 31 04:58 /dev/console
```

I don't have change any thing in grub, it's an official 24.10.0-rc4 firmware, installed on Virtualbox
```
root@OpenWrt:~# cat /boot/grub/grub.cfg 
serial --unit=0 --speed=115200 --word=8 --parity=no --stop=1 --rtscts=off
terminal_input console serial; terminal_output console serial

set default="0"
set timeout="5"
search -l kernel -s root

menuentry "OpenWrt" {
        linux /boot/vmlinuz root=PARTUUID=c08eae04-04d7-3b08-205e-fa50fb5f3d02 rootwait  console=tty1 console=ttyS0,115200n8 noinitrd
}
menuentry "OpenWrt (failsafe)" {
        linux /boot/vmlinuz failsafe=true root=PARTUUID=c08eae04-04d7-3b08-205e-fa50fb5f3d02 rootwait  console=tty1 console=ttyS0,115200n8 noinitrd
}

```

jjm2473, 2024-12-31:
I think this issue is related to this PR https://github.com/openwrt/openwrt/pull/16213

jjm2473, 2024-12-31:
Sorry, there is no output when upgrading the firmware on 22.03.7

namiltd, 2024-12-31:
There will be no error if you configure the serial port in the VirtualBox machine settings.

jjm2473, 2025-01-01:
> There will be no error if you configure the serial port in the VirtualBox machine settings.

I think the message will write to serial port only.