openwrt/openwrt #17443: br-lan device gets a random MAC address after upgrade from 23.05.5 to 24.10.0rc4

grosjo, 2024-12-31:
### Describe the bug

When upgrading from 23.05.5 to 24.10.0rc4 and using the same configuration, the MAC address assignment to interfaces appears to have changed in such a way that (among other things) it is not possible to assign a DHCP-received static address for the 'br-lan' interface, because said interface's MAC is generated randomly each boot. The 'wan' device appears to have been assigned the same MAC as 'br-lan'.

### OpenWrt version

r28218-45bf30e062

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

YunCore/KuWfi CPE830(D)

### Image kind

Self-built image

### Steps to reproduce

Upgrade device from 23.05 to 24.10-rc4

MAC address changed and is changing times to times at each reboot

### Actual behaviour

My devices are connecting via WDS based on respective MC addresses

Upon upgrading the firmware, MACs have changed and devices were not connecting anymore

### Expected behaviour

No impact on prior (23.x) MAC address
No random changes of MAC address

### Additional info

Appears in WDS (not sure about non-WDS connection)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2024-12-31:
YunCore/KuWfi CPE830(D) is not supported in OpenWrt 24.10. Support was only added to main in df222e57be467eb85e77f4f569e169d471c8ba54. Are you using a custom patched version?

grosjo, 2024-12-31:
Same for all my devices (about 20). Will update with another nodel, but issue is the same for all devices

It was also reported here https://github.com/openwrt/openwrt/issues/17384 but they insisted that it is not a general problem, but actually, it is

hauke, 2024-12-31:
Do you have the problem with Comfast CF-WR650AC v2?

Which interface gets the wrong mac address?

This is not a general problem, but it effects some devices. Many devices were converted to get the mac address in kernel space and no using scripts in user space. Probably something broke at this point. It probably broke in one of these commits: e816591e226a540000171a5a676e7033d3f0f128, d07cec6b2bdcf6389c06a2599a5c924ee490136f or 74f2df9dbccd9840edb25d55ad70d58e3ed8303c.

grosjo, 2025-01-01:
This is affecting the bridged interface (br-lan as default name)
All ports on all my devices are bridged (internal network)
Thsi is happening to all my devices, they are all ATH79

grosjo, 2025-01-08:
@hauke how to fix this ? thank you

grosjo, 2025-01-14:
Any update on that ?

Neustradamus, 2025-01-15:
@neheb: Have you seen this problem?

grosjo, 2025-01-23:
How to resolve this ?

AddaxSoft, 2025-06-18:
temporary fix to this issue is to assign a macaddr manually:
I assume you can replace `eth0` with `br-lan` too

```
MACADDR=$(cat /sys/class/net/eth0/address)
uci set network.eth0dev='device'
uci set network.eth0dev.name='eth0'
uci set network.eth0dev.macaddr="$MACADDR"
uci commit network && sync
```