openwrt/openwrt #17448: Huawei E3306 ncm modem - netifd - variable does not exist

pielgrzym, 2025-01-01:
### Describe the bug

I had a working setup of NCM version of E3306 on 23.05, however when I upgraded to 24 rc4 when the modem is connected via usb setting the interface fails:

```
Wed Jan  1 19:50:23 2025 daemon.notice netifd: Interface 'wwan_ncm' is setting up now
' does not exist or is not an array/objectifd: wwan_ncm (23025): WARNING: Variable 'huawei
Wed Jan  1 19:50:25 2025 daemon.notice netifd: wwan_ncm (23025): Unsupported modem
Wed Jan  1 19:50:25 2025 daemon.notice netifd: wwan_ncm (23441): Stopping network wwan_ncm
' does not exist or is not an array/objectifd: wwan_ncm (23441): WARNING: Variable 'huawei
Wed Jan  1 19:50:27 2025 daemon.notice netifd: wwan_ncm (23441): Unsupported modem
Wed Jan  1 19:50:27 2025 daemon.notice netifd: Interface 'wwan_ncm' is now down
```

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

ipq40xx/generic

### Device

GL.iNet GL-A1300

### Image kind

Self-built image

### Steps to reproduce

I configured my NCM modem interface like this:

```
config interface 'wwan_ncm'
        option ifname 'wwan0'
        option proto 'ncm'
        option pdptype 'IP'
        option device '/dev/ttyUSB0'
        option mode 'auto'
        option apn 'internet'
        option delay '15'
        option ttl '65'
        option metric '1022'
```

In 23.05 connecting usb modem successfully brought up the wwan_ncm interface.

### Actual behaviour

Netifd seems to crash when trying to bring wwan_ncm interface - full log:

```
Wed Jan  1 19:50:21 2025 kern.info kernel: [15304.804029] usb 1-1: new high-speed USB device number 25 using xhci-hcd
Wed Jan  1 19:50:21 2025 kern.info kernel: [15305.328921] usb 1-1: USB disconnect, device number 25
Wed Jan  1 19:50:22 2025 kern.info kernel: [15305.884100] usb 1-1: new high-speed USB device number 26 using xhci-hcd
Wed Jan  1 19:50:22 2025 kern.info kernel: [15306.159379] option 1-1:1.0: GSM modem (1-port) converter detected
Wed Jan  1 19:50:22 2025 kern.info kernel: [15306.159746] usb 1-1: GSM modem (1-port) converter now attached to ttyUSB0
Wed Jan  1 19:50:22 2025 kern.info kernel: [15306.165410] option 1-1:1.1: GSM modem (1-port) converter detected
Wed Jan  1 19:50:22 2025 kern.info kernel: [15306.172380] usb 1-1: GSM modem (1-port) converter now attached to ttyUSB1
Wed Jan  1 19:50:22 2025 kern.info kernel: [15306.349663] huawei_cdc_ncm 1-1:1.2: MAC-Address: <REMOVED>
Wed Jan  1 19:50:22 2025 kern.info kernel: [15306.349726] huawei_cdc_ncm 1-1:1.2: setting rx_max = 16384
Wed Jan  1 19:50:22 2025 kern.info kernel: [15306.356021] huawei_cdc_ncm 1-1:1.2: NDP will be placed at end of frame for this device.
Wed Jan  1 19:50:22 2025 kern.info kernel: [15306.360840] huawei_cdc_ncm 1-1:1.2: cdc-wdm0: USB WDM device
Wed Jan  1 19:50:22 2025 kern.info kernel: [15306.370145] huawei_cdc_ncm 1-1:1.2 wwan0: register 'huawei_cdc_ncm' at usb-xhci-hcd.0.auto-1, Huawei CDC NCM device, <REMOVED>
[... 144 lines cut ...]
CONFIG_TARGET_DEVICE_ipq40xx_generic_DEVICE_zte_mf18a=y
CONFIG_TARGET_DEVICE_PACKAGES_ipq40xx_generic_DEVICE_zte_mf18a=""
CONFIG_TARGET_DEVICE_ipq40xx_generic_DEVICE_zte_mf282plus=y
CONFIG_TARGET_DEVICE_PACKAGES_ipq40xx_generic_DEVICE_zte_mf282plus=""
CONFIG_TARGET_DEVICE_ipq40xx_generic_DEVICE_zte_mf286d=y
CONFIG_TARGET_DEVICE_PACKAGES_ipq40xx_generic_DEVICE_zte_mf286d=""
CONFIG_TARGET_DEVICE_ipq40xx_generic_DEVICE_zte_mf287=y
CONFIG_TARGET_DEVICE_PACKAGES_ipq40xx_generic_DEVICE_zte_mf287=""
CONFIG_TARGET_DEVICE_ipq40xx_generic_DEVICE_zte_mf287plus=y
CONFIG_TARGET_DEVICE_PACKAGES_ipq40xx_generic_DEVICE_zte_mf287plus=""
CONFIG_TARGET_DEVICE_ipq40xx_generic_DEVICE_zte_mf287pro=y
CONFIG_TARGET_DEVICE_PACKAGES_ipq40xx_generic_DEVICE_zte_mf287pro=""
CONFIG_TARGET_DEVICE_ipq40xx_generic_DEVICE_zte_mf289f=y
CONFIG_TARGET_DEVICE_PACKAGES_ipq40xx_generic_DEVICE_zte_mf289f=""
CONFIG_TARGET_DEVICE_ipq40xx_generic_DEVICE_zyxel_nbg6617=y
CONFIG_TARGET_DEVICE_PACKAGES_ipq40xx_generic_DEVICE_zyxel_nbg6617=""
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pielgrzym, 2025-01-01:
The issue seems to stemp in line 115-116 of `/lib/netifd/proto/ncm.sh` - perhaps the json parsing is failing somehow.

pielgrzym, 2025-01-01:
Ok I think I  tracked down the issue: in line 93 of `/lib/netifd/proto/ncm.sh` we are obtaining manufacturer of the modem using gcom and `/dev/ttyUSB0` (in my case). I suspected there might be some unprintable characters in the output and I was right:

```
gcom -d "/dev/ttyUSB0" -s /etc/gcom/getcardinfo.gcom  | awk 'NF && $0 !~ /AT\+CGMI/ { sub(/\+CGMI: /,""); print tolower($1); exit; }' > /tmp/test
xxd /tmp/test
00000000: 6875 6177 6569 0d0a                      huawei..
```
Line ends with `\r\n`. I changed line 93 to explicitly remove trailing newline using `tr -d '\r\n'` and modem started working again. Line 93 after modification:
```
manufacturer=$(gcom -d "$device" -s /etc/gcom/getcardinfo.gcom | awk 'NF && $0 !~ /AT\+CGMI/ { sub(/\+CGMI: /,""); print tolower($1); exit; }' | tr -d '\r\n')
```

Hope that helps!

Jalakas, 2025-01-03:
Wouldn't it be good if you formalized your finding into a pull request?

morfikov, 2025-01-19:
I had the same issue with my Huawei E3372s-153, and I can confirm that the fix with changing the line 93 works pretty well.

jlabuz, 2025-04-11:
I have the same problem, but with my Huawei E3372**h**, not E3372**s** modem, tested on two different devices. 

Unlike the cases described above by @pielgrzym https://github.com/openwrt/openwrt/issues/17448#issuecomment-2600984427 and @morfikov https://github.com/openwrt/openwrt/issues/17448#issuecomment-2600984427, I get an additional empty line ending with `^M` at the very beginning of the text returned from `gcom`:

```
$ cat -vET gcom-output.txt
^M$
huawei^M$
^M$
OK^M$
```
That makes `awk` treat the first line as non-empty and detect the manufacturer as `^M` so then interface initialization fails.

This is similar to gcom behaviour described in #17998 by @pcislocked, @Zredy https://github.com/openwrt/openwrt/issues/17998#issuecomment-2690692229 and @shalzz https://github.com/openwrt/openwrt/issues/17998#issuecomment-2691982541, as @naseef pointed out.

I solved this by modifying the `RS` variable in the `awk` call in line 93 to handle both Linux- and Windows-style line breaks the same way:
```
manufacturer=$(gcom -d "$device" -s /etc/gcom/getcardinfo.gcom | awk -v RS='\r?\n' 'NF && $0 !~ /AT\+CGMI/ { sub(/\+CGMI: /,""); print tolower($1); exit; }')
```

I think this approach solves both this issue and #17998, because adding `tr -d '\r\n'` at the end will work only when there is no that "empty" first line like in my case.

commit 2eb7e4bf (2025-04-17): comgt-ncm: fix modem manufacturer detection

Fix an issue where NCM interface initialization fails because of wrong
modem manufacturer detection.

gcom call returns an output with Windows-style line breaks (containing \r)
what makes awk call return empty or malformed manufacturer name. Changing
awk RS variable to handle both \n and \r\n as line break fixes this issue.

Fixes #17448 and #17998 GitHub issues.

Link: https://github.com/openwrt/openwrt/pull/18460
PR #18460 (2025-04-17): comgt-ncm: fix modem manufacturer detection
Fix an issue where NCM interface initialization fails because of wrong modem manufacturer detection.

gcom call returns an output with Windows-style line breaks (containing \r) what makes awk call return empty or malformed manufacturer name. Changing awk RS variable to handle both \n and \r\n as line break fixes this issue.

Fixes #17448 and #17998 GitHub issues.
closed 2025-04-17 by openwrt-bot