openwrt/openwrt #17452: NanoPi R5C WAN port (eth1) initialization fails on boot

ataluts, 2025-01-02:
### Describe the bug

WAN interface ('_eth1_' device) initialization fails on boot so WAN port is unavailable.
In Luci _Network -> Interfaces_ tab there is an error in '_wan_' and '_wan6_' interfaces stating "Network device is not present".
'_eth1_' is not even mentioned in the log because the device is not found ("_Phy link never came up_" in the log). 

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R5C

### Image kind

Official downloaded image

### Steps to reproduce

Flash official OpenWrt image from firmware selector and boot.

### Actual behaviour

eth1 interface is not initialized.

### Expected behaviour

eth1 interface is initialized.

### Additional info

This is definitely not a hardware issue because:
1. on FriendlyWrt this port works without any issues on every boot;
2. quite rarely it even manages to work on OpenWrt.

Several times I had managed to boot OpenWrt with 'eth1' working by powering the device off, connecting cable only to WAN port and powering it on again. In this case the port works without any issues but after reboot it fails again.
Seems that it is some kind of timing issue during PCIe initialization.

Also while searching for this problem I have found [this post](https://forum.openwrt.org/t/nanopi-r5c-rockchip-rk3568b2-2-pcie-2-5gbps/148431/21) with very similar behaviour.

In the following files there are '_dmesg_' outputs for successful and unsuccessful boot, also for FriendlyWrt boot.
[dmesg - eth1 fail.txt](https://github.com/user-attachments/files/18287163/dmesg.-.eth1.fail.txt)
[dmesg - eth1 ok.txt](https://github.com/user-attachments/files/18287164/dmesg.-.eth1.ok.txt)
[dmesg - friendlywrt.txt](https://github.com/user-attachments/files/18287166/dmesg.-.friendlywrt.txt)

Side note:
I'm flashing OpenWrt to an uSD-card and boot from it because I have fully functional FriendlyWrt on internal eMMC. When I flash FriendlyWrt on the same uSD-card it works without issues.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

1715173329, 2025-01-05:
uninstalling `kmod-r8169` and installing `kmod-r8125-rss` should help this issue.

ataluts, 2025-01-05:
> uninstalling `kmod-r8169` and installing `kmod-r8125-rss` should help this issue.

Nope, doesn't solve the problem. Everything stays the same.
I think it's not the issue with the Ethernet PHY itself but rather something going wrong with PCIe. But I'm not an expert in this at all so it's only my layman guess.

ataluts, 2025-01-09:
Investigating what can be the cause of this first pulse, I probed power lines _VDD3V3_25GLAN_ which comes from _VCC_3V3_ which is produced by PMIC (RK809-5) _SW1_ channel from _VCC3V3_SYS_. And _VCC_3V3_ goes down right at that moment which means that this pulse is not caused by CPU switching PERSTB signal state but by PMIC turning off that power bus.
On FriendlyWrt there is no such behaviour of that bus - voltage stays constant throughout the entire boot process.

CH1 (yellow) - PERSTB of WAN port, CH2 (green) - VCC_3V3.
[image]

Averell7, 2025-01-10:
Interesting is the remark about cold/warm : our experience, with a total of 7 R5C with the last Dietpi is : we had few problems with wlan0 eand lan0. The problem is mainly with the §WiFi, which is the WiFi installed by the Manufacturer. Generally, with some exceptions, the WiFi fails to start on cold boots, but is recognized on reboot. The reason may be the temperature ? 
In 2025, the R5C is integrated in the Linux kernel and maybe the problem is solved. But we have not been able to test yet.

ataluts, 2025-01-11:
If I blow hot air right underneath the SoC (on the bottom side of the board) while the rest of it is still cold, the WAN port starts almost right away on a boot. Acts like ordinary soldering issue except that on every FriendlyElec build it always work no matter what (I even put the board in a fridge and it still behaved exactly the same) and it works in OpenWrt just fine even if I cool down the board after a successful boot, but as soon as I press reset it fails to start again.
I even pulled down (with a 100Ohm resistor) the VDD3V3_25GLAN power line and PERSTB all the way to the ground during that short power dropout - didn't help. The problem is that I have only one board and can't check if something similar happens on another one or is it just this particular copy that is broken.
For now I have no idea what's going wrong. The only thing to experiment with, as i see now, is to eliminate that short power dropout and watch what will happen.

hauke, 2025-01-12:
After Linux is started it will deactivate all unused clocks. When you add `clk_ignore_unused` to the boot arguments it will keep the clocks the bootloader already activated running. If this removes your short drop you should find which clock is actually needed. 

Maybe the reset of 125ms is too short. Which pin is it? Did you check the in the device tree if you can configure a longer time.

ataluts, 2025-01-13:
1. Found out that this power dropout is not during boot, it's during system shutdown when the board is reset. OpenWrt resets the PMIC while FriendlyWrt is not. I gues it can be ruled out since when external power is supplied for the first time there is no such pulse but the board behaves the same.
2. Soldered a jumper that PERSTB signal from the LAN port also resets the WAN port - didn't make any difference. So I guess duration of this pulse is not the issue. (In the PHY datasheet minimum time for reset pulse is declared to be 100us so "short" pulse is already orders of magnitute longer than required)

PERSTB for WAN is pin AA20 (GPIO0_B6), PERSTB for LAN is pin AG27 (GPIO0_A0) according to schematics.
As for the clocks and device tree It is extremely problematic for me to check anything like that because I am not a programmer but a hadware design engineer and I don't know how to build or compile anything myself.

Averell7, 2025-01-13:
Since the device works fine with Friendly Elec OS, based on Debian, I think this means the problem is more on the OS side. I did not test OpenWrt, I just know that on our six R5C, we do not have problems with the wan or lan ports, it is only the wifi which does not start at cold boot, it starts only on reboot, but not always. 
We have done what we good, with the help of the Dietpi founder, but we have not been able to fix this problem.

1715173329, 2025-01-14:
> Since the device works fine with Friendly Elec OS, based on Debian, I think this means the problem is more on the OS side.

well FriendlyElec OS and FriendlyWrt use BSP kernel where tricks may apply, while OpenWrt uses mainline kernel.

dsakurai, 2025-01-19:
> > Describe the bug
>
> WAN interface ('eth1' device) initialization fails on boot so WAN port is unavailable.
In Luci Network -> Interfaces tab there is an error in 'wan' and 'wan6' interfaces stating "Network device is not present".
'eth1' is not even mentioned in the log because the device is not found ("Phy link never came up" in the log).

I have the same problem on the same device (R5C) but with 24.10.0-rc5. I haven't tried warming it up.
Here's the log from `dmesg`.
~~~
root@OpenWrt:~# dmesg | grep -i ethernet
[    2.089617] rk_gmac-dwmac fe2a0000.ethernet: IRQ eth_lpi not found
[    2.090683] rk_gmac-dwmac fe2a0000.ethernet: supply phy not found, using dummy regulator
[    2.091592] rk_gmac-dwmac fe2a0000.ethernet: clock input or output? (output).
[    2.092234] rk_gmac-dwmac fe2a0000.ethernet: TX delay(0x3c).
[    2.092740] rk_gmac-dwmac fe2a0000.ethernet: RX delay(0x2f).
[    2.093258] rk_gmac-dwmac fe2a0000.ethernet: integrated PHY? (no).
[    2.098859] rk_gmac-dwmac fe2a0000.ethernet: init for RGMII
[    2.099638] rk_gmac-dwmac fe2a0000.ethernet: User ID: 0x30, Synopsys ID: 0x51
[    2.100283] rk_gmac-dwmac fe2a0000.ethernet: 	DWMAC4/5
[    2.100918] rk_gmac-dwmac fe2a0000.ethernet: DMA HW capability register supported
[    2.101586] rk_gmac-dwmac fe2a0000.ethernet: RX Checksum Offload Engine supported
[    2.102246] rk_gmac-dwmac fe2a0000.ethernet: TX Checksum insertion supported
[    2.102868] rk_gmac-dwmac fe2a0000.ethernet: Wake-Up On Lan supported
[    2.103438] rk_gmac-dwmac fe2a0000.ethernet: TSO supported
[    2.103926] rk_gmac-dwmac fe2a0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    2.104634] rk_gmac-dwmac fe2a0000.ethernet: Enabled RFS Flow TC (entries=10)
[    2.105274] rk_gmac-dwmac fe2a0000.ethernet: TSO feature enabled
[    2.105811] rk_gmac-dwmac fe2a0000.ethernet: Using 32/32 bits DMA host/device width
[   12.073611] rk_gmac-dwmac fe2a0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   12.076034] rk_gmac-dwmac fe2a0000.ethernet eth0: PHY [stmmac-0:01] driver [Generic PHY] (irq=POLL)
[   13.080525] rk_gmac-dwmac fe2a0000.ethernet: Failed to reset the dma
[   13.081113] rk_gmac-dwmac fe2a0000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   13.081911] rk_gmac-dwmac fe2a0000.ethernet eth0: __stmmac_open: Hw setup failed
~~~

> uninstalling kmod-r8169 and installing kmod-r8125-rss should help this issue.

I'm planning to try this one and come back.

1715173329, 2025-01-19:
How could this happen? There's no PHY connected to MAC on R5C, and the MAC node is not even enabled.

dsakurai, 2025-01-19:
I realized that R5C is [not supported](https://openwrt.org/toh/friendlyarm/nanopi_r5c).

1715173329, 2026-09-09:
https://lore.kernel.org/linux-rockchip/1788158204-161347-1-git-send-email-shawn.lin@rock-chips.com/
I guess this patch may help.

closed 2026-09-13 by openwrt-bot