openwrt/openwrt #17453: `Mesh Point` `Encryption` is always none

bilogic, 2025-01-02:
### Describe the bug

I created a 802.11s Mesh Point interface and set encryption as `WPA3-SAE`, but under `Wireless Overview`, it still displays `Encryption` as `None`

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA8100

### Image kind

Official downloaded image

### Steps to reproduce

1. Create a 802.11s
2. Set encryption as SAE

### Actual behaviour

After creation, look at `Wireless Overview`, it displays `None` for `Encryption`

### Expected behaviour

After creation, `Encryption` should be whatever was defined

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

McGiverGim, 2025-01-02:
This was a known bug, fixed in the next 24.10 version (that is at RC stage). I don't know if it can be fixed for version 23.
[image]

castillofrancodamian, 2025-01-02:
> This was a known bug, fixed in the next 24.10 version (that is at RC stage). I don't know if it can be fixed for version 23. ![Screenshot_20250102-122448](https://private-user-images.githubusercontent.com/2673520/399687087-431a2e41-456a-479a-9600-f92aad4ed00a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU4NDA0NDAsIm5iZiI6MTczNTg0MDE0MCwicGF0aCI6Ii8yNjczNTIwLzM5OTY4NzA4Ny00MzFhMmU0MS00NTZhLTQ3OWEtOTYwMC1mOTJhYWQ0ZWQwMGEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDJUMTc0OTAwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODZjZDFiNjBjZDliODJmZjJhZWM5ZjhjNWFhM2QwNjM4NmQ5NGNkMjVjZGRkMjVjMmE2NTQyZGM2MzVjNDNlZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.n71TwrPGwN_fu8dzaEGeV_7EiAZxS5TMdjAUvxxXyh8)

Now we just need to get the mode translated.

bilogic, 2025-01-03:
> This was a known bug, fixed in the next 24.10 version (that is at RC stage)

Thanks, I opened the issue because it has been there for years and could not find another reporting it.

bilogic, 2025-03-02:
This remains an issue in `OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.014.55016~7046a1c`

blackthorne7, 2025-04-26:
also seeing the same bug, in 24.10.1

nanosonde, 2025-05-05:
Having the same issue that it flips around when enabling/disabling the mesh point interface:

[image]

Model | GL.iNet GL-MT6000
Architecture | ARMv8 Processor rev 4
Target Platform | mediatek/filogic
Firmware Version | OpenWrt 24.10.1 r28597-0425664679 / LuCI openwrt-24.10 branch 25.119.71430~be769af
Kernel Version | 6.6.86

```
phy0-mesh0 ESSID: "my-mesh"
          Access Point: 96:83:C4:A6:B1:40
          Mode: Mesh Point  Channel: 9 (2.452 GHz)  HT Mode: HT20
          Center Channel 1: 9 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -91 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [MediaTek MT7986]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```