openwrt/openwrt #17454: Perfomance issue on Nanopi R3S ( may also on all Rockchip )

issuex, 2025-01-02:
### Describe the bug

from discussion in
https://forum.openwrt.org/t/a-wireguard-comparison-db/187586/54
https://forum.openwrt.org/t/openwrt-support-for-nanopi-r3s/212571/43

On kernel for Rockchip is built with the configuration `CONFIG_PREEMPT=y` make CPU lower performance.
While all the others, Mediatek, Broadcom, Qualcomm, x86, disabled in the kernel configuration by `CONFIG_PREEMPT_NONE_BUILD=y`

i wish someone fix it in official build

similar case topic
https://github.com/cyyself/wg-bench/issues/5

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R3S

### Image kind

Official downloaded image

### Steps to reproduce

- just test performance by https://github.com/cyyself/wg-bench and compare between build

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

samara1532, 2025-01-02:
same on Nanopi R5C
Performance is two times lower than on non-official firmware.
mj22226, Actions-FriendlyWrt

lessload, 2025-01-02:
same issue on Nanopi R2S, FriendlyWRT have better performance than 24-rc4

kien-truong, 2025-01-03:
Indeed, rockchip target is using a configuration for a preemptible kernel, which is more suitable for a desktop system. This has been in use since the very first release with supported rockchip devices (21.02), so the reason for this config is unclear.

For an embedded distro like Openwrt, disabling preemption with `CONFIG_PREEMPT_NONE_BUILD=y` might indeed be a better option.

fakemanhk, 2025-01-05:
For a beast like NanoPi R6S (RK3588) I am observing the same thing, YouTuber tested with FriendlyElec's Wrt flavor and it gave ~1Gbps while official OpenWrt is not getting even half of that number.

kien-truong, 2025-01-06:
It seems FriendlyElec is using `CONFIG_PREEMPT_VOLUNTARY=y`

https://github.com/friendlyarm/kernel-rockchip/blob/6bd846aff11336b7250bb638eb4097207543fc88/arch/arm64/configs/nanopi6_linux_defconfig#L10

devastgh, 2025-01-09:
> to summary how to fix performance issue is.
> 
> ```
> -CONFIG_PREEMPT=y
> -CONFIG_PREEMPTION=y
> -CONFIG_PREEMPT_BUILD=y
> -CONFIG_PREEMPT_COUNT=y
> -CONFIG_PREEMPT_RCU=y
> +CONFIG_PREEMPT_NONE=y
> ```
> 
> wait for testing result from others

Yes, pretty straightforward. Someone has to merge it though.

closed 2025-01-12 by openwrt-bot

TyroSpy, 2025-01-25:
PREEMPT provides a balance between performance and throughput, and it's optimal for routers. While PREEMPT_NONE is only provides the throughput, but gives worse performance. There is no reason to change PREEMP to improve network performance. It might be a bad network driver implementation

issuex, 2025-01-25:
> PREEMPT provides a balance between performance and throughput, and it's optimal for routers. While PREEMPT_NONE is only provides the throughput, but gives worse performance. There is no reason to change PREEMP to improve network performance. It might be a bad Rockchip network driver implementation

show me some benchmark that relate what you reply.

TyroSpy, 2025-01-26:
The problem with low network performance on the NanoPi devices is that OpenWrt used only open source, in particular Realtek network drivers, which have bad implementation, poor performance and speed reduction. This has been discussed for many years and this is easy to verify by install FrendlyWrt or ImmortalWrt and compare network performance. Please revert all changes from approved request https://github.com/openwrt/openwrt/pull/17575

TyroSpy, 2025-01-26:
> show me some benchmark that relate what you reply.
definitely yes, with disabled preampt we will get a **_slight_** increase network speed, but at the expense of stability and performance all other services.

kien-truong, 2025-01-26:
@TyroSpy stop repeating the same argument like a LLM and give us some real use case, in a different GH issue and/or forum post. Don't hijack an already resolved ticket.

issuex, 2025-01-26:
@TyroSpy just show us real test case that shown `stability and performance all other services` was impact. ?

TyroSpy, 2025-01-26:
do whatever you want together china copypaster mainteners

TyroSpy, 2025-01-26:
> It seems FriendlyElec is using `CONFIG_PREEMPT_VOLUNTARY=y`
YES! You have your own Chinese projects FriendlyWrt and ImmortalWrt, go there and do whatever you want. We didn't ask you to help us.

issuex, 2025-01-26:
> china copypaster

For better community, We should talk about lab, test, benchmark and not accept all kind of racism in this community.

kayakyakr, 2025-03-13:
To close out some of this discussion: I'm now using SNAPSHOT with the merged change on an R5C. I'm able to pull 2.35gbps maxing out and even going beyond my 2 gbps connection. I'll update if I see any stability issues, but so far pretty pleased.

Before the update was previously getting 400/down, 800/up.

commit 1c178f36 (2025-01-12): rockchip: disable kernel preemption

This setting is more suitable for device running OpenWRT.
Most OpenWRT targets are already default to this configuration,
and it has shown better performance in VPN (wireguard).

Fix: #17454

Link: https://github.com/openwrt/openwrt/pull/17575
PR #17575 (2025-01-12): rockchip: disable kernel preemption
This setting is more suitable for device running OpenWRT. Most OpenWRT targets are already default to this configuration, and it has shown better performance in VPN (wireguard).

Fix: #17454
commit e9d80cdb (2025-01-12): rockchip: disable kernel preemption

This setting is more suitable for device running OpenWRT.
Most OpenWRT targets are already default to this configuration,
and it has shown better performance in VPN (wireguard).

Fix: #17454

Link: https://github.com/openwrt/openwrt/pull/17575

(cherry picked from commit 1c178f364454c5b73eb453fa0b101ae2f6454d4c)