openwrt/openwrt #17476: Hanging commands with 24.10.0-rc4 release on wrt1200ac target

lespinasse, 2025-01-04:
### Describe the bug

Some commands such as `dmesg`, `ls /proc`, `cat /proc/interrupts` are repeatably hanging on wrt1200ac target.

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1200AC

### Image kind

Self-built image

### Steps to reproduce

ssh into the target and issue one of the following commands:
- dmesg
- ls /proc
- cat /proc/interrupts
(I'm sure there are others)

### Actual behaviour

The listed commands hang. The target doesn't crash - I can still ssh onto it from another terminal - but the commands never complete.
The target can be recovered by issuing the reboot command, and then flashing back a 23.05.x image on it.

### Expected behaviour

The listed commands should quickly complete. The shell should then show the prompt to enter the next command.

### Additional info

This is on wrt1200ac target. I am building my own image using the image builder, inserting a minimal /etc/config/network (contents below) before packing the image. I am able to reproduce the issue without adding any extra packages compared to the default.

/etc/config/network contents as below (nothing special about it, it's just so that the image will be easy to use on my network):
```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1/8'

config device
        option name 'br-trunk'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-trunk.100'
        option proto 'static'
        option ipaddr '10.0.0.1/16'
```

I'm seeing the issue only on wrt1200ac. I have a few other targets I tried 24.10.0-rc4 on (r7800, rpi-4, rpi-0, nanopi r6s), but the hangs are only seen on wrt1200ac target.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2025-01-04:
This looks like a locking problem in some kernel code. 

Please compile the image with these additional configuration options:
```
CONFIG_KERNEL_DETECT_HUNG_TASK=y
CONFIG_KERNEL_HARDLOCKUP_DETECTOR=y
CONFIG_KERNEL_PROVE_LOCKING=y
CONFIG_KERNEL_SOFTLOCKUP_DETECTOR=y
CONFIG_KERNEL_WQ_WATCHDOG=y
```

Maybe it shows us what is hanging. Please connect a UART to the device, trigger the hang and wait for 3 minutes it and check if you got some messages from the kernel.
Please post the output of: ` ./scripts/diffconfig.sh`

lespinasse, 2025-01-05:
I agree a UART connection would help a lot, but I don't currently have one on the device. I feel uneasy about adding one right now as the device is used as my main internet gateway. This could change when 24.10.0 gets out, at which point I could switch to the r6s as my main gateway and have less concern about hacking on the wrt1200ac.

Until then, are there ways I could help that would only involve software ? Like, would ramoops or netconsole be viable options for looking into this ?

lespinasse, 2025-01-12:
So, I am going to close as I can't reproduce the issue anymore. I actually got curious and switched my main gateway earlier than I had planned, so I could use the wrt1200ac device to investigate the issue. After I did so, I wasn't able to reproduce the issue anymore. I switched the main gateway back to the wrt1200ac device, and still no luck as the issue won't trigger.

The good news is that I'm now set up to investigate with serial line debugging if this ever shows up again.

closed 2025-01-12 by lespinasse