openwrt/openwrt #17480: block-mount. 24.10.0-rc4. library link broken by remaining old version SO name.

mic-user, 2025-01-04:
### Describe the bug

As I mentioned at [the article](https://forum.openwrt.org/t/a-bug-report-on-24-10-0-rc4-regarding-libubus/220458)
[image]

### OpenWrt version

24.10.0.-rc4

### OpenWrt release

24.10.0.-rc4

### OpenWrt target/subtarget

mvebu / cortexa9

### Device

Linksys WRT32X

### Image kind

Self-built image

### Steps to reproduce

# block
or
# block info

### Actual behaviour

No error

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Loafter, 2025-01-05:
Same issues with X86 release
it's completely ruin mount system on 3 my devices:)

```
root@OpenWrt:~# /sbin/block
Error loading shared library libubus.so.20241020: No such file or directory (needed by /sbin/block)
Error relocating /sbin/block: ubus_free: symbol not found
Error relocating /sbin/block: ubus_lookup_id: symbol not found
Error relocating /sbin/block: ubus_connect: symbol not found
Error relocating /sbin/block: ubus_invoke_fd: symbol not found
```

robimarko, 2025-01-05:
Yeah, I can confirm that this is a bug.
Somehow builbots pushed package updates to an already tagged release thus breaking all images generated via IB.

robimarko, 2025-01-05:
Ok, packages are continously rebuilt, my fault.

But the issue seems to be that fstools that provide `/sbin/block` are nonshared thus being built with the target which will not get rebuilt while ubus was updated.

robimarko, 2025-01-08:
Updating to RC5 should fix this

mic-user, 2025-01-12:
Fix confirmed.

closed 2025-01-12 by mic-user