openwrt/openwrt #17482: Nmap-full OpenSSL DTLS support

hotfur, 2025-01-04:
### Describe the bug

Install OpenWRT 24.10-rc4 on Xiaomi CR6606, then okpg update and okpg install **nmap-full**. Specify service scan, which resulted in error "nsock_pool_dtls_init called with no OpenSSL DTLS support".

Expected behavior: nmap scan success without error.

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router CR6606

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hotfur, 2025-01-04:
If install OpenWRT 23.05.5 instead of 24.10-rc4, scan completed without any error.

andriej, 2025-01-19:
Can confirm the same on rockchip/armv8 aarch64_generic 24.10.0-rc5

closed 2025-02-19 by hotfur