openwrt/openwrt #17489: wifi txpower value is very low

ef1f, 2025-01-05:
### Describe the bug

wifi txpower value is very low. Selecting a country code does not affect the current txpower

phy0.0-ap0 ESSID: "OpenWrt"
          Access Point: A6:80:59:35:05:EE
          Mode: Master  Channel: 6 (2.437 GHz)  HT Mode: HE40
          Center Channel 1: 8 2: unknown
          Tx-Power: 6 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -83 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy0.1-ap0 ESSID: "OpenWrt"
          Access Point: A6:80:59:35:05:FE
          Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: HE160
          Center Channel 1: 50 2: unknown
          Tx-Power: 7 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -83 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy0.2-ap0 ESSID: "OpenWrt"
          Access Point: A6:80:59:35:25:0E
          Mode: Master  Channel: 53 (6.215 GHz)  HT Mode: HE160
          Center Channel 1: 47 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -83 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

root@OpenWrt:~#  iwinfo phy0.0-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
*  6 dBm (   3 mW)

root@OpenWrt:~#  iwinfo phy0.1-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
*  7 dBm (   5 mW)

root@OpenWrt:~#  iwinfo phy0.2-ap0 txpowerlist
 255 dBm (2147483647 mW)

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pmarques, 2025-01-06:
I have the same issue with all 20.10 release candidates as well as with mtk feed.

lukasz1992, 2025-01-07:
I belive v24.10.0-rc5 should have it fixed

pmarques, 2025-01-07:
I'm still limited to 7dbm
```
root@OpenWrt:~# iwinfo phy0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
*  7 dBm (   5 mW)
root@OpenWrt:~# iw reg get
global
country GB: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
	(5925 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
	(57000 - 71000 @ 2160), (N/A, 40), (N/A)
```

ef1f, 2025-01-07:
RC5, problem is not solved

Gall978, 2025-01-08:
Подтверждаю не исправили:(

Gall978, 2025-01-09:
> RC5, problem is not solved

Yes

hauke, 2025-01-14:
@nbd168  Are you aware of this problem with MT7996E?

ef1f, 2025-01-15:
If you install the firmware from the manufacturer Sinovoip, the Wi-Fi will work fine

pmarques, 2025-01-15:
> If you install the firmware from the manufacturer Sinovoip, the Wi-Fi will work fine

Are you referring to OpenWRT MTK MP4.0?

ef1f, 2025-01-15:
> > If you install the firmware from the manufacturer Sinovoip, the Wi-Fi will work fine
> 
> Are you referring to OpenWRT MTK MP4.0?

Yes, BPI-R4-BE1350-WIFI_MP4_0-SDK-20240620

RonWilliam547, 2025-01-15:
> > > If you install the firmware from the manufacturer Sinovoip, the Wi-Fi will work fine
> > 
> > 
> > Are you referring to OpenWRT MTK MP4.0?
> 
> Yes, BPI-R4-BE1350-WIFI_MP4_0-SDK-20240620

That's proprietary Chinese firmware though. I personally think I'd rather just wait for the open source driver issue to be resolved.

pmarques, 2025-01-15:
> That's proprietary Chinese firmware though. I personally think I'd rather just wait for the open source driver issue to be resolved.

Indeed, not sure if I trust that version of OpenWrt.

What I don’t understand is how other people is apparently using it 🫤

RonWilliam547, 2025-01-15:
> > That's proprietary Chinese firmware though. I personally think I'd rather just wait for the open source driver issue to be resolved.
> 
> Indeed, not sure if I trust that version of OpenWrt.
> 
> What I don’t understand is how other people is apparently using it 🫤

Yeah, I mean it might be fine but I personally don't trust it. Just like I wouldn't trust the stock firmware on those GLiNet routers or really any router honestly. Hopefully this bug isn't too difficult to fix and is able to be resolved in the next release. I'm not a developer so I don't know what all goes into it but I sincerely appreciate every single person who contributes to OpenWRT.

dsbaha, 2025-01-18:
This appears to be resolved in the OpenWrt SNAPSHOT (r28605-9decfc82dc) version provided by the openwrt firmware selector.
```
root@ap1:~# iwinfo phy0.0-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
  25 dBm ( 316 mW)
  26 dBm ( 398 mW)
* 27 dBm ( 501 mW)
```

RonWilliam547, 2025-01-18:
> This appears to be resolved in the OpenWrt SNAPSHOT (r28605-9decfc82dc) version provided by the openwrt firmware selector.
>
> ```
> root@ap1:~# iwinfo phy0.0-ap0 txpowerlist
>    0 dBm (   1 mW)
>    1 dBm [...]

I installed that version and it doesn't appear to be in Luci but I will use the CLI and see. Thank you for the update. I just assumed it hadn't been fixed yet due to Luci.

RonWilliam547, 2025-01-18:
> This appears to be resolved in the OpenWrt SNAPSHOT (r28605-9decfc82dc) version provided by the openwrt firmware selector.
>
> ```
> root@ap1:~# iwinfo phy0.0-ap0 txpowerlist
>    0 dBm (   1 mW)
>    1 dBm [...]

Upon further inspection, I actually had a different snapshot version. I had version r28608-693c1ea81a. The firmware selector page appears to be down at the moment though.

pmarques, 2025-01-18:
I just tried the latest snapshot (`693c1ea81a`) and still have the same power limitation.

<details>
<summary> /etc/openwrt_release </summary>

```sh
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r28608-693c1ea81a'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r28608-693c1ea81a'
DISTRIB_TAINTS=''
```
</details>

```
root@OpenWrt:~# iwinfo 
phy0.0-ap0 ESSID: "OpenWrt 2.4Ghz"
          Access Point: EE:F7:63:3D:EF:62
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: EHT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 6 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

root@OpenWrt:~# iwinfo phy0.0-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
*  6 dBm (   3 mW)
root@OpenWrt:~# iw reg get
global
country GB: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
        (5925 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 71000 @ 2160), (N/A, 40), (N/A)

```

<details>
<summary> iw list frequencies </summary>

```
@OpenWrt:~# iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

root@OpenWrt:~# iw list |  grep -A 10 Freque
                Frequencies:
                        * 2412.0 MHz [1] (6.0 dBm)
                        * 2417.0 MHz [2] (6.0 dBm)
                        * 2422.0 MHz [3] (6.0 dBm)
                        * 2427.0 MHz [4] (6.0 dBm)
                        * 2432.0 MHz [5] (6.0 dBm)
                        * 2437.0 MHz [6] (6.0 dBm)
                        * 2442.0 MHz [7] (6.0 dBm)
                        * 2447.0 MHz [8] (6.0 dBm)
                        * 2452.0 MHz [9] (6.0 dBm)
                        * 2457.0 MHz [10] (6.0 dBm)
--
                Frequencies:
                        * 5180.0 MHz [36] (7.0 dBm)
                        * 5200.0 MHz [40] (7.0 dBm)
                        * 5220.0 MHz [44] (7.0 dBm)
                        * 5240.0 MHz [48] (7.0 dBm)
                        * 5260.0 MHz [52] (7.0 dBm) (no IR, radar detection)
                        * 5280.0 MHz [56] (7.0 dBm) (no IR, radar detection)
                        * 5300.0 MHz [60] (7.0 dBm) (no IR, radar detection)
                        * 5320.0 MHz [64] (7.0 dBm) (no IR, radar detection)
                        * 5500.0 MHz [100] (7.0 dBm) (no IR, radar detection)
                        * 5520.0 MHz [104] (7.0 dBm) (no IR, radar detection)
--
                Frequencies:
                        * 5955.0 MHz [1] (disabled)
                        * 5975.0 MHz [5] (disabled)
                        * 5995.0 MHz [9] (disabled)
                        * 6015.0 MHz [13] (disabled)
                        * 6035.0 MHz [17] (disabled)
                        * 6055.0 MHz [21] (disabled)
                        * 6075.0 MHz [25] (disabled)
                        * 6095.0 MHz [29] (disabled)
                        * 6115.0 MHz [33] (disabled)
                        * 6135.0 MHz [37] (disabled)
```
</details>

RonWilliam547, 2025-01-18:
> I just tried the latest snapshot (`693c1ea81a`) and still have the same power limitation.
> 
> /etc/openwrt_release
> DISTRIB_ID='OpenWrt'
> DISTRIB_RELEASE='SNAPSHOT'
> DISTRIB_REVISION='r28608-693c1ea81a'
> DISTRIB_TARGET='mediatek/filogic'
> DISTRIB_ARCH='aarch64_cortex-a53'
> DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r28608-693c1ea81a'
> DISTRIB_TAINTS=''
> ```
> root@OpenWrt:~# iwinfo 
> phy0.0-ap0 ESSID: "OpenWrt 2.4Ghz"
>           Access Point: EE:F7:63:3D:EF:62
>           Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: EHT20
>           Center Channel 1: 1 2: unknown
>           Tx-Power: 6 dBm  Link Quality: unknown/70
>           Signal: unknown  Noise: -92 dBm
>           Bit Rate: unknown
>           Encryption: WPA3 SAE (CCMP)
>           Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
>           Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
>           TX power offset: none
>           Frequency offset: none
>           Supports VAPs: yes  PHY name: phy0
> 
> root@OpenWrt:~# iwinfo phy0.0-ap0 txpowerlist
>    0 dBm (   1 mW)
>    1 dBm (   1 mW)
>    2 dBm (   1 mW)
>    3 dBm (   1 mW)
>    4 dBm (   2 mW)
>    5 dBm (   3 mW)
> *  6 dBm (   3 mW)
> root@OpenWrt:~# iw reg get
> global
> country GB: DFS-ETSI
>         (2400 - 2483 @ 40), (N/A, 20), (N/A)
>         (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
>         (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
>         (5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
>         (5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
>         (5925 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
>         (57000 - 71000 @ 2160), (N/A, 40), (N/A)
> ```
> 
> iw list frequencies

Yeah, I still have the same power limitation on that version as well but it's a different version number than the one that they mentioned. Maybe there was a newer one that got lost when they were restoring connection to the download server? 🤔

pmarques, 2025-01-18:
The version that @dsbaha mentioned is from the previous day, the one we tested is more recent.

[image]

RonWilliam547, 2025-01-18:
> The version that [@dsbaha](https://github.com/dsbaha) mentioned is from the previous day, the one we tested is more recent.
> 
> <img alt="Image" width="1373" src="https://private-user-images.githubusercontent.com/441283/404550674-e15a6719-9207-4147-b70e-46a712f5eee2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzcyMjEzNjgsIm5iZiI6MTczNzIyMTA2OCwicGF0aCI6Ii80NDEyODMvNDA0NTUwNjc0LWUxNWE2NzE5LTkyMDctNDE0Ny1iNzBlLTQ2YTcxMmY1ZWVlMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExOFQxNzI0MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMmRiNWZmMDU0NTg5NWRiZTU0N2NhNDUzMmMzNmRlYzgwYjkyM2I3ODQyN2Y2YTM0ZWNmNTJmZjE2ZTY3OTE3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Ci-VnLnTjmpRSAxlGHxcrbNCPd0GM6U6EeDjY0pbu6s">

Interesting. I think I tried the snapshot 2 days ago but only in Luci and in Luci it didn't give more options. Now it's possible that it had actually been fixed but not reflected in Luci but at the time I didn't test the CLI for it and I don't have that version anymore. I wonder if @dsbaha would be interested in sharing that version for testing purposes or if there is a way to go back and get that version.

twwn, 2025-01-18:
> Looks like an issue in https://wireless.docs.kernel.org/en/latest/en/developers/regulatory/wireless-regdb.html . Neither OpenWrt nor mt76 driver can change or modify whatever TX power limits have been set by wireless-regdb for your country. Please bring this up at wireless-regdb mailing list at http://lists.infradead.org/mailman/listinfo/wireless-regdb .
> 
> Disclaimer: I am not a dev, neither at OpenWrt nor at wireless-regdb. I am just a user.

I've built & installed rany2's fork (@ f22feba), which patches in [an unrestricted](https://github.com/rany2/openwrt/commit/76f574574af869210600576c603e2ca315c8ce70) "ZZ" test country that I set. `iw` still reports a 6/7 dBm limit for all frequencies.

dsbaha, 2025-01-19:
Hi, hope this helps.
```
root@ap1:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r28608-693c1ea81a'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r28608-693c1ea81a'
DISTRIB_TAINTS=''
```
```
root@ap1:~# iwinfo
phy0.0-ap0 ESSID: "jNetGP"
          Access Point: 1E:55:BB:24:7B:DD
          Mode: Master  Channel: 11 (2.462 GHz)  HT Mode: EHT20
          Center Channel 1: 11 2: unknown
          Tx-Power: 27 dBm  Link Quality: 61/70
          Signal: -49 dBm  Noise: -88 dBm
          Bit Rate: 65.0 MBit/s
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy0.0-ap1 ESSID: unknown
          Access Point: 1E:55:BB:24:7B:DC
          Mode: Master  Channel: 11 (2.462 GHz)  HT Mode: EHT20
          Center Channel 1: 11 2: unknown
          Tx-Power: 27 dBm  Link Quality: 56/70
          Signal: -54 dBm  Noise: -88 dBm
          Bit Rate: 46.5 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy0.0-ap2 ESSID: unknown
          Access Point: 1E:55:BB:24:7B:DF
          Mode: Master  Channel: 11 (2.462 GHz)  HT Mode: EHT20
          Center Channel 1: 11 2: unknown
          Tx-Power: 27 dBm  Link Quality: 59/70
          Signal: -51 dBm  Noise: -88 dBm
          Bit Rate: 77.9 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy0.1-ap0 ESSID: "jNetGP"
          Access Point: 1E:55:BB:24:7B:ED
          Mode: Master  Channel: 153 (5.765 GHz)  HT Mode: EHT80
          Center Channel 1: 155 2: unknown
          Tx-Power: 26 dBm  Link Quality: 58/70
          Signal: -52 dBm  Noise: -88 dBm
          Bit Rate: 973.6 MBit/s
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy0.1-ap1 ESSID: unknown
[... 146 lines cut ...]
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
  25 dBm ( 316 mW)
  26 dBm ( 398 mW)
* 27 dBm ( 501 mW)
```

rmandrad, 2025-01-19:
what 'iw dev' is showing for those with the txpower issue ?

twwn, 2025-01-19:
@dsbaha 

Thank you. All I can say… weird, I'm only the next day's commits ahead. When I first copied your `/etc/config/wireless` (after `iw reg set US`), there seemed to be higher limits for 2/5G (only in `iw`), but the second I touched the config (cutting down to 1 AP/radio, …) it broke down. Now Luci won't even show the Wireless tab anymore until I delete the file & reboot.

@rmandrad 

> what 'iw dev' is showing for those with the txpower issue ?

Default config:

```
root@OpenWrt:~# iw dev
phy#0
	Interface phy0.2-ap0
		ifindex 15
		wdev 0x4
		addr 2e:90:02:04:e3:bf
		type AP
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
		Radios: 2
	Interface phy0.1-ap0
		ifindex 14
		wdev 0x3
		addr 2e:90:02:04:e3:cf
		ssid OpenWrt
		type AP
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 7.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
		Radios: 0 1 2
	Interface phy0.0-ap0
		ifindex 12
		wdev 0x1
		addr 2e:90:02:04:e3:bf
		ssid OpenWrt
		type AP
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 6.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
		Radios: 0
```

Third 6G radio won't come up for its `txpower` to show there, but `iw dev phy0 info` for `5955.0 MHz [1]` to `6415.0 MHz [93]` prints `23.0 dBm` for it as `Band 4` (there is no _Band 3_).

kartman521, 2025-01-19:
I have the same problem on the latest snapshot ()

`root@OpenWrt:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r28628-0f8a5a5fd6'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r28628-0f8a5a5fd6'
DISTRIB_TAINTS=''`

`root@OpenWrt:~# iwinfo phy0.1-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
*  7 dBm (   5 mW)`

`root@OpenWrt:~# iw dev
phy#0
        Interface phy0.1-ap0
                ifindex 15
                wdev 0x5
                addr 8e:9f:0b:a5:8e:0e
                ssid OpenWrt
                type AP
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 7.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
                Radios: 1
        Interface phy0.0-sta0
                ifindex 14
                wdev 0x4
                addr 8e:9f:0b:a5:7e:fe
                ssid opera
                type managed
                channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
                txpower 6.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
                Radios: 0`

RonWilliam547, 2025-01-19:
I just tested with the latest snapshot and this is what I've got. Also important to mention, if I set the country code to USA and enable 6GHZ, it won't actually start. If I set it to France, it will be after setting the security to WPA3.

[image]

[image]

[image]

[image]

testuser7, 2025-01-20:
> if I set the country code to USA and enable 6GHZ, it won't actually start.

Currently it is not possible to run AP on 6GHz for `US` due to `NO-IR` on all channels: https://forum.openwrt.org/t/adding-openwrt-support-for-zyxel-ex5601-t0/155914/296

testuser7, 2025-01-20:
I think that using AFC will change this situation: https://www.wi-fi.com/wi-fi-afc

RonWilliam547, 2025-01-22:
Question, how many people have installed OpenWRT to the emmc? Do you have the low transmit power issue? How many people are running off an SD card? Do you have the low transmit power issue? I'm currently running off an SD card and I have the low transmit power issue. I wonder if installing to the emmc would fix it or if that has anything to do with this. 🤔

pmarques, 2025-01-22:
I tried all opinions before, although the last snapshots were only from the SD (I don't have it connected to the internet and was a bit of waist of time). I even tried to reset the EMMC with the MTK 4.0 in between

RonWilliam547, 2025-01-22:
> I tried all opinions before, although the last snapshots were only from the SD (I don't have it connected to the internet and was a bit of waist of time). I even tried to reset the EMMC with the MTK 4.0 in between

Yeah, I figured the emmc had nothing to do with this issue but I figured I'd ask just in case.

netzding, 2025-01-25:
> I'm also experiencing the same issue. Changing the Wi-Fi country code or even patching the regdb didn't resolve the low transmit power problem for my BE1400 Wi-Fi card (my hardware revision V1.0) either.

Have the exact same issue with the exact same hardware.

SockToy, 2025-01-26:
BPI R4 board v1.1, same issue. Have tried snapshot & 24.10-RC6. 

Sinovoip build does solve the txpower issue but has many other issues OpenWRT solves. It's not a fix. 

I have run from both the SD slot and the eMMC with RC6.

Details comparing iw outputs and some iperf testing between sinovoip and snapshot here:
https://forum.openwrt.org/t/banana-bpi-r4-wifi7-status/201051/1149

rmandrad, 2025-01-26:
> BPI R4 board v1.1, same issue. Have tried snapshot & 24.10-RC6.
>
> Sinovoip build does solve the txpower issue but has many other issues OpenWRT solves. It's not a fix.
>
> I have run from both the SD [...]

can you check /var/run/hostapd-phy0-*.conf for any power statements ? if there is a suggested on the forum to remove a recent commit update from mac80211.sh files

on your wireless config do you have any country parameter? if not you should add... all radios should have the same country

pmarques, 2025-01-26:
> pls share /var/run/hostapd-phy0-*.conf ... remove the password parameter before sharing
>
> on your wireless config do you have any country parameter? if not you should add... all radios should have the same country

To make sure I've a clean state, I just flashed my SD Card with 24.10 rc6 and setup the wifi using UCI. I haven't touched anything else.

<details><summary>wifi Setup using uci</summary>

```
uci set wireless.radio0.disabled='0'
uci set wireless.radio0.country='GB'
uci set wireless.radio0.txpower='20'
uci set wireless.default_radio0.encryption='sae'
uci set wireless.default_radio0.ssid='OpenWrt 2.4Ghz'
uci set wireless.default_radio0.key="=********"
uci set wireless.radio1.disabled='0'
uci set wireless.radio1.country='GB'
uci set wireless.radio1.txpower='20'
uci set wireless.default_radio1.encryption='sae'
uci set wireless.default_radio1.ssid='OpenWrt 5Ghz'
uci set wireless.default_radio1.key="********"
uci set wireless.radio2.disabled='0'
uci set wireless.radio2.country='GB'
uci set wireless.radio2.txpower='20'
uci set wireless.default_radio2.encryption='sae'
uci set wireless.default_radio2.ssid='OpenWrt 6Ghz'
uci set wireless.default_radio2.key='=********'
uci commit wireless
wifi reload
```

</details> 

<details><summary>cat /var/run/hostapd-phy0.0.conf</summary>

```
$ cat /var/run/hostapd-phy0.0.conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=GB
ieee80211d=1
hw_mode=g
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
beacon_int=100
stationary_ap=1
channel=1
chanlist=1

#num_global_macaddr=1
#macaddr_base=
ieee80211n=1
ht_coex=0
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
ieee80211ax=1
he_su_beamformer=1
he_su_beamformee=1
he_mu_beamformer=1
he_bss_color=128
he_spr_sr_control=3
he_default_pe_duration=4
he_rts_threshold=1023
he_mu_edca_qos_info_param_count=0
he_mu_edca_qos_info_q_ack=0
he_mu_edca_qos_info_queue_request=0
he_mu_edca_qos_info_txop_request=0
he_mu_edca_ac_be_aifsn=8
[... 240 lines cut ...]
sae_pwe=2
wpa_passphrase=********
sae_password_file=/var/run/hostapd-phy0.2-ap0.sae
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=OpenWrt 6Ghz
bridge=br-lan   
wds_bridge=
snoop_iface=br-lan
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=SAE
okc=1
ieee80211w=2
group_mgmt_cipher=AES-128-CMAC
dynamic_vlan=0  
vlan_naming=1   
vlan_no_bridge=1
vlan_file=/var/run/hostapd-phy0.2-ap0.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=3a:76:4d:4f:75:69
#default_macaddr
```

</details>

pmarques, 2025-01-26:
I just compiled 24.10 branch with the mtk feed and seems like the eeprom is the issue:

```
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/band0/txpower_sku | grep eeprom
eeprom maximum power:   6 [dBm]
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/band1/txpower_sku | grep eeprom
eeprom maximum power:   7 [dBm]
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/band2/txpower_sku | grep eeprom
eeprom maximum power:   24 [dBm]
```

This is using the latest available eprom from the mtk feed from Nov 2024
```
[   18.443674] mt7996e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20241106004238a
[   18.660707] mt7996e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20241106004234
[   18.718025] mt7996e 0000:01:00.0: DSP Firmware Version: ____000000, Build Time: 20241106003710
[   18.772135] mt7996e 0000:01:00.0: WA Firmware Version: ____000000, Build Time: 20241106004201
```

rmandrad, 2025-01-26:
that's odd ... I don't have the  /sys/kernel/debug/ieee80211/phy0/mt76/band[0-2]/txpower_sku directory

what is  /sys/kernel/debug/ieee80211/phy0/netdev:phy0.1-ap0/txpower showing ? do you have this directory ?

when you compiled have you chosen any other driver than just mt7996 ? just saying because the debugfs.c for mt7996 doesn't create this directory

what lsmod | grep mt79  ... shows ?

another point have you tried to have phy1 setup on AX ? I see BE on the hostpad conf setup ... also do not set txpower

pmarques, 2025-01-26:
> that's odd ... I don't have the /sys/kernel/debug/ieee80211/phy0/mt76/band[0-2]/txpower_sku directory

I only get this when compiling with https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/refs/heads/master/autobuild/unified/

> what is /sys/kernel/debug/ieee80211/phy0/netdev:phy0.1-ap0/txpower showing ? do you have this directory ?

Not with the MTK version :/

> when you compiled have you chosen any other driver than just mt7996 ? just saying because the debugfs.c for mt7996 doesn't create this directory

Not really, used the default setup / .config (Only remove the `crypto-eip` since was not compiling)

> what lsmod | grep mt79 ... shows ?

```
root@OpenWrt:~# lsmod | grep mt7
cfg80211              753664  4 mt7996e,mt76_connac_lib,mt76,mac80211
mac80211              958464  3 mt7996e,mt76_connac_lib,mt76
mt76                  110592  2 mt7996e,mt76_connac_lib
mt76_connac_lib        49152  1 mt7996e
mt7996e               405504  0 
```

> another point have you tried to have phy1 setup on AX ? I see BE on the hostpad conf setup ... also do not set txpower

Do you mean `phy0.1`? I tried multiple setups before and nothing worked :(

rmandrad, 2025-01-26:
ok so you are using the mtk-sdk version ...there are a number patches that are using sku there and limiting txpower ... don't see something similar with openwrt...

pmarques, 2025-01-26:
Silly question, is `/sys/kernel/debug/ieee80211/phy0/mt76/eeprom` supposed to be the same as one of the eeproms `/lib/firmware/mediatek/mt7996/mt7996_eeprom*`? 

```
root@OpenWrt:~# ls -la /lib/firmware/mediatek/mt7996/mt7996_eeprom*
465b6b60dc996c8037d0d11cf690e291  /lib/firmware/mediatek/mt7996/mt7996_eeprom.bin
cf763d6f08f7f20dd75442907ee74ddb  /lib/firmware/mediatek/mt7996/mt7996_eeprom_233.bin
cf763d6f08f7f20dd75442907ee74ddb  /lib/firmware/mediatek/mt7996/mt7996_eeprom_233_2i5i6i.bin
23a759dc8c54c44ac09a5de4cf325b37  /lib/firmware/mediatek/mt7996/mt7996_eeprom_2i5i6i.bin

root@OpenWrt:~# md5sum /sys/kernel/debug/ieee80211/phy0/mt76/eeprom
81be409e8fb75671c925bc82aaee6a8f  /sys/kernel/debug/ieee80211/phy0/mt76/eeprom
```

rmandrad, 2025-01-26:
no ... the /sys/kernel/debug is a place for the drivers & code to place debug info... where the /lib/firmware are the proprietary files provided by the vendors I am assuming the plan from mediatek is to limit the power via the eeprom (similar to the way qualcomm does) bypassing regdb

pmarques, 2025-01-26:
> you should be using the /lib/firmware/mediatek/mt7996/mt7996_eeprom_233.bin file as i guess you are using the be14 wifi card

I think so, if I remove the files I get an error at least:
```
[   13.363674] mt7996e 0000:01:00.0: Direct firmware load for mediatek/mt7996/mt7996_rom_patch_233.bin failed with error -2
[   13.374548] mt7996e 0000:01:00.0: Falling back to sysfs fallback for: mediatek/mt7996/mt7996_rom_patch_233.bin
[   13.404265] mt7996e: probe of 0000:01:00.0 failed with error -12
```

rmandrad, 2025-01-26:
> [@rmandrad](https://github.com/rmandrad) are you by any chance using a R4 with the BE14? If yes, which version of the hardware do you have?

yes I am using the r4 / be14 don't know the version but bought it about july 24 if i remember right

rmandrad, 2025-01-26:
in openwrt (not the sdk) bring the wifi down (iw dev should not show any devices)

create an interface

iw phy phy0 interface add phy1-ap0 type station

iw reg set GB

what is the power showing ?

rmandrad, 2025-01-26:
# SSID
ssid=example
# PASSPHRASE
wpa_passphrase=password123
# Band: a = 5GHz & 6GHz (a/n/ac/ax), g = 2Ghz (b/g/n)
hw_mode=a
# Channel availability varies by country
# https://en.wikipedia.org/wiki/List_of_WLAN_channels
# US:
# 2.4 GHz (1-11)
# 5 GHz ((36(42) and 149(155))
# 5 GHz DFS ((52(58), 100(106), 116(122), 132(138))
# 6 GHz (see op_class below)
# Channel
channel=acs_survey
#chanlist=1
# The operating class is an indication of the 6 GHz channel size. An
# operating class of 134 indicates a 160 MHz channel. Likewise, 133 
# denotes an 80 MHz channel, 132 indicates a 40 MHz channel, and 131
# indicates a channel size of 20 MHz.
# 133 signifies channel width: 80 MHz
#   channels 7, 23, 39, 55, 71, 87, 103, 119, 135, 151, 167, 183, 199, 215
# 134 signifies channel width: 160 MHz
#   channels 15, 47, 79, 111, 143, 175, 207
#op_class=133
# Country code
# https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
country_code=GB
# All environments
#country3=0x20
# Outdoor
#country3=0x4f
# Indoor environment only
#country3=0x49
# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels if supported (requires ieee80211d=1)
#ieee80211h=1

# Bridge interface
bridge=br-lan
# WiFi interface (wlan0 should be changed to your interface name)
interface=phy0-ap0

# Set hostapd driver (nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers)
driver=nl80211

# hostapd event logger configuration
#
# Module bitfield (ORed bitfield of modules that will be logged; -1 = all
# modules):
# bit 0 (1) = IEEE 802.11
# bit 1 (2) = IEEE 802.1X
# bit 2 (4) = RADIUS
# bit 3 (8) = WPA
# bit 4 (16) = driver interface
# bit 6 (64) = MLME
#
# Levels (minimum value for logged events):
#  0 = verbose debugging
#  1 = debugging
#  2 = informational messages
#  3 = notification
#  4 = warning
#
logger_syslog=1
logger_syslog_level=2

# Control interface
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

# Various settings
#beacon_int=100
#dtim_period=2
#multicast_to_unicast=0
#tx_queue_data2_aifs=1
#tx_queue_data2_cwmin=7
#tx_queue_data2_cwmax=15
#tx_queue_data2_burst=3.0
tx_queue_data2_burst=2.0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
skip_inactivity_poll=1
preamble=1

# Security
# authentication algorithms (3 = both are allowed, use 3 for SAE)
auth_algs=3
# MAC address -based authentication (0 = accept unless in deny list)
macaddr_acl=0
# hidden SSID (1 = hidden)
ignore_broadcast_ssid=0
# Opportunistic Key Caching (1 = enabled)
okc=1
# Enable WPA. (2 is required for WPA2, mixed and WPA3)
wpa=2
# Pairwise cipher
wpa_pairwise=CCMP CCMP-256
rsn_pairwise=CCMP CCMP-256
#
# Activate only one of the following 3 sections
#
# 1. WPA2 Personal
#wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256
#
# 2. WPA2/WPA3 Transitional (mixed)
wpa_key_mgmt=SAE WPA-PSK WPA-PSK-SHA256
ieee80211w=1
sae_require_mfp=1
#
# 3. WPA3 Personal (required for WiFi 6e - 6 GHz)
#wpa_key_mgmt=SAE
#ieee80211w=2

# IEEE 802.11n (WiFi 4) band 2, 5 GHz configuration
ieee80211n=1
wmm_enabled=1
ht_coex=0
#
# generic setting - 20 MHz channel width
#ht_capab=[SHORT-GI-20]
#
# generic setting - 40 MHz channel width (if allowed)
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
#
# mt7921au  (HT capabilities 0x9ff)
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
#

# IEEE 802.11ac (WiFi 5) configuration
ieee80211ac=1
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
#
# generic setting
#vht_capab=[SHORT-GI-80]
#
# mt7921au (VHT capabilities 0x339071b2)
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
#

# IEEE 802.11ax (WiFi 6) configuration
ieee80211ax=1
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
he_oper_chwidth=2
# HE center channel (chan + 6)
he_oper_centr_freq_seg0_idx=42
# used by clients to discern the source of interference
# each AP in your area needs to use a different number
# allowed: 1-63
#he_bss_color=38
#he_default_pe_duration=4
#he_rts_threshold=1023
#he_mu_edca_qos_info_param_count=0
#he_mu_edca_qos_info_q_ack=0
#he_mu_edca_qos_info_queue_request=0
#he_mu_edca_qos_info_txop_request=0
#he_mu_edca_ac_be_aifsn=8
#he_mu_edca_ac_be_aci=0
#he_mu_edca_ac_be_ecwmin=9
#he_mu_edca_ac_be_ecwmax=10
#he_mu_edca_ac_be_timer=255
#he_mu_edca_ac_bk_aifsn=15
#he_mu_edca_ac_bk_aci=1
#he_mu_edca_ac_bk_ecwmin=9
#he_mu_edca_ac_bk_ecwmax=10
#he_mu_edca_ac_bk_timer=255
#he_mu_edca_ac_vi_ecwmin=5
#he_mu_edca_ac_vi_ecwmax=7
#he_mu_edca_ac_vi_aifsn=5
#he_mu_edca_ac_vi_aci=2
#he_mu_edca_ac_vi_timer=255
#he_mu_edca_ac_vo_aifsn=5
#he_mu_edca_ac_vo_aci=3
#he_mu_edca_ac_vo_ecwmin=5
#he_mu_edca_ac_vo_ecwmax=7
#he_mu_edca_ac_vo_timer=255

pmarques, 2025-01-26:
Thanks for following up the whole day. I will need test this tomorrow.

danpawlik, 2025-01-27:
@pmarques hi, I don't remember if I asked you via private email conversation, to make steps:
- write newest image to the sdcard
- change switch to use SD card
- via UART in u-boot menu, choose option 7: install bootloader, system to eMMC
- switch to eMMC

Check results. 
I don't have such issue on my device, but when I get it, I have use that option for all places: eMMC and the same for NAND. 
Does Sinovoip says something?

kartman521, 2025-01-27:
> Check results.

I did this before for snapshots (around January 15th), it did not affect the available power for 2.5 and 5 radios. I also tried various modified uboots (with NVME support etc., it did not make a difference)

SockToy, 2025-01-28:
> in openwrt (not the sdk) bring the wifi down (iw dev should not show any devices)
> 
> create an interface
> 
> iw phy phy0 interface add phy0-ap0 type station

Below is the result from:
- Fresh RC6 install to system
- NO wifi config via uci/luci
- iw dev is empty
- create device with iw phy phy0 interface add phy0-ap0 type station
- NO hostapd / interface up yet
- 
```
Linux ap-bpi-24-rc6 6.6.73 #0 SMP Wed Jan 22 19:52:54 2025 aarch64 GNU/Linux
root@ap-bpi-24-rc6:~# 
root@ap-bpi-24-rc6:~# iw list | grep -i dbm
```
			* 2412.0 MHz [1] (6.0 dBm)
			* 2417.0 MHz [2] (6.0 dBm)
			* 2422.0 MHz [3] (6.0 dBm)
			* 2427.0 MHz [4] (6.0 dBm)
			* 2432.0 MHz [5] (6.0 dBm)
			* 2437.0 MHz [6] (6.0 dBm)
			* 2442.0 MHz [7] (6.0 dBm)
			* 2447.0 MHz [8] (6.0 dBm)
			* 2452.0 MHz [9] (6.0 dBm)
			* 2457.0 MHz [10] (6.0 dBm)
			* 2462.0 MHz [11] (6.0 dBm)
			* 2467.0 MHz [12] (6.0 dBm) (no IR)
			* 2472.0 MHz [13] (6.0 dBm) (no IR)
			* 2484.0 MHz [14] (6.0 dBm) (no IR)
			* 5180.0 MHz [36] (7.0 dBm)
			* 5200.0 MHz [40] (7.0 dBm)
			* 5220.0 MHz [44] (7.0 dBm)
			* 5240.0 MHz [48] (7.0 dBm)
			* 5260.0 MHz [52] (7.0 dBm) (no IR, radar detection)
			* 5280.0 MHz [56] (7.0 dBm) (no IR, radar detection)
			* 5300.0 MHz [60] (7.0 dBm) (no IR, radar detection)
			* 5320.0 MHz [64] (7.0 dBm) (no IR, radar detection)
			* 5500.0 MHz [100] (7.0 dBm) (no IR, radar detection)
			* 5520.0 MHz [104] (7.0 dBm) (no IR, radar detection)
			* 5540.0 MHz [108] (7.0 dBm) (no IR, radar detection)
			* 5560.0 MHz [112] (7.0 dBm) (no IR, radar detection)
			* 5580.0 MHz [116] (7.0 dBm) (no IR, radar detection)
			* 5600.0 MHz [120] (7.0 dBm) (no IR, radar detection)
			* 5620.0 MHz [124] (7.0 dBm) (no IR, radar detection)
			* 5640.0 MHz [128] (7.0 dBm) (no IR, radar detection)
			* 5660.0 MHz [132] (7.0 dBm) (no IR, radar detection)
			* 5680.0 MHz [136] (7.0 dBm) (no IR, radar detection)
			* 5700.0 MHz [140] (7.0 dBm) (no IR, radar detection)
			* 5720.0 MHz [144] (7.0 dBm) (no IR, radar detection)
			* 5745.0 MHz [149] (7.0 dBm) (no IR)
			* 5765.0 MHz [153] (7.0 dBm) (no IR)
			* 5785.0 MHz [157] (7.0 dBm) (no IR)
			* 5805.0 MHz [161] (7.0 dBm) (no IR)
			* 5825.0 MHz [165] (7.0 dBm) (no IR)
```
```

In terms of creating a hostapd file, you're requesting one not configured by uci/luci using /etc/config/wireless? Like, go hand craft it? Any particular settings you'd want to see? I could just lift the one from the sinovoip build and see how that works out, but usually I let openwrt just construct what it wants.

rmandrad, 2025-01-28:
@SockToy just want to see the radio starting outside openwrt and use iw to create the interface and hostapd to start the interace and see the txpower reading. So, just use any conf file that works (you can use a version of the conf files that are at /var/run )

note - please use the openwrt build as we are not troubleshooting the mtk-sdk

danpawlik, 2025-01-28:
Wondering if one of those commits are not breaking that: b795e5cbcffc54e858bc8d3bde0031a2ca32377e or 65aff89f641f6adccac80ee73b02b99a904739d5.
From the other side, I'm using image that contains those commits and it works normally (I triggered building new image right now. If someone want to check my image, they are available here: https://github.com/danpawlik/openwrt-builder/releases )

wirl, 2025-01-28:
Could this issue also be responsible for slower wireless speeds in non-AP mode scenarios? My head is spinning trying to figure out why the mt7915e in my router maxes out around 100mbit/s up/down in STA mode.

kartman521, 2025-01-28:
> If someone want to check my image, they are available here: https://github.com/danpawlik/openwrt-builder/releases )

Unfortunately, the power on the new build is also very low(

[image]

danpawlik, 2025-01-28:
Eh, that's really odd. Last what comes to my mind right now is to check community build available in that [thread](https://forum.openwrt.org/t/banana-bpi-r4-all-related-to-mtk-sdk/221080). Just to check, if that will help.

RonWilliam547, 2025-01-28:
> [@kartman521](https://github.com/kartman521) Eh, that's really odd. Last what comes to my mind right now is to check community build available in that [thread](https://forum.openwrt.org/t/banana-bpi-r4-all-related-to-mtk-sdk/221080). Just to check, if that will help.
> 
> EDIT: I see that [@pmarques](https://github.com/pmarques) already tried that [#17489 (comment)](https://github.com/openwrt/openwrt/issues/17489#issuecomment-2614325108) ...

I can confirm on my end using your latest image, it's probably the best OpenWRT image we have available right now but the txpower issue is still present. The txpower for 2.4 GHZ only goes up to 6 dBm. The txpower for 5 GHZ goes up to 7 dBm. The txpower for 6 GHZ goes up to 7 dBm in Luci and when using the command iwinfo phy0.2-ap0 txpowerlist but when leaving all the txpower values as "driver default", 2.4 GHZ goes to 6 dBm, 5 GHZ goes to 7 dBm, but 6 GHZ goes to 23 dBm. Is this a driver issue at the moment or is there something to try to fix the issue? I know we all really appreciate your assistance thus far and all the work the OpenWRT team does. 😊

rmandrad, 2025-01-29:
what cat /sys/kernel/debug/ieee80211/phy0/netdev:phy0.0-ap0/txpower 

and 

systool -vm mt7996e

shows ?

RonWilliam547, 2025-01-29:
> what cat /sys/kernel/debug/ieee80211/phy0/netdev:phy0.0-ap0/txpower
> 
> and
> 
> systool -vm mt7996e
> 
> shows ?

This is what I get

```bash
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/netdev:phy0.0-ap0/txpower
6
root@OpenWrt:~# systool -vm mt7996e
Module = "mt7996e"

  Attributes:
    coresize            = "118784"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "0"
    taint               = "O"
    uevent              = <store method only>

  Parameters:
    sr_scene_detect     = "Y"
    wed_enable          = "N"

  Sections:
    .altinstructions    = "0xffffffc0790ef030"
    .bss                = "0xffffffc0790ec680"
    .data..once         = "0xffffffc0790ec398"
    .data               = "0xffffffc0790ec040"
    .exit.data          = "0xffffffc0790ec3a0"
    .exit.text          = "0xffffffc0790e9ba8"
    .gnu.linkonce.this_module= "0xffffffc0790ec3c0"
    .init.data          = "0xffffffc0790f6000"
    .init.plt           = "0xffffffc0790f40c0"
    .init.text          = "0xffffffc0790f4000"
    .note.Linux         = "0xffffffc0790ef000"
    .plt                = "0xffffffc0790e9c00"
    .rodata             = "0xffffffc0790ef078"
    .rodata.str1.8      = "0xffffffc0790f1988"
    .strtab             = "0xffffffc0790f87c8"
    .symtab             = "0xffffffc0790f6008"
    .text.ftrace_trampoline= "0xffffffc0790ea5fc"
    .text               = "0xffffffc0790d4000"
    __bug_table         = "0xffffffc0790ec000"
    __param             = "0xffffffc0790f2f98"

root@OpenWrt:~#
```

rmandrad, 2025-01-29:
I am assuming that "SW4 is turned to the "ON" position, and the 12V LED will lights up when power on" is this right ?

RonWilliam547, 2025-01-29:
> I am assuming that "SW4 is turned to the "ON" position, and the 12V LED will lights up when power on" is this right ?
> 
> [image]

Yup, sure is. I can connect devices to the network but the range is not good. I just don't ave any devices connected at the moment.

pmarques, 2025-01-29:
I see @RonWilliam547  followed on previous questions, here are my results

1. Using SD Card with `OpenWrt 24.10.0-rc6, r28388-58d0057481`
2. `iw phy phy0 interface add phy0-ap0 type station`
3. `cat /sys/kernel/debug/ieee80211/phy0/netdev:phy0-0-ap0/txpower`
```
cat: can't open '/sys/kernel/debug/ieee80211/phy0/netdev:phy0-0-ap0/txpower': No such file or directory
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/netdev:phy0-ap0/txpower
-2147483648
````
4. `systool -vm mt7996e`
```
Module = "mt7996e"

  Attributes:
    coresize            = "118784"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "0"
    taint               = "O"
    uevent              = <store method only>

  Parameters:
    coredump_memdump    = "N"
    sr_scene_detect     = "Y"
    wed_enable          = "N"
```

---

I tried to get the values after setting up the AP via UCI and this is an interesting one:

```
cat /sys/kernel/debug/ieee80211/phy0/netdev:phy0.0-ap0/txpower
6

cat /sys/kernel/debug/ieee80211/phy0/netdev:phy0.0-ap0/ap_power_level 
-2147483648

cat /sys/kernel/debug/ieee80211/phy0/netdev:phy0.0-ap0/user_power_level 
20
```

----

> I am assuming that "SW4 is turned to the "ON" position, and the 12V LED will lights up when power on" is this right ?

@rmandrad if the switch is off, you can't even see the PCIe / device since the card doesn't start (the power light doesn't come up). I also tried multiple power sources and still have the same limitation

[297 more replies]