openwrt/openwrt #17490: APK does not install local files

grosjo, 2025-01-05:
### Describe the bug

Trying to install and packages.apk from the local filsystem. apk tries irremediably to reach something web site on teh internet....

### OpenWrt version

r28363-f10ee1e209

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Comfast CF-WR650AC v2

### Image kind

Self-built image

### Steps to reproduce

Select "M" in the building menu
Upload/Scp the .apk file to the device
apk add file.apk
-> apk tries to connect some web site... :(

### Expected behaviour

apk install the designated apk file

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2025-01-06:
It will install the local APK, but will first try to update the remote manifest.

grosjo, 2025-01-06:
No, it does not

grosjo, 2025-01-06:
and the search on the internet relates to a " standard built"  which is a bit of a nonsense for a self-built firmware

robimarko, 2025-01-06:
It for sure does, as I have used apk add on local apk files plenty of times when debugging the initial APK support

grosjo, 2025-01-08:
I recompiled my firmware, I get

```
root@OpenWrt:/tmp# apk add -v ./kmod-ath9k-6.6.69.6.12.6-r1.apk 
ERROR: Unable to read database: No such file or directory
ERROR: Failed to open apk database: No such file or directory

```

robimarko, 2025-01-08:
What does `apk info` say?
And what do you have in `/lib/apk/db`?

Cause this error screams something basic is broken in your case

robimarko, 2025-01-08:
Well, that is your problem, you dont even have an APK database.

Something is really broken, what is your builds diffconfig?

grosjo, 2025-01-08:
here:

```
# ./scripts/diffconfig.sh 
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_kuwfi_wb5g08=y
CONFIG_ATH10K_THERMAL=y
# CONFIG_ATH9K_HWRNG is not set
CONFIG_ATH9K_SUPPORT_PCOEM=y
# CONFIG_ATH_USER_REGD is not set
CONFIG_CLEAN_IPKG=y
# CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set
# CONFIG_DROPBEAR_AGENTFORWARD is not set
# CONFIG_DROPBEAR_CHACHA20POLY1305 is not set
# CONFIG_DROPBEAR_CURVE25519 is not set
# CONFIG_DROPBEAR_DBCLIENT is not set
# CONFIG_DROPBEAR_ED25519 is not set
# CONFIG_DROPBEAR_SCP is not set
# CONFIG_DROPBEAR_U2F is not set
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
# CONFIG_KERNEL_DEVPTS_MULTIPLE_INSTANCES is not set
# CONFIG_KERNEL_ELF_CORE is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_KERNEL_IPV6_SEG6_LWTUNNEL is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_KALLSYMS is not set
# CONFIG_KERNEL_KEYS is not set
# CONFIG_KERNEL_MPTCP_IPV6 is not set
# CONFIG_KERNEL_NAMESPACES is not set
# CONFIG_KERNEL_NET_L3_MASTER_DEV is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_NF_CONNTRACK_TIMEOUT is not set
# CONFIG_KERNEL_POSIX_MQUEUE is not set
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_SECCOMP is not set
# CONFIG_KERNEL_SWAP is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
CONFIG_LUCI_SRCDIET=y
# CONFIG_PACKAGE_MAC80211_MESH is not set
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_getrandom is not set
CONFIG_PACKAGE_kmod-ath9k=m
CONFIG_PACKAGE_kmod-ath9k-common=m
CONFIG_PACKAGE_kmod-br-netfilter=y
# CONFIG_PACKAGE_kmod-gpio-button-hotplug is not set
CONFIG_PACKAGE_kmod-hwmon-core=y
CONFIG_PACKAGE_kmod-ipt-condition=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-iface=y
CONFIG_PACKAGE_kmod-ipt-iprange=y
CONFIG_PACKAGE_kmod-ipt-ipv4options=y
CONFIG_PACKAGE_kmod-ipt-physdev=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-libphy=y
CONFIG_PACKAGE_kmod-loop=y
CONFIG_PACKAGE_kmod-mdio-devres=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nft-bridge=y
CONFIG_PACKAGE_kmod-nft-compat=y
CONFIG_PACKAGE_kmod-nls-base=y
CONFIG_PACKAGE_kmod-nls-utf8=y
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-swconfig=y
CONFIG_PACKAGE_kmod-switch-ar8xxx=y
CONFIG_PACKAGE_kmod-thermal=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-filebrowser=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-json=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-rpc=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_openwrt-keyring is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
# CONFIG_PACKAGE_swconfig is not set
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_urandom-seed is not set
# CONFIG_PACKAGE_urngd is not set
# CONFIG_PACKAGE_usign is not set
CONFIG_PACKAGE_wpad-basic=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_USE_APK is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_MKLIBS=y
# CONFIG_USE_SECCOMP is not set
# CONFIG_WPA_MBO_SUPPORT is not set
CONFIG_PACKAGE_apk-mbedtls=y
CONFIG_PACKAGE_kmod-random-core=y
# CONFIG_PACKAGE_libncurses-dev is not set
# CONFIG_PACKAGE_libpthread is not set
# CONFIG_PACKAGE_opkg is not set
CONFIG_PACKAGE_zlib=y
# CONFIG_PACKAGE_zlib-dev is not set
# CONFIG_ZLIB_OPTIMIZE_SPEED is not set

```

robimarko, 2025-01-08:
And there is your problem:
`# CONFIG_USE_APK is not set`

robimarko, 2025-01-08:
Can you upload the whole .config?

grosjo, 2025-01-08:
[Uploading config_kuwfiWB.txt…]()

grosjo, 2025-01-08:
[config_kuwfiWB.txt](https://github.com/user-attachments/files/18348552/config_kuwfiWB.txt)

grosjo, 2025-01-08:
Why apk files of packages are not included in the compiled firmware, instead of having to upload them on teh device (if ever APK would work) ?

grosjo, 2025-01-08:
In the new compilation, I put M in front of ath9k, but no ath9k-xxx.apk found in the folder....

it seems this needs a lot of debug

grosjo, 2025-01-08:
On another device, I get the following :

WHY does this is looking at some external website ? WHy not just loading the apk files ? why should one go to the complicated steps of uploading the desired apk files to the device, whereas compiled as M in the built ?

```
root@OpenWrt:/tmp# apk add -v kmod-ath9k-6.6.69.6.12.6-r1.apk 
fetch https://downloads.openwrt.org/snapshots/targets/ath79/generic/packages/packages.adb
Failed to send request: Operation not permitted
WARNING: updating and opening https://downloads.openwrt.org/snapshots/targets/ath79/generic/packages/packages.adb: network error (check Internet connection and firewall)
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/base/packages.adb
Failed to send request: Operation not permitted
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/mips_24kc/base/packages.adb: network error (check Internet connection and firewall)
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/packages.adb
Failed to send request: Operation not permitted
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/packages.adb: network error (check Internet connection and firewall)
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/packages.adb
Failed to send request: Operation not permitted
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/packages.adb: network error (check Internet connection and firewall)
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/packages.adb
Failed to send request: Operation not permitted
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/packages.adb: network error (check Internet connection and firewall)
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony/packages.adb
Failed to send request: Operation not permitted
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony/packages.adb: network error (check Internet connection and firewall)
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/video/packages.adb
Failed to send request: Operation not permitted
WARNING: updating and opening https://downloads.openwrt.org/snapshots/packages/mips_24kc/video/packages.adb: network error (check Internet connection and firewall)
(1/1) Installing kmod-ath9k (6.6.69.6.12.6-r1)
Executing kmod-ath9k-6.6.69.6.12.6-r1.post-install

```

KanjiMonster, 2025-01-08:
> Why apk files of packages are not included in the compiled firmware, instead of having to upload them on teh device (if ever APK would work) ?

`M` in OpenWrt's config means "create a package, but don't include in the image". This also applies to kernel modules.

If you want to have packages included in the image, you need to select them as y/`*`.

grosjo, 2025-01-09:
1 -No, the 'y' compile in placed in the executed firmware, not just the storage space. Purpose of modules is to load/unload them on request, is not to make it impossible to use them
2 - this is total non-sense to look for external website when using APK. For those will to use that "feature", maybe add a specific command line option to specify " search on the web"

robimarko, 2025-01-09:
That is just how APK works, it updates the remote feeds to populate the DB and then proceeds to install whatever you told it to

KanjiMonster, 2025-01-09:
> 1 -No, the 'y' compile in placed in the executed firmware, not just the storage space. Purpose of modules is to load/unload them on request, is not to make it impossible to use them

Sorry, I cannot follow what you are saying here. In OpenWrt's config, you only configure packages, but you do not control whether kernel drivers are built-in or a module. If it is a `kmod-*` package, it will always be a module (.ko), regardless of selecting `y` or `m`. The `y` or `m` only controls whether the resulting module will be included in the rootfs of the firmware.

> That is just how APK works, it updates the remote feeds to populate the DB and then proceeds to install whatever you told it to

I agree though that it shouldn't do that when trying to install a local package. I asked it to install a package, not to update the feeds.

robimarko, 2025-01-09:
> 
> I agree though that it shouldn't do that when trying to install a local package. I asked it to install a package, not to update the feeds.

I agree its weird

grosjo, 2025-01-10:
If "M" is not to allow the packages to be loaded at a later time, then what 'M" does even exist ?

Agree the curent behavior of APK is just non-sense

ynezz, 2025-01-10:
You've two options

```shell
podman run --network none --rm -it alpine:edge  \
    sh -c "touch foo.apk && apk add --no-network --allow-untrusted foo.apk"
WARNING: opening from cache https://dl-cdn.alpinelinux.org/alpine/edge/main: No such file or directory
WARNING: opening from cache https://dl-cdn.alpinelinux.org/alpine/edge/community: No such file or directory
ERROR: foo.apk: BAD archive
```

```shell
podman run --network none --rm -it alpine:edge \
    sh -c "touch foo.apk && apk add --repositories-file /dev/null --allow-untrusted foo.apk"
ERROR: foo.apk: BAD archive
```

closed 2025-01-10 by ynezz

ynezz, 2025-01-10:
Documented it https://openwrt.org/docs/guide-user/additional-software/opkg-to-apk-cheatsheet#add_and_remove under ` Default repositories handling`

ynezz, 2025-01-10:
> I agree though that it shouldn't do that when trying to install a local package. I asked it to install a package, not to update the feeds.

I agree as well, just don't have enough time to create a merge request and propose improvement upstream :)