openwrt/openwrt #17491: "Set Static" Button for DHCPv6 Lease in Overview does not work

MichaelUray, 2025-01-05:
### Describe the bug

Click in `cgi-bin/luci/admin/status/overview` - `Active DHCPv6 Leases` on the `Set Static` button
The button starts turning a circle but it never finishs, IPv4 works fine.
[image]

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

x86/64

### Device

LXC in Proxmox

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-01-05:
Go to LuCI issues page: https://github.com/openwrt/luci/issues

MichaelUray, 2025-01-05:
@jow- could you give users the permission to move their own issues to another repository?