openwrt/openwrt #17492: LUCI: Browser looses selection/focus every few seconds

MichaelUray, 2025-01-05:
### Describe the bug

If you select a text in the webbrowser LUCI interface, then is looses its selection within a couple seconds at the next data refresh.
On this way it is pretty hard to select and copy a text.
https://github.com/user-attachments/assets/ebfb178f-ee39-4b39-9fca-e024b94a2aab

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

x86/64

### Device

LXC in Proxmox

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-01-05:
Go to LUCI issues page: https://github.com/openwrt/luci/issues

tapper82, 2025-01-05:
Same here. Makes it hard to copy an IP address when using a screen reader. Happens to me when in the DHCP table.