openwrt/openwrt #17497: hostapd memleak

HaiPengSUPER, 2025-01-06:
### Describe the bug

https://github.com/openwrt/openwrt/blob/main/package/network/services/hostapd/patches/320-optional_rfkill.patch

In 320-optional_rfkill.patch, the following changes have been made:
+static inline struct rfkill_data * rfkill_init(struct rfkill_config *cfg)
+{

return (void *) 1;
+}
Returning 1 will result in memory not being freed. For example, in the wpa_driver_nl80211_drv_init_rfkill within the file driver_nl80211.c.

A memory leak occurred in the current actual environment, and it was eventually traced back to this location. The return value of 1 has no special purpose, so why return 1?

### OpenWrt version

DISTRIB_REVISION='r19685-512e76967f'

### OpenWrt release

DISTRIB_RELEASE='3.1.0'

### OpenWrt target/subtarget

DISTRIB_TARGET='tr6560/generic'

### Device

THG6500-TAX2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2025-01-06:
@cotequeiroz @nbd168 @ynezz  ping

ynezz, 2025-01-10:
[image]

> DISTRIB_TARGET='tr6560/generic'
> DISTRIB_RELEASE='3.1.0'

This is not OpenWrt.

> DISTRIB_REVISION='r19685-512e76967f'

This is 22.03 EOL (unsupported) release.

> A memory leak occurred in the current actual environment, and it was eventually traced back to this location.

Ok, lets try to reproduce with current OpenWrt snapshot, 23.05 or 24.10 releases on some supported target and provide a complete output from the tool about the memory leak (valgrind?), ideally a reproduction scenario.

closed 2025-01-10 by ynezz