openwrt/openwrt #17500: lantiq: xrx200_legacy: apparently slow flash I/O results in poor response or apparent hang, sometimes rebooting

aimacintyre, 2025-01-06:
### Describe the bug

The LUCI web interface appears to be very sluggish and can become completely unresponsive; sometimes the modem becomes unresponsive after boot without successfully syncing (VDSL2).  In the completely unresponsive state the DSL connection quite often drops and sometimes the device reboots.

NB: I've been able to get the device into fail-safe mode reliably if it appears to go unresponsive during boot, and can successfully flash a known good older image to get the device usable again from fail-safe mode.

### OpenWrt version

r28306-50dcebeb8a

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

lantiq/xrx200_legacy

### Device

TP-LINK TD-W8980

### Image kind

Self-built image

### Steps to reproduce

Install any SNAPSHOT (main) or 24.10 release image and reboot the device, then
- attempt to login to LUCI and configure the device or just navigate the LUCI interface
- attempt to login via ssh and start `top` and monitor

### Actual behaviour

- when you can log in to LUCI and the sluggish behaviour is evident, the 1 minute load factor is high (at least 5-6) and continues to rise.  Giving the device a "breather" (no UI activity) for several minutes sometimes allows it to recover, other times the load factor keeps rising until LUCI becomes completely unresponsive. In the completely unresponsive state the DSL connection quite often drops and sometimes the device reboots.
- logging in via ssh and running `top` continuously, the most obvious marker of the behaviour when apparent is an io% value higher than 20% - values as high as 50% have been observed before top itself becomes unresponsive.
- if the device boots, establishes a DSL connection and passes traffic, it will usually continue apparently normal operation until some periodic process initiates a burst of process activity that pushes it into the unresponsive state (I've seen as long as 30 hours).

### Expected behaviour

device operates normally, load factor stays below 1-2 during normal operation and DSL connections are held for weeks at a time (my DSL connection isn't fast but is usually very stable with 10 week uptimes common, though sometimes with SRA events resulting in sync rate adjustments without a fully re-sync).

### Additional info

The 2.4GHz WiFi is disabled (5GHz WiFi not supported on TD-W8980) at all times.

A DM200 running an image built from the same sources (same `make` invocation from diffconfig below) doesn't experience the completely unresponsive state but elevated io% can be seen via `top` and LUCI can seem a bit sluggish at times; it just never seems to get out of control.

Builds of 22.03 don't show any of the behaviour; builds of 23.05 show hints of the elevated io% in `top` but the behaviour doesn't appear to get out of control.

I have experimented with kernels compiled for speed (.config modified after creation from diffconfig below) and increasing the fragment cache to 3 (from the default 2), but either or both changes appear to at best only slightly improve the behaviour.

NB:  I require a non-standard DSL chipset firmware for my connection and there is not quite sufficient free space to install it after installing the standard image, hence the removed packages in the diffconfig below; the required file is included in my images via a `files` subdirectory in my checkouts.

### Diffconfig

```text
CONFIG_TARGET_lantiq=y
CONFIG_TARGET_lantiq_xrx200_legacy=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_lantiq_xrx200_legacy_DEVICE_netgear_dm200=y
CONFIG_TARGET_DEVICE_PACKAGES_lantiq_xrx200_legacy_DEVICE_netgear_dm200="-ltq-vdsl-vr9-vectoring-fw-installer -dsl-vrx200-firmware-xdsl-a -dsl-vrx200-firmware-xdsl-b-patch"
CONFIG_TARGET_DEVICE_lantiq_xrx200_legacy_DEVICE_tplink_tdw8970=y
CONFIG_TARGET_DEVICE_PACKAGES_lantiq_xrx200_legacy_DEVICE_tplink_tdw8970="-ltq-vdsl-vr9-vectoring-fw-installer -dsl-vrx200-firmware-xdsl-a -dsl-vrx200-firmware-xdsl-b-patch"
CONFIG_TARGET_DEVICE_lantiq_xrx200_legacy_DEVICE_tplink_tdw8980=y
CONFIG_TARGET_DEVICE_PACKAGES_lantiq_xrx200_legacy_DEVICE_tplink_tdw8980="-ltq-vdsl-vr9-vectoring-fw-installer -dsl-vrx200-firmware-xdsl-a -dsl-vrx200-firmware-xdsl-b-patch"
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
CONFIG_PACKAGE_dsl-vrx200-firmware-xdsl-a=m
CONFIG_PACKAGE_dsl-vrx200-firmware-xdsl-b-patch=m
CONFIG_PACKAGE_ltq-vdsl-vr9-vectoring-fw-installer=m
CONFIG_PACKAGE_luci-ssl=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-01-07:
> Invalid Target/Subtarget reported. `lantiq/xrx200_legacy` Is this from a supported device?

Yes

Rupurudu, 2025-01-07:
I believe this is partly because of the SquashFS blocksize, the IO bandwith might not be able to keep up reading 1M chunksizes.

https://github.com/openwrt/openwrt/blob/285bdd29cb55900a2009e26303ab5abb66e165e9/config/Config-images.in#L157