openwrt/openwrt #17505: Cudy TR3000: MT7981 ethernet disconnects

divinehawk, 2025-01-07:
### Describe the bug

eth1 (br-lan) loses connectivity. eth0 (wan) seems to recover.   Only a reboot seems to fix.

[1641206.401457] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[1641206.408615] WARNING: CPU: 0 PID: 0 at 0xffffffc0086c6464
[1641206.414095] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_policy xt_multiport xt_mark xt_mac xt_limit xt_esp xt_comment xt_TCPMSS xt_LOG xfrm_interface slhc poly1305_neon nfnetlink nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha iptable_mangle iptable_filter ipt_ah ip_tables compat crypto_safexcel ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6
[1641206.414290]  tunnel4 tun xfrm_user xfrm_ipcomp af_key xfrm_algo sha1_generic seqiv md5 echainiv des_generic libdes cbc authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common aquantia crc_ccitt
[1641206.523464] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.15.167 #0
[1641206.529719] Hardware name: Cudy TR3000 v1 (DT)
[1641206.534322] pstate: 20400005 (nzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[1641206.541446] pc : 0xffffffc0086c6464
[1641206.545096] lr : 0xffffffc0086c6464
[1641206.548745] sp : ffffffc008003db0
[1641206.552220] x29: ffffffc008003db0 x28: 0000000000000140 x27: 00000000ffffffff
[1641206.559518] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff80003434c0
[1641206.566817] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008a46000
[1641206.574114] x20: ffffff8000343000 x19: 0000000000000000 x18: ffffffc008a5a2a8
[1641206.581413] x17: ffffffc017482000 x16: ffffffc008000000 x15: 0000000000000345
[1641206.588711] x14: 0000000000000117 x13: ffffffc008003ae8 x12: ffffffc008ab22a8
[1641206.596010] x11: 712074696d736e61 x10: ffffffc008ab22a8 x9 : 0000000000000000
[1641206.603308] x8 : ffffffc008a5a258 x7 : ffffffc008a5a2a8 x6 : 0000000000000001
[1641206.610605] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[1641206.617903] x2 : 0000000000000040 x1 : 0000000000000002 x0 : 000000000000003f
[1641206.625202] Call trace:
[1641206.627810]  0xffffffc0086c6464
[1641206.631112]  0xffffffc0080a5b50
[1641206.634415]  0xffffffc0080a63c8
[1641206.637717]  0xffffffc0080a6918
[1641206.641020]  0xffffffc00803b04c
[1641206.644322]  0xffffffc00803b404
[1641206.647624]  0xffffffc00803b5d0
[1641206.650927]  0xffffffc008089454
[1641206.654230]  0xffffffc0080100e0
[1641206.657532]  0xffffffc00801584c
[1641206.660834]  0xffffffc00801609c
[1641206.664136]  0xffffffc008835d60
[1641206.667438]  0xffffffc008836514
[1641206.670740]  0xffffffc008011374
[1641206.674043]  0xffffffc008836af4
[1641206.677345]  0xffffffc008069350
[1641206.680648]  0xffffffc008069560
[1641206.683950]  0xffffffc008836e94
[1641206.687253]  0xffffffc0089d0954
[1641206.690555]  0xffffffc0089d0f2c
[1641206.693857]  0xffffffc0089d034c
[1641206.697160] ---[ end trace 7adc065e7be54b8b ]---
[1641206.701962] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[1641206.840530] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[1641206.956621] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[1641207.497411] mtk_soc_eth 15100000.ethernet: warm reset failed
[1641207.796327] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)
[1641207.811486] mtk_soc_eth 15100000.ethernet eth0: configuring for phy/sgmii link mode
[1641207.871506] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[1641207.881253] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/gmii link mode
[1641207.892613] br-lan: port 1(eth1) entered disabled state
[1641210.956019] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[1641211.045751] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[1641211.045775] br-lan: port 1(eth1) entered blocking state
[1641211.059599] br-lan: port 1(eth1) entered forwarding state

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy TR3000 v1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

divinehawk, 2025-01-07:
I will try and report back:
ethtool -K eth0 tso off
ethtool -K eth1 tso off

xianren78, 2025-01-07:
Fan-cool your device first to see if it disappears.