openwrt/openwrt #17506: The system log keeps repeating the cron log.

zdz2019, 2025-01-07:
### Describe the bug

The system log keeps repeating the cron log, this causes other logs to be overwritten.

[image]

### OpenWrt version

24.10-RC4

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

x86/64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-01-07:
In version 24.10 you need to change cron log level to disabled

/etc/config/system:
```
config system
->        option cronloglevel '9'
```

zdz2019, 2025-01-07:
> In version 24.10 you need to change cron log level to disabled
> 
> /etc/config/system:
> 
> ```
> config system
> ->        option cronloglevel '9'
> ```
Yes, I've already tried disabling logging, but it didn't work.

config system
        ...
	option conloglevel '4'
	option cronloglevel '9'