openwrt/openwrt #17510: Unable to Use Ed25519 Private Key for SSH Connection

ZLHuo, 2025-01-07:
### Describe the bug

I was able to use my Ed25519 private key in 23.05.5. However, after updating to 24.10-rc4, my device consistently rejects my private key. I generated a new RSA-2048 key pair, and it works fine.

Is this an issue with Dropbear? Any help would be greatly appreciated.

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Xiaomi Mi Router 4C

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ZLHuo, 2025-01-07:
Replaced dropear with openssh-server, and the key is working.

aimacintyre, 2025-01-07:
`mt76x8` subtarget is now SMALL_FLASH which results in ED25519 support not being included in `dropbear` with a default config.

closed 2025-01-07 by ZLHuo