openwrt/openwrt #17511: `dnsmasq` does not use values from `addnhosts`

bilogic, 2025-01-07:
### Describe the bug

I can't spoof an IP for a hostname provided via addnhosts

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA8100

### Image kind

Official downloaded image

### Steps to reproduce

```
clear
ping -w 1 github.com # see the real IP

uci show | grep addnhosts
echo "0.0.0.0 github.com" > /root/blocked/distraction.hosts

service dnsmasq reload
sleep 10
ping -w 1 github.com # still sees the real IP

service dnsmasq restart
sleep 10
ping -w 1 github.com # still sees the real IP
```

### Actual behaviour

Real IP shows even though host file was provided via addnhost

### Expected behaviour

Fake IP (0.0.0.0) should show instead

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-01-10:
Do you get the same results from a client as from the router? Is the router `/etc/resolv.conf` actually pointing to 127.0.0.1?

bilogic, 2025-01-10:
I had never touch `/etc/resolv.conf`, is it necessary?

```
root@OpenWrt:~# cat /etc/resolv.conf
search lan
nameserver 127.0.0.1
nameserver ::1
```

dave14305, 2025-01-10:
Do you see dnsmasq log entries after a restart showing it read the file?
```
service dnsmasq restart
grep addn-hosts /var/etc/dnsmasq.conf*
logread -e distraction
```

bilogic, 2025-01-10:
Thanks, I think we are making progress, I clearly have the file specified in the UI, but the logs seem to show otherwise.

[image]

```
root@Hall:~# service dnsmasq restart
udhcpc: started, v1.36.1
udhcpc: broadcasting discover
udhcpc: no lease, failing
udhcpc: started, v1.36.1
udhcpc: broadcasting discover
udhcpc: no lease, failing
udhcpc: started, v1.36.1
udhcpc: broadcasting discover
udhcpc: no lease, failing
udhcpc: started, v1.36.1
udhcpc: broadcasting discover
udhcpc: no lease, failing
root@Hall:~# grep addn-hosts /var/etc/dnsmasq.conf*
/var/etc/dnsmasq.conf.cfg01411c:addn-hosts=/tmp/hosts
/var/etc/dnsmasq.conf.cfg01411c.24946:addn-hosts=/tmp/hosts
root@Hall:~# logread -e distraction
root@Hall:~#
```

bilogic, 2025-01-10:
ok, I changed the value to `/root/blocked` and now i'm getting 0.0.0.0 for the specified hosts.

Apparently `Additional hosts files` is actually asking for a folder (not file), this is highly misleading.

KanjiMonster, 2025-01-10:
The documentation for `--addn-hosts` says it supports both files and folders:

```
-H, --addn-hosts=<file>
    Additional hosts file. Read the specified file as well as /etc/hosts. If --no-hosts is given, read only the specified file. This option may be repeated for more than one additional hosts file. If a directory is given, then read all the files contained in that directory in alphabetical order. 
```

So files *should* work. The question is why they don't.

bilogic, 2025-01-11:
1. `uci` had the file
2. but `/var/etc/dnsmasq.conf*` did not

Culprit seems to be on the `uci` side of things when translating to dnsmasq config

dave14305, 2025-01-11:
How many dnsmasq instances in your dhcp config file? Care to share the content? Feels like something might be corrupt in there.

bilogic, 2025-01-11:
Any commands I can run to check?

dave14305, 2025-01-11:
Just run `uci export dhcp`.

bilogic, 2025-01-11:
It is a very long list, I'm not sure how many instances of dnsmasq. 

Reason is, I have additional subnets, e.g. IoT, Guest etc

dave14305, 2025-01-11:
Post each `config dnsmasq` section from the output. Is there more than one?

bilogic, 2025-01-11:
Only 1
```
root@Hall:~# uci export dhcp | grep dnsmasq
config dnsmasq
        option confdir '/tmp/dnsmasq.d'
root@Hall:~#
```

dave14305, 2025-01-11:
Run the same but grep for addnhosts. It should be a list type, but perhaps it’s written as an option type?

bilogic, 2025-01-11:
```
root@Hall:~# uci export dhcp | grep addnhosts
        option addnhosts '/root/blocked/distraction.hosts'
root@Hall:~#
```

dave14305, 2025-01-11:
Edit `/etc/config/dhcp` and change that `option` keyword to `list` and see it it works after a dnsmasq restart.

bilogic, 2025-01-11:
Yes, it works now and I think I found the root cause. 

My OpenWrt config is in the form of a bash script and it added `option addnhosts ...`, thanks so much for the help!

bilogic, 2025-01-11:
1. My OpenWrt is configured via a bash script, but that bash script does not contain `addnhosts` yet
2. I used LuCI to remove and add `addnhosts` again, this time it was `config list`
3. I can't track down why it was `config option`

But really appreciate the help to find the cause of my problem. Thank you!

bilogic, 2025-01-11:
```
root@Hall:~# uci show dhcp | grep @dnsmasq
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].cachesize='1000'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].confdir='/tmp/dnsmasq.d'
dhcp.@dnsmasq[0].addnhosts='/root/blocked/distraction.hosts'
```

1. I tracked down the issue, when I made a change in the UI, the only line that was added is `dhcp.@dnsmasq[0].addnhosts='/root/blocked/distraction.hosts'`
2. If I were to then run `uci commit dhcp` and `service dnsmasq restart`, that leaeds to `option addnhosts ...`
3. Any idea why?

bilogic, 2025-01-11:
https://serverfault.com/a/1117696/930498 

The solution is to use `add_list`, e.g. `uci add_list dhcp.@dnsmasq[0].address='/example.local/192.168.123.123'`

dave14305, 2025-01-11:
> 3\. Any idea why?

If you add more than one entry via LuCI, does it write it out as list? I’m guessing it treats a single value as option and multiple values as list. Try splitting your hosts file and see if LuCI behaves better.

bilogic, 2025-01-11:
I tested, it seems `add_list` and `set` produces the same `dhcp.@dnsmasq[0].addnhosts='/root/blocked/distraction.hosts'`, but generates different output when `uci commit dhcp`.

closed 2025-01-12 by bilogic