openwrt/openwrt #17522: NanoPi R5C wireless device not found

Sceptically, 2025-01-08:
### Describe the bug

With a fresh install of the 24.10-rc5 image, no wireless device is showing up. In FriendlyWrt the rtl8822ce m.2 wireless card was found and working.

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R5C

### Image kind

Official downloaded image

### Steps to reproduce

Booting a stock 24.10.0-rc5 squashfs image.

### Actual behaviour

The wireless device is not found (iwinfo returns an empty output and, after installing pciutils, lspci does not list the wireless device).

### Expected behaviour

The wireless should just work (tm).

### Additional info

When running [lspci -v](https://github.com/user-attachments/files/18341406/openwrt-24.10.0-rc5-nowifi-lspci-v.txt) the following error shows up after the initial 0000:00:00.0 pci bridge device line:

lspci: Unable to load libkmod resources: error -2

Not much in [dmesg](https://github.com/user-attachments/files/18341385/openwrt-24.10.0-rc5-nowifi-dmesg.txt) looks suspicious to me other than the lines like
`[    2.082026] pci_bus 0002:01: busn_res: can not insert [bus 01-ff] under [bus 00-0f]`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Sceptically, 2025-01-08:
Here's a [dmesg from a working friendlywrt install](https://github.com/user-attachments/files/18343945/friendlywrt-dmesg.txt), and [the output of lspci -v on friendlywrt](https://github.com/user-attachments/files/18343947/friendlywrt-lspci-v.txt).

ataluts, 2025-01-09:
I suspect that we might have the same issue with PCIe but with different devices affected.
Here is my opened issue with the same board #17452.

bam80, 2026-11-10:
Any progress?

Sceptically, 2026-11-11:
> Any progress?

The network device is actually showing up in lspci on release 25.12.6, so it looks like it's fixed.

closed 2026-11-11 by Sceptically