openwrt/openwrt #17524: [23.05.5] Archer AX23: unable to peristently set device autoneg/speed/duplex in /etc/config/network

x-qq, 2025-01-08:
### Describe the bug

I need to set wan device autoneg/speed/duplex to no/100/full persistenly so that these settings are applied at boot and maintaned consistently, but I am unable to do so.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer AX23 v1

### Image kind

Official downloaded image

### Steps to reproduce

Based on information in https://github.com/openwrt/netifd/blob/openwrt-23.05/device.c#L67 I have modified /etc/config/network to look like this:

```
root@OpenWrt:~# cat /etc/config/network 

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd71:128e:1c4e::/48'
	option packet_steering '1'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.3.99'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'wan'
	option proto 'dhcp'
	option hostname '*'

config device
	option name 'wan'
	option ipv6 '0'
	option autoneg 0
	option speed 100
	option duplex 1
root@OpenWrt:~#
```

After which I reboot the device.

### Actual behaviour

After reboot, the wan interface has autoneg on and speed 1000 - the settings in config file were ignored.

```
root@OpenWrt:~# ethtool wan
Settings for wan:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Full 
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 4
	Transceiver: external
	Auto-negotiation: on
	Current message level: 0x000000ff (255)
			       drv probe link timer ifdown ifup rx_err tx_err
	Link detected: yes
root@OpenWrt:~#
```

### Expected behaviour

The settings in the config file should be applied and maintained consistently.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-01-09:
Do you get any errors in the system log when you boot with this configuration?

Can you successfully set the WAN state you desire using `ethtool`?

x-qq, 2025-01-09:
I don't see any errors in the system log after boot.

[system-log-after-boot.txt](https://github.com/user-attachments/files/18364717/system-log-after-boot.txt)

After reboot I can ssh into the router and fix the interface manually with:

```
ethtool -s wan speed 100 duplex full autoneg off
```

after which I restart the interface from LuCI and it is able to communicate with the ISP.

x-qq, 2025-04-17:
Is any additional info required to get this fixed?