openwrt/openwrt #17530: Regression in 24.10 on MediaTek MT7915E (Netgear Wax 206): Sta WDS Client does not use HE160 anymore

marcust, 2025-01-08:
### Describe the bug

I have been playing with HE160 and meshing a lot (https://github.com/openwrt/openwrt/issues/13880), after I gave up on meshing I realized that I could not get back to my former performance anymore because the WAX206 was only doing 80 Mhz width now even as WDS client.
[image]
`iw dev`:
```Interface wl1-sta0
		ifindex 11
		wdev 0x100000002
		addr 02:0c:43:26:46:b0
		ssid ThiesenwaveBER_5
		type managed
		channel 48 (5240 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
		4addr: on
```

This is a regression in `24.10.0-rc5`, as I get it back when I downgrade to `23.05.5` in the same configuration:

[image]

```
	Interface wl1-sta0
		ifindex 11
		wdev 0x100000002
		addr 00:0c:43:26:46:b0
		ssid ThiesenwaveBER_5
		type managed
		channel 48 (5240 MHz), width: 160 MHz, center1: 5250 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
		4addr: on
```

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Netgear WAX206

### Image kind

Official downloaded image

### Steps to reproduce

- Use a WAX206 in WDS client mode:
```
config wifi-iface 'wifinet3'
	option device 'radio1'
	option mode 'sta'
	option network 'lan'
	option encryption 'psk2'
	option key 'redacted'
	option wds '1'
	option disassoc_low_ack '0'
	option bssid 'redacted'
	option ssid 'ThiesenwaveBER_5'
```
- Upgrade to 24.10(-rc5)
- See lower bandwith on the connection

### Actual behaviour

Link is established with 80Mhz channel width

### Expected behaviour

Link is established with 160Mhz channel width

### Additional info

I also have a D-Link AQUILA PRO AI M30 A1(MediaTek MT7981 802.11ac/ax/n) that does not show this behaviour, link is fine with 24.10 in 160Mhz width. Master router is a MERCUSYS MR90X v1 (MediaTek MT7986 802.11ac/ax/n).

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Underworld6, 2025-02-03:
I am experiencing the same issue with my Belkin RT3200 using the same chipset as your router with rc7. On top that it connects on 80Mhz the speed is very slow, as in your example. However if I switch off the AP mode of it and leave just the WDS client mode- the speed is fine - I can not remember if in that case it was connecting @ 160 Mhz.

wirl, 2025-02-06:
This could be related to [my issue](https://github.com/openwrt/openwrt/issues/17770) too. Let's hope we can get this bug identified and squashed!

gentrydeng, 2025-02-16:
Hi guys, my question may be relevant, so I'm commenting here.

I've been working on the GL.iNet GL-MT6000, the CMCC RAX3000M and the Qihoo 360T7 with OpenWrt 24.10.0 installed and intend to link via WDS.

However, I have noticed that the downstream CMCC RAX3000M and Qihoo 360T7 networks do not work at all.
I referenced this article: https://openwrt.org/docs/guide-user/network/wifi/wifiextenders/wds

Since this is my production network and simple troubleshooting couldn't find the problem, I quickly used [Relayd](https://openwrt.org/docs/guide-user/network/wifi/relay_configuration) instead. And Relayd worked fine.

I apologise for not being able to provide more data, but I sincerely hope that the problem can be resolved. WDS is much simpler and more efficient than Relayd.

wirl, 2025-02-17:
All we can do is hope someone from OpenWrt has a look at these issues at some point.

oxavelar, 2025-10-25:
Anyone has a further clue about this?