openwrt/openwrt #17547: SSH via wifi slows down

Yurannich2, 2025-01-09:
### Describe the bug

SSH via wifi slows down

### OpenWrt version

r28515-70610a5240

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Steps to reproduce

Connect with SSH to router via Wi-Fi 

### Actual behaviour

The command line either does not respond to commands at all or is very slow when entering

### Expected behaviour

Normal operation without freezing in СLI

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

fs30000, 2025-05-08:
I'm facing the same issue, ssh sometimes gets slow, not necessarily to openwrt itself. Version OpenWrt 24.10.1 r28597-0425664679 / LuCI openwrt-24.10 branch 25.123.45881~e061a5f on a banana pi r4

Denis11212, 2025-05-20:
I confirm the problem. Also a BPI-R4 device. This problem has been around since 2024 in the Snapshot version.

Yurannich2, 2025-05-20:
Clarification: connection slowdown, inability to connect and other problems occur exclusively when connecting via Wi-Fi 6; when connecting in AX and H mode, this does not occur.

mikestarman, 2025-05-20:
Same, happens whenever I add a 5ghz/6ghz network in AX or BE mode

fs30000, 2025-05-23:
I have  swapped dropbear with openssh-server, to see if it fixed something, even tough ssh gets slow to other hosts other than openwrt itself, no change.

fs30000, 2025-06-26:
> I have the same problem. Very slow ssh only via wifi. 

Exactly, ssh via cable works fine.

fs30000, 2025-06-27:
With 24.10.2, this got much worse. To all machines, not just openwrt itself.

hsand, 2025-09-26:
Appears to be fixed in 24.10.3