openwrt/openwrt #17548: dnsmasq-full crush on parsing nftset direcive, network not working on reboot

cspwizard, 2025-01-09:
### Describe the bug

dnsmasq (full version) crush on parsing nftset config line. No interfaces are working after that because dhcp is not working.

config was generated bu the Luci. Couldn't figure out what is wrong on my own. Except the case that it's about domain of third level. When I add amazon.com - it works fine, when i add aws.amazon.com - it failes to start.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Steps to reproduce

Add ipset via Luci UI for me it reproduce with 3rd level domain (aws.amazon.com, api.google.com, etc.) Dnsmasq fail to start after that. With error "bad option line 44" - the next line (empty one) after nftset rule.
With domains of higher levels it works (tested with speedtest.tngnet.com.prod.hosts.ooklaserver.net). 

### Actual behaviour

Dnsmasq fail to start with error "bad option line 44" - the next line (empty one) after nftset rule. In case if it will be rebooted it would stop working completely as no client could be connected to the network.

### Expected behaviour

Everything should work fine, or at least the nftset should be rejected, so the network could operate fine.

### Additional info

```
daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
daemon.info dnsmasq[1]: DNSSEC validation enabled
daemon.info dnsmasq[1]: configured with trust anchor for <root> keytag 20326
daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
daemon.info dnsmasq[1]: using nameserver 127.0.0.53#53
daemon.info dnsmasq[1]: using only locally-known addresses for test
daemon.info dnsmasq[1]: using only locally-known addresses for onion
daemon.info dnsmasq[1]: using only locally-known addresses for localhost
daemon.info dnsmasq[1]: using only locally-known addresses for local
daemon.info dnsmasq[1]: using only locally-known addresses for invalid
daemon.info dnsmasq[1]: using only locally-known addresses for bind
daemon.info dnsmasq[1]: using only locally-known addresses for lan
daemon.info dnsmasq[1]: read /etc/hosts - 12 names
daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
daemon.crit dnsmasq[1]: bad option at line 44 of /var/etc/dnsmasq.conf.cfg01411c
daemon.crit dnsmasq[1]: FAILED to start up
daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 12 crashes, 0 seconds since last crash
```
example of the nftset config. (only domains changed)
```
nftset=/amazon.com/4#inet#fw4#vpn_dst
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-01-10:
What is the actual list of domains that fail? There is a limit to the number of characters dnsmasq can parse from one option (I don’t know what it is), but if your domain list is really long, it might need to be broken up into multiple nftset options.

cspwizard, 2025-01-10:
Good point, thanks! 
Looks like this is the issue. Right now its 1025 chars. If I try to add any new domain it fails, but adding new list works fine. 
So this seems to be Luci issue. Not the dnsmasq itself.

rmandrad, 2025-01-10:
> Good point, thanks! Looks like this is the issue. Right now it is 1025 chars. If I try to add any new domain it fails, but adding new list works fine. So this seems to be Luci issue. Not the dnsmasq itself.

if so the issue should be created on openwrt/luci instead of openwrt/openwrt

closed 2025-01-10 by cspwizard

dave14305, 2025-01-10:
My opinion is that this should be solved in the dnsmasq init script, if and only if the user is using the newer ipset config sections versus the older single line ipset option. Somehow (easier said than done), the init script should keep track of the total length of the domains and nftset family/type/table/name and “break” at 1025 chars and start a new nftset= line in dnsmasq.conf for the remaining domains.

cspwizard, 2025-01-10:
Yeah, probably the main issue is that network doesn't allow new connections with dhcp. So it would be nice if dnsmasq at least would log warning and ignore the line, rather then failing completely

Fail-Safe, 2025-07-13:
@cspwizard && @dave14305 -- I see this was closed as "not planned". Where did things land with this? Was there a resolution?

FWIW, I am running into this issue now and it's pretty impactful.

Fail-Safe, 2025-07-13:
@cspwizard I took on @dave14305's challenge:
> Somehow (easier said than done), the init script should keep track of the total length of the domains and nftset family/type/table/name and “break” at 1025 chars and start a new nftset= line in dnsmasq.conf for the remaining domains.

Would you mind giving this patch a try? This is working well for me and fixes the issue of `dnsmasq` not starting when I was previously exceeding the line length limitation.

```patch
diff --git a/package/network/services/dnsmasq/files/dnsmasq.init b/package/network/services/dnsmasq/files/dnsmasq.init
index a0a644fcbd..0997615bfd 100755
--- a/package/network/services/dnsmasq/files/dnsmasq.init
+++ b/package/network/services/dnsmasq/files/dnsmasq.init
@@ -850,7 +850,7 @@ dhcp_relay_add() {

 dnsmasq_ipset_add() {
 	local cfg="$1"
-	local ipsets nftsets domains
+	local ipsets nftsets domains chunk_domains chunk_line

 	add_ipset() {
 		ipsets="${ipsets:+$ipsets,}$1"
@@ -896,8 +896,33 @@ dnsmasq_ipset_add() {
 		return 0
 	fi

+	# Build each --nftset= line as close to 1024 chars as possible
+	nftset_prefix="--nftset=/"
+	maxlen=1024
+	chunk_domains=""
+	# Remove leading slash for iteration
+	domain_iter="${domains#/}"
+	IFS='/'
+	for domain in $domain_iter; do
+		test_chunk="$chunk_domains/$domain"
+		# Calculate the length of the line if we add this domain
+		line_len=$(( ${#test_chunk} + 1 + ${#nftsets} + ${#nftset_prefix} ))
+		if [ $line_len -gt $maxlen ] && [ -n "$chunk_domains" ]; then
+			chunk_line="$chunk_domains/$nftsets"
+			xappend "--nftset=$chunk_line"
+			chunk_domains="/$domain"
+		else
+			chunk_domains="$test_chunk"
+		fi
+	done
+	unset IFS
+	# Output any remaining domains
+	if [ -n "$chunk_domains" ]; then
+		chunk_line="$chunk_domains/$nftsets"
+		xappend "--nftset=$chunk_line"
+	fi
+
 	xappend "--ipset=$domains/$ipsets"
-	xappend "--nftset=$domains/$nftsets"
 }

 dnsmasq_start()
```

Example of prior output to `/var/etc/dnsmasq.conf.cfg01411c`:
```config
...
nftset=/assets.mediadelivery.net/cdn.masterbooks.com/filters.1blocker.com/4#inet#fw4#allowed_cdns_4,6#inet#fw4#allowed_cdns_6
nftset=/my.nextdns.io/dns.nextdns.io/dns1.nextdns.io/dns2.nextdns.io/dns1.steering.nextdns.io/dns2.steering.nextdns.io/anycast.dns.nextdns.io/anycast.dns1.nextdns.io/anycast.dns2.nextdns.io/steering.nextdns.io/cname.vercel-dns.com/ipv4-hydron-clt-1.edge.nextdns.io/ipv6-hydron-clt-1.edge.nextdns.io/ipv4-tier-clt-1.edge.nextdns.io/ipv6-tier-clt-1.edge.nextdns.io/ipv4-anexia-mnz-1.edge.nextdns.io/ipv6-anexia-mnz-1.edge.nextdns.io/ipv4-zepto-xrs-1.edge.nextdns.io/ipv6-zepto-xrs-1.edge.nextdns.io/ipv4-zepto-iad-1.edge.nextdns.io/ipv6-zepto-iad-1.edge.nextdns.io/ipv4-hetzner-iad-1.edge.nextdns.io/ipv6-hetzner-iad-1.edge.nextdns.io/ipv4-teraswitch-pit-1.edge.nextdns.io/ipv4-cloudzy-pit-1.edge.nextdns.io/ipv4-anexia-atl-1.edge.nextdns.io/ipv6-anexia-atl-1.edge.nextdns.io/ipv4-vultr-atl-1.edge.nextdns.io/ipv6-vultr-atl-1.edge.nextdns.io/doh3.steering.nextdns.io/doh3.dns.nextdns.io/doh3.dns1.nextdns.io/doh3.dns2.nextdns.io/probe-test.dns.nextdns.io/ping.nextdns.io/test.nextdns.io/router.nextdns.io/4#inet#fw4#nextdns_hosts_4,6#inet#fw4#nextdns_hosts_6
```

With patch above:
```config
...
nftset=/assets.mediadelivery.net/cdn.masterbooks.com/filters.1blocker.com/4#inet#fw4#allowed_cdns_4,6#inet#fw4#allowed_cdns_6
nftset=/my.nextdns.io/dns.nextdns.io/dns1.nextdns.io/dns2.nextdns.io/dns1.steering.nextdns.io/dns2.steering.nextdns.io/anycast.dns.nextdns.io/anycast.dns1.nextdns.io/anycast.dns2.nextdns.io/steering.nextdns.io/cname.vercel-dns.com/ipv4-hydron-clt-1.edge.nextdns.io/ipv6-hydron-clt-1.edge.nextdns.io/ipv4-tier-clt-1.edge.nextdns.io/ipv6-tier-clt-1.edge.nextdns.io/ipv4-anexia-mnz-1.edge.nextdns.io/ipv6-anexia-mnz-1.edge.nextdns.io/ipv4-zepto-xrs-1.edge.nextdns.io/ipv6-zepto-xrs-1.edge.nextdns.io/ipv4-zepto-iad-1.edge.nextdns.io/ipv6-zepto-iad-1.edge.nextdns.io/ipv4-hetzner-iad-1.edge.nextdns.io/ipv6-hetzner-iad-1.edge.nextdns.io/ipv4-teraswitch-pit-1.edge.nextdns.io/ipv4-cloudzy-pit-1.edge.nextdns.io/ipv4-anexia-atl-1.edge.nextdns.io/ipv6-anexia-atl-1.edge.nextdns.io/ipv4-vultr-atl-1.edge.nextdns.io/ipv6-vultr-atl-1.edge.nextdns.io/doh3.steering.nextdns.io/doh3.dns.nextdns.io/doh3.dns1.nextdns.io/doh3.dns2.nextdns.io/probe-test.dns.nextdns.io/4#inet#fw4#nextdns_hosts_4,6#inet#fw4#nextdns_hosts_6
nftset=/ping.nextdns.io/test.nextdns.io/router.nextdns.io/4#inet#fw4#nextdns_hosts_4,6#inet#fw4#nextdns_hosts_6
```

wuchang1123, 2025-12-27:
> [@cspwizard](https://github.com/cspwizard) I took on [@dave14305](https://github.com/dave14305)'s challenge:
> 
> > Somehow (easier said than done), the init script should keep track of the total length of the domains and nftset family/type/table/name and “break” at 1025 chars and start a new nftset= line in dnsmasq.conf for the remaining domains.
> 
> Would you mind giving this patch a try? This is working well for me and fixes the issue of `dnsmasq` not starting when I was previously exceeding the line length limitation.
> 
> `package/network/services/dnsmasq/patches/900-nftset-chunking.patch`
> 
> diff --git a/package/network/services/dnsmasq/files/dnsmasq.init b/package/network/services/dnsmasq/files/dnsmasq.init
> index a0a644fcbd..0997615bfd 100755
> --- a/package/network/services/dnsmasq/files/dnsmasq.init
> +++ b/package/network/services/dnsmasq/files/dnsmasq.init
> @@ -850,7 +850,7 @@ dhcp_relay_add() {
> 
>  dnsmasq_ipset_add() {
>  	local cfg="$1"
> -	local ipsets nftsets domains
> +	local ipsets nftsets domains chunk_domains chunk_line
> 
>  	add_ipset() {
>  		ipsets="${ipsets:+$ipsets,}$1"
> @@ -896,8 +896,33 @@ dnsmasq_ipset_add() {
>  		return 0
>  	fi
> 
> +	# Build each --nftset= line as close to 1024 chars as possible
> +	nftset_prefix="--nftset=/"
> +	maxlen=1024
> +	chunk_domains=""
> +	# Remove leading slash for iteration
> +	domain_iter="${domains#/}"
> +	IFS='/'
> +	for domain in $domain_iter; do
> +		test_chunk="$chunk_domains/$domain"
> +		# Calculate the length of the line if we add this domain
> +		line_len=$(( ${#test_chunk} + 1 + ${#nftsets} + ${#nftset_prefix} ))
> +		if [ $line_len -gt $maxlen ] && [ -n "$chunk_domains" ]; then
> +			chunk_line="$chunk_domains/$nftsets"
> +			xappend "--nftset=$chunk_line"
> +			chunk_domains="/$domain"
> +		else
> +			chunk_domains="$test_chunk"
> +		fi
> +	done
> +	unset IFS
> +	# Output any remaining domains
> +	if [ -n "$chunk_domains" ]; then
> +		chunk_line="$chunk_domains/$nftsets"
> +		xappend "--nftset=$chunk_line"
> +	fi
> +
>  	xappend "--ipset=$domains/$ipsets"
> -	xappend "--nftset=$domains/$nftsets"
>  }
> 
>  dnsmasq_start()
> Example of prior output to `/var/etc/dnsmasq.conf.cfg01411c`:
> 
> ```
> ...
> nftset=/assets.mediadelivery.net/cdn.masterbooks.com/filters.1blocker.com/4#inet#fw4#allowed_cdns_4,6#inet#fw4#allowed_cdns_6
> nftset=/my.nextdns.io/dns.nextdns.io/dns1.nextdns.io/dns2.nextdns.io/dns1.steering.nextdns.io/dns2.steering.nextdns.io/anycast.dns.nextdns.io/anycast.dns1.nextdns.io/anycast.dns2.nextdns.io/steering.nextdns.io/cname.vercel-dns.com/ipv4-hydron-clt-1.edge.nextdns.io/ipv6-hydron-clt-1.edge.nextdns.io/ipv4-tier-clt-1.edge.nextdns.io/ipv6-tier-clt-1.edge.nextdns.io/ipv4-anexia-mnz-1.edge.nextdns.io/ipv6-anexia-mnz-1.edge.nextdns.io/ipv4-zepto-xrs-1.edge.nextdns.io/ipv6-zepto-xrs-1.edge.nextdns.io/ipv4-zepto-iad-1.edge.nextdns.io/ipv6-zepto-iad-1.edge.nextdns.io/ipv4-hetzner-iad-1.edge.nextdns.io/ipv6-hetzner-iad-1.edge.nextdns.io/ipv4-teraswitch-pit-1.edge.nextdns.io/ipv4-cloudzy-pit-1.edge.nextdns.io/ipv4-anexia-atl-1.edge.nextdns.io/ipv6-anexia-atl-1.edge.nextdns.io/ipv4-vultr-atl-1.edge.nextdns.io/ipv6-vultr-atl-1.edge.nextdns.io/doh3.steering.nextdns.io/doh3.dns.nextdns.io/doh3.dns1.nextdns.io/doh3.dns2.nextdns.io/probe-test.dns.nextdns.io/ping.nextdns.io/test.nextdns.io/router.nextdns.io/4#inet#fw4#nextdns_hosts_4,6#inet#fw4#nextdns_hosts_6
> ```
> 
> With patch above:
> 
> ```
> ...
> nftset=/assets.mediadelivery.net/cdn.masterbooks.com/filters.1blocker.com/4#inet#fw4#allowed_cdns_4,6#inet#fw4#allowed_cdns_6
> nftset=/my.nextdns.io/dns.nextdns.io/dns1.nextdns.io/dns2.nextdns.io/dns1.steering.nextdns.io/dns2.steering.nextdns.io/anycast.dns.nextdns.io/anycast.dns1.nextdns.io/anycast.dns2.nextdns.io/steering.nextdns.io/cname.vercel-dns.com/ipv4-hydron-clt-1.edge.nextdns.io/ipv6-hydron-clt-1.edge.nextdns.io/ipv4-tier-clt-1.edge.nextdns.io/ipv6-tier-clt-1.edge.nextdns.io/ipv4-anexia-mnz-1.edge.nextdns.io/ipv6-anexia-mnz-1.edge.nextdns.io/ipv4-zepto-xrs-1.edge.nextdns.io/ipv6-zepto-xrs-1.edge.nextdns.io/ipv4-zepto-iad-1.edge.nextdns.io/ipv6-zepto-iad-1.edge.nextdns.io/ipv4-hetzner-iad-1.edge.nextdns.io/ipv6-hetzner-iad-1.edge.nextdns.io/ipv4-teraswitch-pit-1.edge.nextdns.io/ipv4-cloudzy-pit-1.edge.nextdns.io/ipv4-anexia-atl-1.edge.nextdns.io/ipv6-anexia-atl-1.edge.nextdns.io/ipv4-vultr-atl-1.edge.nextdns.io/ipv6-vultr-atl-1.edge.nextdns.io/doh3.steering.nextdns.io/doh3.dns.nextdns.io/doh3.dns1.nextdns.io/doh3.dns2.nextdns.io/probe-test.dns.nextdns.io/4#inet#fw4#nextdns_hosts_4,6#inet#fw4#nextdns_hosts_6
> nftset=/ping.nextdns.io/test.nextdns.io/router.nextdns.io/4#inet#fw4#nextdns_hosts_4,6#inet#fw4#nextdns_hosts_6
> ```

不错，解决了

Fail-Safe, 2025-12-28:
> [@cspwizard](https://github.com/cspwizard) I took on [@dave14305](https://github.com/dave14305)'s challenge:
>
>
> Would you mind giving this patch a try? This is working well for me and fixes the issue [...]

@cspwizard && @dave14305 -- Was there any interest in trying to move forward with this patch?

dave14305, 2025-12-28:
> Was there any interest in trying to move forward with this patch?

Seems like a worthy effort for those that use dnsmasq-full for set population.