openwrt/openwrt #17555: why error

Dr-chen99, 2025-01-10:
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe7 in position 13: invalid continuation byte

../../../build_dir/target-x86_64_musl/util-linux-2.40.2/meson.build:252:15: ERROR: Unhandled python exception

    This is a Meson bug and should be reported!
make[3]: *** [Makefile:912: /media/ta/Linux/openwrt/build_dir/target-x86_64_musl/util-linux-2.40.2/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[3]: Leaving directory '/media/ta/Linux/openwrt/package/utils/util-linux'
time: package/utils/util-linux/compile#2.38#0.77#3.01
    ERROR: package/utils/util-linux failed to build.

brada4, 2025-01-10:
Mind to follow issue template? util-linux is not builtt by default.

neheb, 2025-07-29:
@Dr-chen99 meson cannot handle Unicode paths. I'm guessing the path has Chinese in it.