openwrt/openwrt #17557: Misleading UI that requires `folders`, not `files`

bilogic, 2025-01-10:
### Describe the bug

1. "Paths of additional hosts files" instead of "Additional hosts files"
2. OR "Paths for `addn-hosts`" instead of "Additional hosts files"
3. Placeholder should be `/tmp/hosts/` instead of `/etc/dnsmasq.hosts`

[image]

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA8100

### Image kind

Official downloaded image

### Steps to reproduce

see #17511

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-10:
World writable include place to fix gui?

bilogic, 2025-01-12:
I had this issue because I was using the `uci` CLI to set things up

Reason: https://github.com/openwrt/openwrt/issues/17511#issuecomment-2585194191
Solution: https://github.com/openwrt/openwrt/issues/17511#issuecomment-2585191032

closed 2025-01-12 by bilogic