openwrt/openwrt #17562: package/system/apk [host] failed to build. (master branch)

GrumpyMeow, 2025-01-10:
### Describe the bug

I've just started with OpenWRT development and get an error setting up my environment. I'm following these instructions: https://openwrt.org/docs/guide-developer/toolchain/building_openwrt_on_openwrt.

Eventually i get an error when running: `make package/system/apk/compile`
```
make[2]: Entering directory '/root/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/root/openwrt/scripts/config'
make[1]: Entering directory '/root/openwrt'
+ mkdir -p /root/openwrt/staging_dir/target-x86_64_musl
+ cd /root/openwrt/staging_dir/target-x86_64_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /root/openwrt/build_dir/target-x86_64_musl/stamp
touch /root/openwrt/staging_dir/target-x86_64_musl/.prepared
make[2]: Entering directory '/root/openwrt/package/utils/lua'
make[2]: Leaving directory '/root/openwrt/package/utils/lua'
time: package/utils/lua/host-compile#0.76#0.13#0.84
make[2]: Entering directory '/root/openwrt/package/system/apk'
VERSION=3.0.0_pre20241130 MAKEFLAGS="" /root/openwrt/staging_dir/host/bin/ninja  -j1 -C /root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/openwrt-build 
ninja: Entering directory `/root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/openwrt-build'
[1/1] Linking target src/apk
FAILED: src/apk 
/root/openwrt/staging_dir/host/bin/gcc  -o src/apk src/apk.p/apk.c.o src/apk.p/app_adbdump.c.o src/apk.p/app_adbsign.c.o src/apk.p/app_add.c.o src/apk.p/app_audit.c.o src/apk.p/app_cache.c.o src/apk.p/app_convdb.c.o src/apk.p/app_convndx.c.o src/apk.p/app_del.c.o src/apk.p/app_dot.c.o src/apk.p/app_extract.c.o src/apk.p/app_fetch.c.o src/apk.p/app_fix.c.o src/apk.p/app_index.c.o src/apk.p/app_info.c.o src/apk.p/app_list.c.o src/apk.p/app_manifest.c.o src/apk.p/app_mkndx.c.o src/apk.p/app_mkpkg.c.o src/apk.p/app_policy.c.o src/apk.p/app_update.c.o src/apk.p/app_upgrade.c.o src/apk.p/app_search.c.o src/apk.p/app_stats.c.o src/apk.p/app_verify.c.o src/apk.p/app_version.c.o src/apk.p/applet.c.o -L/root/openwrt/staging_dir/host/lib -L/root/openwrt/staging_dir/hostpkg/lib -L/root/openwrt/staging_dir/target-x86_64_musl/host/lib -flto -Wl,--as-needed -Wl,--no-undefined -Wl,--start-group src/libapk.a /root/openwrt/staging_dir/host/lib/libssl.a /root/openwrt/staging_dir/host/lib/libcrypto.a /root/openwrt/staging_dir/host/lib/libz.a /root/openwrt/staging_dir/host/lib/libzstd.a -Wl,--end-group -pthread
/usr/bin/ld: /root/openwrt/tmp/cclmjApF.ltrans0.ltrans.o: in function `stats_main':
/root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/openwrt-build/../src/app_stats.c:30:(.text+0x711): undefined reference to `apk_out_fmt'
/usr/bin/ld: /root/openwrt/tmp/cclmjApF.ltrans0.ltrans.o: in function `ver_indexes':
/root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/openwrt-build/../src/app_version.c:33:(.text+0x76e): undefined reference to `apk_out_fmt'
...
/root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/openwrt-build/../src/apk_defines.h:171:(.text.startup+0x4e1): undefined reference to `_apk_array__free'
/usr/bin/ld: /root/openwrt/tmp/cclmjApF.ltrans0.ltrans.o: in function `main':
/root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/openwrt-build/../src/apk.c:477:(.text.startup+0x50f): undefined reference to `apk_out_fmt'
/usr/bin/ld: /root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/openwrt-build/../src/apk.c:35:(.text.startup+0x5a5): undefined reference to `apk_out_fmt'
/usr/bin/ld: /root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/openwrt-build/../src/apk.c:504:(.text.startup+0x5c7): undefined reference to `apk_error_str'
/usr/bin/ld: /root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/openwrt-build/../src/apk.c:504:(.text.startup+0x5e0): undefined reference to `apk_out_fmt'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:97: /root/openwrt/build_dir/hostpkg/apk-3.0.0_pre20241130/.built] Error 1
make[2]: Leaving directory '/root/openwrt/package/system/apk'
time: package/system/apk/host-compile#4.82#0.77#5.68
    ERROR: package/system/apk [host] failed to build.
make[1]: *** [package/Makefile:183: package/system/apk/host/compile] Error 1
make[1]: Leaving directory '/root/openwrt'
make: *** [/root/openwrt/include/toplevel.mk:233: package/system/apk/compile] Error 2
```

Is this a known issue? Or 

### OpenWrt version

r28534-f5b1d340be

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-01-11:
The build instructions you're following are not from the project's developers but an OpenWrt forum member.  As this guide is not supported by the project's developers you should follow-up with the guide author about the likely source of the issue.

The build instructions in the [OpenWrt Developer's Guide](https://openwrt.org/docs/guide-developer/start) are the supported way of compiling firmwares.

GrumpyMeow, 2025-01-11:
I totally understand that I’m following a non-standard path. It feels like I’m almost there though and that it would allow for a smaller development-deployment-loop. As all other packages build fine and only the apk-package fails, I suspected that some kind of wrong apk-reference is being used. As I also saw some other commits about this kind of issue in the openwrt/packages repository: https://github.com/openwrt/packages/issues/25456
I also have an environment (devcontainer) which is conform the guide which i use to build images.

GrumpyMeow, 2025-01-11:
I did some more digging and find it "weird" that on one side `/root/openwrt/staging_dir/host/bin/gcc` is being used, but then `/usr/bin/ld` used. 
On OpenWRT environment: `GNU ld (GNU Binutils) 2.42`, while on ci-build-devcontainer an older version is used: `GNU ld (GNU Binutils for Debian) 2.35.2`. I can imagine this is causing my issues.

GrumpyMeow, 2025-01-11:
I've switched to using the SDK. The road of openwrt-on-openwrt was too complicated for me.

closed 2025-01-11 by GrumpyMeow