openwrt/openwrt #17565: Wireless 2.4GHZ not working when enabled - Linksys E8450 - mt7622

Dvineone, 2025-01-10:
### Describe the bug

After a fresh install of Openwrt 24.10 rc5 also same error on Openwrt 23.05.5 on Linksys E8450 radio mt7622 using signed UBI installer 1.1.3 the 24.ghz wireless will not start.

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

Go to Wireless setup page in luci and click enable.

### Actual behaviour

the wireless 2.4ghz and the radio comes up but the wireless network is still disabled 

### Expected behaviour

Expect Radio to activate and wireless network to become active and start broadcasting the ssid

### Additional info

[image]

--------------------------------------------- Kernal logs----------------------------------------------------------------------

`Fri Jan 10 16:44:45 2025 daemon.notice netifd: radio0 (6048): WARNING: Variable 'data' does not exist or is not an array/object
Fri Jan 10 16:44:45 2025 daemon.notice hostapd: Set new config for phy wl0:
Fri Jan 10 16:44:45 2025 daemon.notice wpa_supplicant[1671]: Set new config for phy wl0
Fri Jan 10 16:44:45 2025 daemon.notice wpa_supplicant[1671]: Set new config for phy wl0
Fri Jan 10 16:44:45 2025 daemon.notice hostapd: Set new config for phy wl0: /var/run/hostapd-wl0.conf
Fri Jan 10 16:44:45 2025 daemon.notice hostapd: Restart interface for phy wl0
Fri Jan 10 16:44:45 2025 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 hw_mode=g supported_rates=60 90 120 180 240 360 480 540 basic_rates=60 120 240 beacon_int=100 stationary_ap=1 chanlist=1 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] channel=1  interface=wl0-ap0 bssid=80:69:1a:23:e2:a3 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 ocv=0 sae_require_mfp=1 sae_pwe=2 wpa_passphrase=**** wpa_psk_file=/var/run/hostapd-wl0-ap0.psk sae_password_file=/var/run/hostapd-wl0-ap0.sae auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=*** bridge=br-lan wds_bridge= snoop_iface=br-lan wpa_disable_eapol_key_retries=0 wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256 SAE okc=1 ieee802
Fri Jan 10 16:44:45 2025 daemon.err hostapd: Could not set interface wl0-ap0 flags (UP): I/O error
Fri Jan 10 16:44:45 2025 daemon.err hostapd: nl80211: Could not set interface 'wl0-ap0' UP
Fri Jan 10 16:44:45 2025 daemon.notice hostapd: nl80211: deinit ifname=wl0-ap0 disabled_11b_rates=0
Fri Jan 10 16:44:45 2025 daemon.err hostapd: nl80211 driver initialization failed.
Fri Jan 10 16:44:45 2025 daemon.notice hostapd: wl0-ap0: CTRL-EVENT-TERMINATING
Fri Jan 10 16:44:45 2025 daemon.err hostapd: hostapd_free_hapd_data: Interface wl0-ap0 wasn't started
Fri Jan 10 16:44:45 2025 daemon.notice hostapd: hostapd.add_iface failed for phy wl0 ifname=wl0-ap0
Fri Jan 10 16:44:45 2025 daemon.notice netifd: Wireless device 'radio0' is now up
`

-------------------------------------------------------------------- dmesg output at boot up --------------------------------------------------------------

`[   29.938765] mt7622-wmac 18000000.wmac: Message 00000010 (seq 1) timeout
[   29.945418] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   30.169193] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   30.389218] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   30.609192] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   30.829189] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   31.049198] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   31.269115] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   31.489110] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   31.709113] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   31.929185] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   32.149190] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
`

-------------------------------------------------------------- /etc/config/wireless -------------------------------------------------------------------------

`config wifi-device 'radio0'
	option type 'mac80211'
	option phy 'wl0'
	option country 'US'
	option cell_density '0'
	option band '2g'
	option channel '1'
	option txpower '25'
	option htmode 'HT20'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid '***'
	option encryption 'sae-mixed'
	option key '****'
	option ocv '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option phy 'wl1'
	option cell_density '0'
	option htmode 'HE20'
	option band '5g'
	option channel 'auto'
	option country 'US'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid '***'
	option encryption 'sae-mixed'
	option key '***'
	option ocv '0'
`

------------------------------------------------------------ /var/run/hostapd-wl0.conf ---------------------------------------------------------------------
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
hw_mode=g
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
beacon_int=100
stationary_ap=1
channel=1
chanlist=1

#num_global_macaddr=1
ieee80211n=1
ht_coex=0
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]

interface=wl0-ap0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
ocv=0
sae_require_mfp=1
sae_pwe=2
wpa_passphrase=***
wpa_psk_file=/var/run/hostapd-wl0-ap0.psk
sae_password_file=/var/run/hostapd-wl0-ap0.sae
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=***
bridge=br-lan
wds_bridge=
snoop_iface=br-lan
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256 SAE
okc=1
ieee80211w=1
group_mgmt_cipher=AES-128-CMAC
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-wl0-ap0.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=80:69:1a:23:e2:a3
#default_macaddr

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

PSzczepanski1996, 2025-02-07:
Can also confirm that. After updating my NetGear WAX206 router with OpenWrt 24.10.0 r28427-6df0e3d02a for some reason my router Wi-Fi 2.4GHz stopped working at all. Can't connect my Ezviz cameras, my S23 mobile phone, and Tapo T110 Hub which was also linked is unable to connect.

DraganBocevski, 2025-04-08:
Same behaviour on Xiaomi R3G (ramips/mt7621) after upgrade from 23.x to OpenWrt 24.10.0 r28427-6df0e3d02a

NicolasFBorges, 2025-05-10:
Anyone found how to downgrade?

davygravy, 2026-01-17:
Confirmed on upgrading from 24.10.5 to rc2-25.12.0.  

This problem may be the same one where deleting a SSID and then re-creating it via Luci fixes it. 

May be related to some json/naming of the wifi interfaces. ? phy0 vs wl0 or some such. 

A similar/same? issue is mentioned in the original github site where the first UBI images were developed and posted.