openwrt/openwrt #17573: RTL8188GU not detected after installing kmod-rtl8xxxu

MightyCoderX, 2025-01-11:
### Describe the bug

I installed the kernel module `kmod-rtl8xxxu` and rebooted but cannot see my adapter in `ip link`'s output, it's only visible in `lsusb`'s (line 3):
```
Bus 001 Device 005: ID 0424:7800
Bus 001 Device 003: ID 0424:2514
Bus 001 Device 006: ID 0bda:c811 Realtek 802.11ac NIC
Bus 001 Device 002: ID 0424:2514
Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.167 dwc_otg_hcd DWC OTG Controller
```
doesn't show up in `iw list` either

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

bcm27xx/bcm2710

### Device

Raspberry Pi 3 Model B Plus Rev 1.3

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

1715173329, 2025-01-12:
RTL8188GU is not supported on OpenWrt 23.05 (mac80211 backports 6.1). You need to upgrade to 24.10.

hauke, 2025-01-12:
Support for the RTL8188GU was added in https://github.com/torvalds/linux/commit/70664495e3d24803ec47ccaccee9a822220558a0 in upstream Linux. This was merged into Linux 6.4. OpenWrt 23.05 uses the wireless subsystem from Linux 6.1. This device is only supported with OpenWrt 24.10 and later.

closed 2025-01-12 by hauke

MightyCoderX, 2025-01-12:
> RTL8188GU is not supported on OpenWrt 23.05 (mac80211 backports 6.1). You need to upgrade to 24.10.

This is the latest version supported by raspi according to your website, is 24.10 in beta perhaps?

MightyCoderX, 2025-01-12:
> Does it show up like in ubuntu or fedora live cd?

It works on my arch machine without extra drivers, it just requires to run this command to switch it to nic mode from cdrom mode, `usb_modeswitch -KW -v 0bda -p 1a2b`

MightyCoderX, 2025-01-12:
> Support for the RTL8188GU was added in https://github.com/torvalds/linux/commit/70664495e3d24803ec47ccaccee9a822220558a0 in upstream Linux. This was merged into Linux 6.4. OpenWrt 23.05 uses the wireless subsystem from Linux 6.1. This device is only supported with OpenWrt 24.10 and later. 

I'll check if it is available for my device and update, excuse me for the trouble of an extra issue I should have put a bit more effort in my research.

elschman, 2025-08-25:
I tried 24.10 but still the device is just seen with lsusb but not detected as network device...
which module is need to be installed/loaded with lsmod ?