openwrt/openwrt #17577: OpenWrt 24.10.0-rc5 /overlay is too small on ipq806x/generic TP-Link Talon AD7200

Konstantin-Glukhov, 2025-01-11:
### Describe the bug

/overlay partition is only 20MB
It is way smaller then the prior releases.
It is impossible to install an essential package samba4-server

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Talon AD7200

### Image kind

Official downloaded image

### Steps to reproduce

```
df -h
Filesystem           Type            Size      Used Available Use% Mounted on
/dev/root            squashfs        4.5M      4.5M         0 100% /rom
tmpfs                tmpfs         232.5M      1.3M    231.2M   1% /tmp
tmpfs                tmpfs         232.5M     56.0K    232.4M   0% /tmp/root
tmpfs                tmpfs         512.0K         0    512.0K   0% /dev
/dev/mtdblock13      jffs2          20.5M      6.8M     13.7M  33% /overlay
overlayfs:/overlay   overlay        20.5M      6.8M     13.7M  33% /
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-01-11:
The device only has a 32MB flash so there's no chance to increase the usable space.  This is not a bug - it's a hardware limitation.

You have some options:
- try using `ksmbd` instead of `samba`
- as the device has USB ports, add a USB storage device to [expand the root filesystem](https://openwrt.org/docs/guide-user/additional-software/extroot_configuration) (aka extroot)
- use the image builder (or the online equivalents such as [Attended Sysupgrade](https://openwrt.org/docs/guide-user/installation/attended.sysupgrade) or [owut](https://openwrt.org/docs/guide-user/installation/sysupgrade.owut)) to build a custom firmware containing the required packages
- compile a custom firmware from sources with the required packages built-in
- replace the device with one with more usable internal storage

BTW: Building the packages into an images takes advantage of SquashFS compression, so the package occupies less flash space than if it is installed into the overlay, at the cost of requiring assembling a new image each time you want to upgrade a package.

Konstantin-Glukhov, 2025-01-11:
Version 23.05 had enough space in /overlay. Why did it drop so dramatically?

Konstantin-Glukhov, 2025-01-11:
After further investigation I suspect there is a bug in opkg install script for samba4-libs.

The size of /overlay only increased by 0.3MB in OpenWrt-24.10.
samba4-libs package size even decreased for OpenWrt-24.10.

OpenWrt-23.05:
Size: 9004395
Filename: samba4-libs_4.18.8-1_arm_cortex-a15_neon-vfpv4.ipk

OpenWrt-24.10
Size: 8933404
Filename: samba4-libs_4.18.8-r1_arm_cortex-a15_neon-vfpv4.ipk

But opkg install give the the following estimation error in OpenWrt-24.10:

opkg install samba4-libs
Installing samba4-libs (4.18.8-r1) to root...
Collected errors:
 * verify_pkg_installable: Only have 18092kb available on filesystem /overlay, pkg samba4-libs needs 26770
 * opkg_install_cmd: Cannot install package samba4-libs.

aimacintyre, 2025-01-12:
> I suspect there is a bug in opkg install script for samba4-libs.

the installable components in .ipk files are compressed; they are decompressed on installation.  If you examine the .ipk carefully (e.g. with 7Zip on Windows) you will find there is no estimation error by opkg.

Smollejan, 2025-01-12:
I am experiencing the same in version 24.10.0, the `/overlay/upper` location is different compared to version 23.05.5. There is a `/run` file that causes an endless loop. Could this be related somehow?
[image]

aimacintyre, 2025-01-12:
> I am experiencing the same in version 24.10.0

24.10.0 doesn't exist yet - what version did you actually install?

> the `/overlay/upper` location is different compared to version 23.05.5. There is a `/run` file that causes an endless loop. Could this be related somehow?

This doesn't appear related to the original report at all; did you attempt to preserve the configuration during flashing or restore a saved configuration from 23.05.5 after flashing?

Smollejan, 2025-01-12:
> 24.10.0 doesn't exist yet - what version did you actually install?

My bad: 24.10.0-rc5

> This doesn't appear related to the original report at all; did you attempt to preserve the configuration during flashing or restore a saved configuration from 23.05.5 after flashing?

Alright, thanks for your reply! No I didn't save my configuration. Clean flash on Archer C7 v2. But got the same overlay error with the dnscrypt-proxy 2 package.

aimacintyre, 2025-01-13:
> No I didn't save my configuration. Clean flash on Archer C7 v2. But got the same overlay error with the dnscrypt-proxy 2 package.

I would suggest posting about your problem in the `Installing and Using OpenWrt` section of the forums with as much detail as you can about your device, exactly which image(s) have been flashed, how you have flashed them and what other configuration changes you have made to get to this state.

Konstantin-Glukhov, 2025-01-13:
> > I suspect there is a bug in opkg install script for samba4-libs.
> 
> the installable components in .ipk files are compressed; they are decompressed on installation. If you examine the .ipk carefully (e.g. with 7Zip on Windows) you will find there is no estimation error by opkg.

Then explain the fact that the same package but bigger size installs on 23.05.5 and the smaller size one does not install on 24.10.r5

As I mentioned earlier the size of samba4-libs reported by opkg is smaller for 24.10.r5:

```
OpenWrt-23.05:
Size: 9004395
Filename: samba4-libs_4.18.8-1_arm_cortex-a15_neon-vfpv4.ipk

OpenWrt-24.10
Size: 8933404
Filename: samba4-libs_4.18.8-r1_arm_cortex-a15_neon-vfpv4.ipk
```

aimacintyre, 2025-01-13:
> Then explain the fact that the same package but bigger size installs on 23.05.5 and the smaller size one does not install on 24.10.r5

Have you actually rebuilt your 23.05.5 installation from an official image to validate that?  I ask because the software breakdown I can see based on files from the OpenWrt website looks like this:
- device has 29MB of 32MB flash available for kernel + root filesystem + overlay filesystem
- official image (kernel + root filesystem) is ~8.25MB
- overlay file system is therefore ~20.75MB
- Samba 4 Server package + dependencies (decompressed): >26MB (samba4-libs expands to 25.75MB, other dependencies could easily add 2+MB)

This obviously doesn't compute as there's no way 26+MB can fit into ~20.75MB.  If you can make it compute, you need to analyse your 23.05.5 system to find out where either: some packages aren't being installed that could/should be, compression has been used in some way or some extra storage is available.

Perhaps surprisingly, on this target the 23.05. to 24.10 transition doesn't seem to involve much change (~3%) in the calculus...

Konstantin-Glukhov, 2025-01-13:
Yes, I did install both 23.05.5 and 24.10.rc5 from scratch. After install 23.05 shows 20.8MB size on /overlay and 24.10 shows 20.5MB (the difference is only 0.3MB).
I am not an expert on opkg, but 'opkg info' shows the bigger size for samba4-libs on 23.05 than on 24.10.
This is why I suspected a bug in accounting in 24.10. I cannot imagine that the samba4-libs went up by 15MB for OpenWrt 24.10.

KanjiMonster, 2025-01-13:
What you are running into is a change in how the (expected) Installed Size is calculated:

23.05:

```
Package: samba4-libs
Version: 4.18.8-1
...
Installed-Size: 9120239 <--
...
Size: 9015016

```
24.10:

```
Package: samba4-libs
Version: 4.18.8-r1
...
Installed-Size: 27504640 <--
...
Size: 8947353
```

In 23.05, the Package index had the compressed package size as installed-size, while 24.10 uses the uncompressed package size (see https://github.com/openwrt/openwrt/commit/49e8f532989c342fff44c1c558f8182585673833 for the change).

The core issue here is that the package will take up more space than the .ipk once unpacked, and the compression of the overlay filesystem (if there is one) is worse than gzip (due to various factors).

So with the old value it was easy to run into a situation where the system ran out of space while trying to install a package, leaving a broken system, with the opkg state corrupted.

With the new value you sometimes run into situations where opkg refuses to install, but it's a better alternative than accidentally bricking your system by installing too much.

At your own risk, you can tell `opkg` to ignore the space check via `--force-space`, but you may end up with a broken system where only factory resetting the system will get you in a valid state again.

Konstantin-Glukhov, 2025-01-14:
Well, in 24.10 the size is still calculated incorrectly.
I was able to successfully install the samba4-libs package using --force-space.
It takes only 13MB on /overlay, and not 26MB as reported.
So, in this case "opkg install" requires twice more space than it needs.
It is obviously a bug.

By the way, was this change clearly documented somewhere in 24.10 release notes?

aimacintyre, 2025-01-14:
> Well, in 24.10 the size is still calculated incorrectly. I was able to successfully install the samba4-libs package using --force-space. It takes only 13MB on /overlay, and not 26MB as reported. So, in this case "opkg install" requires twice more space than it needs. It is obviously a bug.

I hadn't realised that JFFS2 was able to use compression but it appears that LZMA compression is enabled by default in OpenWrt in most cases so that explains how you can get Samba 4 into your device.  If you read about JFFS2, you will find that it is impossible to predict how much space will be used before writing files so opkg cannot effectively predict how much space will be used once a package is installed.

Konstantin-Glukhov, 2025-01-14:
Maybe for JFFS2 "opkg install" should not estimate the space then?

KanjiMonster, 2025-01-14:
> Maybe for JFFS2 "opkg install" should not estimate the space then?

As said earlier, the space check is to prevent people from accidentally breaking their systems by running out of space. Which would be then very easy if opkg did not do a space check anymore.

JFFS2 isn't the only type of filsystem used for the overlay; it may also be ubifs, f2fs or ext4. Not all of these use compression, and how well they compress will also differ. And at the same time how well it compress will also differ with the package contents themselves, so you cannot reliably estimate it just from the type of filesystem and the uncompressed size.

Also opkg does not know or care what type of underlying filesystem is used, so all it can rely on is the "Installed-Size".

Konstantin-Glukhov, 2025-01-17:
What is the point of doing space check if you cannot estimate it at all for JFFS2? Can't you just put an "if" statement to skip checking for JFFS2? Or at least divided the estimation size by 2? At least you will get some approximation of the space requirement.

KanjiMonster, 2025-01-17:
> What is the point of doing space check if you cannot estimate it at all for JFFS2? Can't you just put an "if" statement to skip checking for JFFS2? Or at least divided the estimation size by 2? At least you will get some approximation of the space requirement.

As I've now explained multiple times, the point is preventing users from easily breaking their systems. Which will happen if you try to install something that does not fit.

Konstantin-Glukhov, 2025-01-27:
Maybe you are being to presumptive? What evidence do you have for 'users breaking their systems by trying to install packages that do not feet into the storage'?

KanjiMonster, 2025-01-27:
> Maybe you are being to presumptive? What evidence do you have for 'users breaking their systems by trying to install packages that do not feet into the storage'?

https://forum.openwrt.org/search?q=opkg%20%22no%20space%20left%22

https://dev.archive.openwrt.org/ticket/13031

Anyway, this won't be reverted. You can always skip the check with `--force-space` if you are confident it will fit anyway.