openwrt/openwrt #17578: wifi-scripts /  "Use new ucode based scripts "  -> Config error

grosjo, 2025-01-11:
### Describe the bug

Upon compilation, choose "Use new ucode based scripts "  in the "Base system"
This lead to the following error

```
Sat Jan 11 19:41:41 2025 daemon.notice netifd: radio0 (2077): wifi-scripts: Preparing interface: phy0-ap0 with MAC: 00:00:00:00:00:01
Sat Jan 11 19:41:41 2025 daemon.notice netifd: radio0 (2077): wifi-scripts: network is not present in the schema

```
Luci does not manage to solve the config

### OpenWrt version

r28531-5141e2d861

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

YunCore/KuWfi CPE830(D)

### Image kind

Self-built image

### Steps to reproduce

Choose "Use new ucode based scripts" in "Base system"

### Actual behaviour

Wireless not setup

### Expected behaviour

Wireless configuration transitioned properly from legacy wifi-scripts

### Diffconfig

```text
# ./scripts/diffconfig.sh
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_yuncore_cpe830=y
# CONFIG_ATH9K_HWRNG is not set
CONFIG_ATH9K_SUPPORT_PCOEM=y
# CONFIG_ATH9K_UBNTHSR is not set
# CONFIG_ATH_USER_REGD is not set
# CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set
# CONFIG_DROPBEAR_AGENTFORWARD is not set
# CONFIG_DROPBEAR_CHACHA20POLY1305 is not set
# CONFIG_DROPBEAR_CURVE25519 is not set
# CONFIG_DROPBEAR_DBCLIENT_AGENTFORWARD is not set
# CONFIG_DROPBEAR_ED25519 is not set
# CONFIG_DROPBEAR_U2F is not set
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
CONFIG_KERNEL_DEVKMEM=y
CONFIG_KERNEL_DEVMEM=y
# CONFIG_KERNEL_DEVPTS_MULTIPLE_INSTANCES is not set
CONFIG_KERNEL_DEVTMPFS=y
CONFIG_KERNEL_DEVTMPFS_MOUNT=y
CONFIG_KERNEL_DYNAMIC_DEBUG=y
# CONFIG_KERNEL_ELF_CORE is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_KERNEL_IPV6_SEG6_LWTUNNEL is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_KALLSYMS is not set
# CONFIG_KERNEL_KEYS is not set
# CONFIG_KERNEL_MAGIC_SYSRQ is not set
# CONFIG_KERNEL_MPTCP is not set
# CONFIG_KERNEL_NAMESPACES is not set
# CONFIG_KERNEL_NET_L3_MASTER_DEV is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_NF_CONNTRACK_TIMEOUT is not set
# CONFIG_KERNEL_POSIX_MQUEUE is not set
CONFIG_KERNEL_PROC_PAGE_MONITOR=y
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_SECCOMP is not set
# CONFIG_KERNEL_SWAP is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_WERROR is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
CONFIG_LUCI_SRCDIET=y
# CONFIG_PACKAGE_MAC80211_MESH is not set
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_getrandom is not set
CONFIG_PACKAGE_iptables-mod-condition=y
CONFIG_PACKAGE_iptables-mod-ipv4options=y
CONFIG_PACKAGE_iptables-mod-nfqueue=y
CONFIG_PACKAGE_iptables-mod-physdev=y
CONFIG_PACKAGE_iptables-mod-proto=y
CONFIG_PACKAGE_iptables-nft=y
CONFIG_PACKAGE_kmod-br-netfilter=y
# CONFIG_PACKAGE_kmod-gpio-button-hotplug is not set
CONFIG_PACKAGE_kmod-ipt-condition=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipv4options=y
CONFIG_PACKAGE_kmod-ipt-nfqueue=y
CONFIG_PACKAGE_kmod-ipt-physdev=y
CONFIG_PACKAGE_kmod-ipt-proto=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nfnetlink-queue=y
CONFIG_PACKAGE_kmod-nft-compat=y
# CONFIG_PACKAGE_kmod-ppp is not set
# CONFIG_PACKAGE_knot-resolver_dnstap is not set
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-filebrowser=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_openwrt-keyring is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
# CONFIG_PACKAGE_rssileds is not set
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-lua=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_urandom-seed is not set
# CONFIG_PACKAGE_urngd is not set
# CONFIG_PACKAGE_usign is not set
CONFIG_PACKAGE_wpad-basic=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zoneinfo-asia=y
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PROCD_SHOW_BOOT=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_MKLIBS=y
# CONFIG_USE_SECCOMP is not set
CONFIG_WIFI_SCRIPTS_UCODE=y
# CONFIG_WPA_MBO_SUPPORT is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

sppmasterspp, 2025-01-13:
I'm not sure but probably the error I see is related to this issue.
I don't get error on compilation but when I run the scripts.
I get this running on qualcommax/ipq807x

```
daemon.notice netifd: radio0 (17956): wifi-scripts: Tearing down phy0
daemon.notice hostapd: Set new config for phy phy0:
daemon.notice wpa_supplicant[2884]: Set new config for phy phy0
daemon.notice netifd: Wireless device 'radio0' is now down
daemon.notice netifd: radio0 (17971): wifi-scripts: Starting
daemon.notice hostapd: Set new config for phy phy0:
daemon.notice wpa_supplicant[2884]: Set new config for phy phy0
daemon.notice netifd: radio0 (17971): command failed: Not supported (-95)
daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
daemon.notice hostapd: Restart interface for phy phy0
daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211h=1 ieee80211d=1 spectrum_mgmt_required=0 beacon_int=100 noscan=1 airtime_mode=0 rssi_reject_assoc_rssi=0 rssi_ignore_probe_request=0 acs_exclude_dfs=0 enable_background_radar=1 min_tx_power=0 hw_mode=a ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][MAX-AMSDU-7935][DSSS_CCK-40][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=0 vht_capab=[RXLDPC][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][MAX-MPDU-7991] he_bss_color=128 he_spr_sr_control=3 ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=0 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_twt_required=0 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=

kern.info kernel: [  241.633470] ath11k c000000.wifi phy0-ap0: entered allmulticast mode
daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
daemon.err hostapd: Interface initialization failed
daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->DISABLED
daemon.notice hostapd: phy0-ap0: AP-DISABLED
daemon.err hostapd: phy0-ap0: Unable to setup interface.
daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
kern.info kernel: [  241.639271] ath11k c000000.wifi phy0-ap0: entered promiscuous mode
user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (phy0-ap0)
kern.info kernel: [  241.689270] ath11k c000000.wifi phy0-ap0: left allmulticast mode
kern.info kernel: [  241.689329] ath11k c000000.wifi phy0-ap0: left promiscuous mode
kern.info kernel: [  241.694472] br-lan: port 6(phy0-ap0) entered disabled state
daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
daemon.err hostapd: hostapd_free_hapd_data: Interface phy0-ap0 wasn't started
daemon.notice hostapd: hostapd.add_iface failed for phy phy0 ifname=phy0-ap0
daemon.notice netifd: Wireless device 'radio0' is now up
daemon.notice netifd: radio0 (17971): wifi-scripts: Configuring 'phy0' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
daemon.notice netifd: radio0 (17971): wifi-scripts: network is not present in the schema
daemon.notice netifd: radio0 (17971): wifi-scripts: Preparing interface: phy0-ap0 with MAC: null
```
Radio0 (5GHz) doesn't start on my build.
Radio1 (2.4GHz) works.

rmandrad, 2025-01-19:
same error in the banana pi4 / mediatek/filogic

grosjo, 2025-02-16:
Who can help ?

rmandrad, 2025-02-16:
@blogic ?

grosjo, 2025-08-03:
@rmandrad This is still not working

For now, I get the following errors enable this option:
```
Sun Aug  3 18:51:27 2025 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Sun Aug  3 18:51:27 2025 daemon.notice hostapd: Reloaded settings for phy phy0
Sun Aug  3 18:51:27 2025 daemon.notice wpa_supplicant[1275]: Set new config for phy phy0
Sun Aug  3 18:51:27 2025 daemon.notice netifd: radio0 (2287): wifi-scripts: type is not present in the schema
Sun Aug  3 18:51:27 2025 daemon.notice netifd: radio0 (2287): wifi-scripts: Configuring 'phy0' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
Sun Aug  3 18:51:27 2025 daemon.notice netifd: radio0 (2287): wifi-scripts: device is not present in the schema
Sun Aug  3 18:51:27 2025 daemon.notice netifd: radio0 (2287): wifi-scripts: radios is not present in the schema
Sun Aug  3 18:51:27 2025 daemon.notice netifd: radio0 (2287): wifi-scripts: Preparing interface: phy0-ap0 with MAC: 44:d1:fa:75:41:06
Sun Aug  3 18:51:27 2025 daemon.notice netifd: radio0 (2287): wifi-scripts: device is not present in the schema
Sun Aug  3 18:51:27 2025 daemon.notice netifd: radio0 (2287): wifi-scripts: radios is not present in the schema
Sun Aug  3 18:51:27 2025 daemon.notice netifd: radio0 (2287): wifi-scripts: Preparing interface: phy0-sta0 with MAC: 46:d1:fa:75:41:06

```

grosjo, 2025-08-03:
@blogic ?

grosjo, 2025-08-09:
@rmandrad @sppmasterspp @blogic @nbd168 any idea how to fix this ?

sppmasterspp, 2025-08-09:
> [@rmandrad](https://github.com/rmandrad) [@sppmasterspp](https://github.com/sppmasterspp) [@blogic](https://github.com/blogic) [@nbd168](https://github.com/nbd168) any idea how to fix this ?

Does it work if you use the old wifi scripts not ucode ones.

grosjo, 2025-08-09:
yes, all ok with legacy scripts

But this ucode thing is officially in teh tree for so many months and is just not working and it seems nobidy cares. So maybe shall it be removed alltogether ?

grosjo, 2025-08-09:
> device is indeed missing from the schema and radios is not a valid option, those errors are just cosmetic and can be ignored

Not, it can not be ignored, as the wifi stops working

nbd168, 2025-08-09:
Please set `CONFIG_WPA_MSG_MIN_PRIORITY=0` in your .config, enable the `udebugd` and `udebug-cli` packages.
On the router do this:
```
uci set udebug.hostapd.enabled=1
uci commit
reload_config
```
Afterwards run:
```
udebug -d 1 logstream > /tmp/log.txt &
wifi
```
If everything works, `/tmp/log.txt` should have detailed log data. Please post it here.

grosjo, 2025-08-10:
```
# cat /tmp/log.txt 
nl80211: multicast to unicast disabled on interface phy0-ap0
nl80211: Fetch survey data
nl80211: Freq survey dump event (freq=2452 MHz noise=-95 channel_time=55529 busy_time=2902 tx_time=175 rx_time=1730 filled=001f
phy0-ap1: Event SURVEY (46) received
Single Channel Survey: (freq=2452 channel_time=55529 channel_time_busy=2902)
Channel Utilization: 11
nl80211: Set beacon (beacon_set=1)
nl80211: Beacon head - hexdump: 80 00 00 00 ff ff ff ff ff ff 7e 27 3c 47 b0 03 7e 27 3c 47 b0 03 00 00 00 00 00 00 00 00 00 00 64 00 31 14 00 00 01 08 8c 12 98 24 b0 48 60 6c 03 01 09
nl80211: Beacon tail - hexdump: 07 06 55 53 20 01 0b 1e 2a 01 00 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 0c 00 0b 05 00 00 0b 00 00 46 05 72 00 00 00 00 3b 02 51 00 2d 1a ed 11 1b ff ff 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 3d 16 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f 08 04 00 00 00 01 00 01 40 c9 11 00 0d 51 09 ff 7c 27 3c 47 b0 03 f5 7e 26 3d 40 7f dd 16 00 50 f2 01 01 00 00 50 f2 04 01 00 00 50 f2 04 01 00 00 50 f2 02 dd 18 00 50 f2 02 01 01 80 00 03 a4 00 00 27 a4 00 00 42 43 5e 00 62 32 2f 00
nl80211: ifindex=6
nl80211: beacon_int=100
nl80211: beacon_rate=0
nl80211: rate_type=0
nl80211: dtim_period=2
nl80211: ssid=VILLAJOLAM
  * beacon_int=100
  * dtim_period=2
nl80211: hidden SSID zero len
nl80211: privacy=3
nl80211: auth_algs=0x1
nl80211: wpa_version=0x3
nl80211: key_mgmt_suites=0x2
nl80211: pairwise_ciphers=0x10
nl80211: group_cipher=0x10
nl80211: beacon_ies - hexdump: 7f 08 04 00 00 00 01 00 01 40
nl80211: proberesp_ies - hexdump: 46 05 72 00 00 00 00 7f 08 04 00 00 00 01 00 01 40
nl80211: assocresp_ies - hexdump: 46 05 72 00 00 00 00 7f 08 04 00 00 00 01 00 01 40
  * freq=2452
  * eht_enabled=0
  * he_enabled=0
  * vht_enabled=0
  * ht_enabled=1
  * radar_background=0
  * sec_channel_offset=0
  * channel_type=1
nl80211: multicast to unicast disabled on interface phy0-ap1
nl80211: Fetch survey data
nl80211: Freq survey dump event (freq=2452 MHz noise=-95 channel_time=58758 busy_time=3069 tx_time=186 rx_time=1834 filled=001f
phy0-ap0: Event SURVEY (46) received
Single Channel Survey: (freq=2452 channel_time=58758 channel_time_busy=3069)
Channel Utilization: 13
nl80211: Set beacon (beacon_set=1)
nl80211: Beacon head - hexdump: 80 00 00 00 ff ff ff ff ff ff 7c 27 3c 47 b0 03 7c 27 3c 47 b0 03 00 00 00 00 00 00 00 00 00 00 64 00 21 14 00 0b 56 32 2d 42 41 43 4b 59 41 52 44 01 08 8c 12 98 24 b0 48 60 6c 03 01 09
nl80211: Beacon tail - hexdump: 07 06 55 53 20 01 0b 1e 2a 01 00 0b 05 00 00 0d 00 00 46 05 72 00 00 00 00 3b 02 51 00 2d 1a ed 11 1b ff ff 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 3d 16 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f 08 04 00 00 00 01 00 01 40 dd 18 00 50 f2 02 01 01 81 00 03 a4 00 00 27 a4 00 00 42 43 5e 00 62 32 2f 00
nl80211: ifindex=5
nl80211: beacon_int=100
nl80211: beacon_rate=0
nl80211: rate_type=0
nl80211: dtim_period=2
nl80211: ssid=V2-BACKYARD
  * beacon_int=100
  * dtim_period=2
nl80211: hidden SSID not in use
nl80211: privacy=0
nl80211: auth_algs=0x1
nl80211: wpa_version=0x0
nl80211: key_mgmt_suites=0x4
nl80211: pairwise_ciphers=0x0
nl80211: group_cipher=0x1
nl80211: beacon_ies - hexdump: 7f 08 04 00 00 00 01 00 01 40
nl80211: proberesp_ies - hexdump: 46 05 72 00 00 00 00 7f 08 04 00 00 00 01 00 01 40
nl80211: assocresp_ies - hexdump: 46 05 72 00 00 00 00 7f 08 04 00 00 00 01 00 01 40
  * freq=2452
  * eht_enabled=0
  * he_enabled=0
  * vht_enabled=0
  * ht_enabled=1
  * radar_background=0
[... 616 lines cut ...]
nl80211: dtim_period=2
nl80211: ssid=VILLAJOLAM
  * beacon_int=100
  * dtim_period=2
nl80211: hidden SSID zero len
nl80211: privacy=3
nl80211: auth_algs=0x1
nl80211: wpa_version=0x3
nl80211: key_mgmt_suites=0x2
nl80211: pairwise_ciphers=0x10
nl80211: group_cipher=0x10
nl80211: beacon_ies - hexdump: 7f 08 04 00 00 00 01 00 01 40
nl80211: proberesp_ies - hexdump: 46 05 72 00 00 00 00 7f 08 04 00 00 00 01 00 01 40
nl80211: assocresp_ies - hexdump: 46 05 72 00 00 00 00 7f 08 04 00 00 00 01 00 01 40
  * freq=2452
  * eht_enabled=0
  * he_enabled=0
  * vht_enabled=0
  * ht_enabled=1
  * radar_background=0
  * sec_channel_offset=0
  * channel_type=1
nl80211: multicast to unicast disabled on interface phy0-ap1

```

grosjo, 2025-08-10:
in both cases, Wifi stops working

closed 2025-08-10 by openwrt-bot