openwrt/openwrt #17579: Unable to build intel-microcode package

vooon, 2025-01-11:
### Describe the bug

I'm unable to build the package on present master, while it was okay at Nov.
Perhaps that's related to opkg -> apk change, as there no changes on the package itself.

```
make[3]: Entering directory '/mnt/LEDE/openwrt/package/firmware/intel-microcode'
rm -f /mnt/LEDE/openwrt/build_dir/target-x86_64_glibc/intel-microcode-3.20240531.1/.built
touch /mnt/LEDE/openwrt/build_dir/target-x86_64_glibc/intel-microcode-3.20240531.1/.built_check
IUCODE_TOOL=/mnt/LEDE/openwrt/staging_dir/target-x86_64_glibc/../host/bin/iucode_tool make -C /mnt/LEDE/openwrt/build_dir/target-x86_64_glibc/intel-microcode-3.20240531.1
make[4]: Entering directory '/mnt/LEDE/openwrt/build_dir/target-x86_64_glibc/intel-microcode-3.20240531.1'
make[4]: Nothing to be done for 'all'.
make[4]: Leaving directory '/mnt/LEDE/openwrt/build_dir/target-x86_64_glibc/intel-microcode-3.20240531.1'
mkdir /mnt/LEDE/openwrt/build_dir/target-x86_64_glibc/intel-microcode-3.20240531.1/intel-ucode-ipkg
mkdir: cannot create directory '/mnt/LEDE/openwrt/build_dir/target-x86_64_glibc/intel-microcode-3.20240531.1/intel-ucode-ipkg': File exists
make[3]: *** [Makefile:56: /mnt/LEDE/openwrt/build_dir/target-x86_64_glibc/intel-microcode-3.20240531.1/.built] Error 1
make[3]: Leaving directory '/mnt/LEDE/openwrt/package/firmware/intel-microcode'
time: package/firmware/intel-microcode/compile#0.42#0.27#0.63
    ERROR: package/firmware/intel-microcode failed to build.
```

### OpenWrt version

17b0d1379a3cd4a2e2dbefa7d0f9bc281269f97e

### OpenWrt release

master

### OpenWrt target/subtarget

x86_64

### Device

x86_64

### Image kind

Self-built image

### Steps to reproduce

```
CONFIG_PACKAGE_intel-microcode=y
```

### Actual behaviour

Build fails.

### Expected behaviour

Build ok.

### Additional info

The fix:

```diff
diff --git a/package/firmware/intel-microcode/Makefile b/package/firmware/intel-microcode/Makefile
index bdd8ae73d8..951a7d6f74 100644
--- a/package/firmware/intel-microcode/Makefile
+++ b/package/firmware/intel-microcode/Makefile
@@ -39,7 +39,7 @@ endef
 
 define Build/Compile
        IUCODE_TOOL=$(STAGING_DIR)/../host/bin/iucode_tool $(MAKE) -C $(PKG_BUILD_DIR)
-       mkdir $(PKG_BUILD_DIR)/intel-ucode-ipkg
+       mkdir -p $(PKG_BUILD_DIR)/intel-ucode-ipkg
[... 427 lines cut ...]
CONFIG_RSYNC_acl=y
CONFIG_RSYNC_lz4=y
CONFIG_RSYNC_xattr=y
CONFIG_RSYNC_xxhash=y
CONFIG_RSYNC_zstd=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_STRIP_ARGS="--strip-all"
CONFIG_TARGET_KERNEL_PARTSIZE=32
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=4096
CONFIG_TARGET_SUFFIX="gnu"
CONFIG_USE_GLIBC=y
CONFIG_USE_STRIP=y
CONFIG_VMDK_IMAGES=y
CONFIG_ZSTD_OPTIMIZE_O3=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vooon, 2025-01-11:
Friendly ping @chunkeey

chunkeey, 2025-01-12:
> Perhaps that's related to opkg -> apk change, as there no changes on the package itself.

Perhaps, if someone really, really wants to know what's going on, he/she can to run a bisect... Though, I don't
think this would be necessary. @aparcar  

Since you fortunately added a fix with your bug report:

> ```diff
> diff --git a/package/firmware/intel-microcode/Makefile b/package/firmware/intel-microcode/Makefile
> index bdd8ae73d8..951a7d6f74 100644
> --- a/package/firmware/intel-microcode/Makefile
> +++ [...]

In the context of the error message you posted:

> mkdir: cannot create directory '/mnt/LEDE/openwrt/build_dir/target-x86_64_glibc/intel-microcode-3.20240531.1/intel-ucode-ipkg': File exists

This makes sense since the "-p" option does: " no error if existing, make parent directories as needed,[...]" (man page). 
which addresses the mkdir error.

I think your are like 99.999...% there. Can you please roll this patch into a PR? Reason being, you wrote it and commits needs proper attribution that only you can really give.

(From what I can tell, the buildbots will not have this issue, as they start out from zero...  running `make clean` should also do the trick... but at the cost of additional build time for the next `make`)
 
Cheers,
Christian

GrumpyMeow, 2025-01-15:
Running this 2 consecutive times does not give any error:
` make package/intel-microcode/clean && make package/intel-microcode/compile`