openwrt/openwrt #17582: 5G Wi-Fi missing after upgrading to 24.10-rc5

ErnestoNCarrea, 2025-01-11:
### Describe the bug

This is a Xiaomi AIoT AC2350 working pretty well with 23.05, but after upgrading to 24.10-rc4 or rc5 5 GHz Wi-Fi is gone. 2.4 GHz still works. Luci says "Device is not active".

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ath79/generic

### Device

Xiaomi AIoT AC2350

### Image kind

Official downloaded image

### Steps to reproduce

Upgrade firmware to 24.10-rc4 or 24.10-rc5.

### Actual behaviour

5G wireless dissapears. Luci 

### Expected behaviour

5G wireless stays active.

### Additional info

 -----------------------------------------------------
 OpenWrt 23.05.3, r23809-234f1a2efa
 -----------------------------------------------------
root@router:~# iwinfo
phy0-ap0  ESSID: "MySSID_5G"
          Access Point: 88:C3:XX:XX:XX:24
          Mode: Master  Channel: 149 (5.745 GHz)  HT Mode: VHT80
          Center Channel 1: 155 2: unknown
          Tx-Power: 30 dBm  Link Quality: 51/70
          Signal: -59 dBm  Noise: -104 dBm
          Bit Rate: 693.3 MBit/s
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 168C:0046 168C:CAFE [Qualcomm Atheros QCA9984]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "MySSID"
          Access Point: 88:C3:XX:XX:XX:25
          Mode: Master  Channel: 6 (2.437 GHz)  HT Mode: HT20
          Center Channel 1: 6 2: unknown
          Tx-Power: 13 dBm  Link Quality: 54/70
          Signal: -56 dBm  Noise: -95 dBm
          Bit Rate: 79.9 MBit/s
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [Qualcomm Atheros QCA9560]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

 -----------------------------------------------------
 OpenWrt 24.10.0-rc5, r28304-6dacba30a7
 -----------------------------------------------------
root@router:~# iwinfo 
phy1-ap0  ESSID: "MySSID"
          Access Point: 88:C3:XX:XX:XX:25
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 13 dBm  Link Quality: 62/70
          Signal: -48 dBm  Noise: -95 dBm
          Bit Rate: 115.5 MBit/s
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [Qualcomm Atheros QCA9560]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-11:
Move out /etc/config/wireless and run `wifi config` -> this comes as detected by default on clean install, maybe DTS device locations changed in the meantime.

castillofrancodamian, 2025-01-12:
@brada4 It is probably one more device that has the same problem reported in https://github.com/openwrt/openwrt/issues/14541.

brada4, 2025-01-12:
Ive got one that needed adjustment for DTS path changes, with same symptoms, not that it is universal, but at least worth trying.

ErnestoNCarrea, 2025-01-13:
> Move out /etc/config/wireless and run `wifi config` -> this comes as detected by default on clean install, maybe DTS device locations changed in the meantime.

After doing that, radio0 does not appear in the configuration file.

castillofrancodamian, 2025-01-13:
> > Move out /etc/config/wireless and run `wifi config` -> this comes as detected by default on clean install, maybe DTS device locations changed in the meantime.
> 
> After doing that, radio0 does not appear in the configuration file.

Then you probably have the same problem reported in my previous comment.

rapcore2, 2025-02-21:
I have the same issue on: OpenWrt 24.10.0 r28427-6df0e3d02a

pasqdnik, 2025-02-23:
Same problem here.

brada4, 2025-02-23:
Looks like `owut check -V 24.10-SNAPSHOT`has the fix.

castillofrancodamian, 2025-02-23:
@brada4 What is the commit that fixed this issue? Because I can't find any related to this.

cujanovic, 2025-02-24:
@castillofrancodamian maybe this one?

https://github.com/openwrt/openwrt/commit/c840629046348d72ff7609433e5e228abb737797

sh4tteredd, 2025-02-25:
> [@castillofrancodamian](https://github.com/castillofrancodamian) maybe this one?
> 
> [c840629](https://github.com/openwrt/openwrt/commit/c840629046348d72ff7609433e5e228abb737797)

is there a way to install a build that has this commit?

cujanovic, 2025-02-26:
> > [@castillofrancodamian](https://github.com/castillofrancodamian) maybe this one?
> > [c840629](https://github.com/openwrt/openwrt/commit/c840629046348d72ff7609433e5e228abb737797)
> 
> is there a way to install a build that has this commit?

I assume that it is included in the SNAPSHOT build r28906-[d6977ab33a](https://github.com/openwrt/openwrt/commit/d6977ab33a2b0296837a81b9c92ce8380cc7d39c) - https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ath79%2Fgeneric&id=xiaomi_aiot-ac2350

sh4tteredd, 2025-02-26:
> I assume that it is included in the SNAPSHOT build r28906-[d6977ab33a](https://github.com/openwrt/openwrt/commit/d6977ab33a2b0296837a81b9c92ce8380cc7d39c) - [...]

tried yesterday, not working sadly :/

mmikolajczykm, 2025-03-03:
Same here. I've tried latest snapshot release (r28926-9a7192c08e) without luck.

cujanovic, 2025-03-12:
This also doesn't work, stable and snapshot:
https://github.com/openwrt/openwrt/issues/14541#issuecomment-2644911526

cujanovic, 2025-03-19:
did someone try this: https://github.com/openwrt/openwrt/issues/17175#issuecomment-2692490767

brada4, 2025-03-20:
@DragonBluep seems last chain of patches "convert to nvmem format" lost something

cujanovic, 2025-03-21:
> > did someone try this: [#17175 (comment)](https://github.com/openwrt/openwrt/issues/17175#issuecomment-2692490767)
> 
> I've just tried it and it works! just downloaded https://github.com/openwrt/openwrt/blob/openwrt-23.05/target/linux/ath79/generic/base-files/etc/hotplug.d/firmware/11-ath10k-caldata and replaced /etc/hotplug.d/firmware/11-ath10k-caldata with it

Thanks, I just did it and yes it works 🎉

cujanovic, 2025-04-05:
It seems like this is a better fix:
Download [board-2.bin](https://github.com/eisaev/r2350/blob/main/fw/board-2.bin) from https://github.com/eisaev/r2350/ and replace file in: `/lib/firmware/ath10k/QCA9984/hw1.0`  then reboot.

ParronS3, 2025-04-19:
Can confirm cujanovic trick works well. 
SSHed into the router CD=> /lib/firmware/ath10k/QCA9984/hw1.0, deleted board-2.bin. Wgetted the one linked by him and reboot. Working like a charm. Thanks

cujanovic, 2025-04-22:
btw this pull request should fix everything: https://github.com/openwrt/firmware_qca-wireless/pull/83

gargullia, 2025-05-17:
@cujanovic Confirmed, it totally works

cujanovic, 2025-06-26:
Still an issue with 24.10.2 version :(

rapcore2, 2025-07-12:
Confirmed not working in OpenWrt 24.10.2 r28739-d9340319c6 / LuCI openwrt-24.10 branch 25.186.44806~9ceef01

mmikolajczykm, 2025-10-05:
seems to be fixed in 24.10.3!

mmikolajczykm, 2025-10-06:
https://openwrt.org/toh/xiaomi/aiot_router_ac2350 needs to be updated. 5G works now.

hauke, 2025-10-19:
5 GHz wifi is working on Xiaomi AIoT AC2350 again since c42f05bf36c5708696cbd5d7abe59cb1d5765c1a and 664424aaeb4a7b74123cc94c536e42fa925cf8c3. backported to 24.10 as 31802948974c5cb67206923f7373f7542fefa0ab and 3db4d8f23ade6bd46a62723533add42d10b66f0a