openwrt/openwrt #17585: kmod-rtl8xxxu driver - failed to parse EFuse

George-RG, 2025-01-11:
### Describe the bug

I am trying to connect an external usb wifi antenna to my router. It is a tplink antenna using a Realtek chip. All I know is that it is identified from lsusb with the following identifier `2357:0108 Realtek 802.11n NIC`. I used my Linux computer to determine the correct driver to use and luckily I found the same one in the OpenWrt packages list `rtl8xxxu`. After installing I get dmesg reports of identifying the device and trying to use the correct driver but it fails with the following report.

```
[  302.591912] usb 1-1: new high-speed USB device number 4 using xhci-mtk
[  302.784601] usb 1-1: This Realtek USB WiFi dongle (0x2357:0x0108) is untested!
[  302.799105] usb 1-1: Please report results to Jes.Sorensen@gmail.com
[  302.905396] usb 1-1: Fatal - failed to parse EFuse
[  302.915029] rtl8xxxu: probe of 1-1:1.0 failed with error -22
```

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ramips/mt7621

### Device

D-Link DIR-2660 A1

### Image kind

Official downloaded image

### Steps to reproduce

Install the kmod-rtl8xxxu and plug the usb wifi antena

### Actual behaviour

Fails to use the antenna with the errors described above

### Expected behaviour

Recognizing and using the external antenna

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2025-01-14:
Probably the USB device is not fully supported by the driver. This looks like a bug in upstream Linux.

Please try Openwrt 24.10 and report back if it works.

George-RG, 2025-01-15:
Thanks for the reply.

I just tried it with version 24.10.0-rc5 but unfortunately, I got exactly the same error.

George-RG, 2025-01-15:
So today I also tried installing OpenWRT on a rpi4 I had. Same version as the router, and the driver/antenna works perfectly. No idea how that's possible

arnold0kang, 2025-01-16:
have you resolve this ?  I have same probleam. 
`[  764.580934] usb 1-2.2: Fatal - failed to parse EFuse
[  764.590909] rtl8xxxu: probe of 1-2.2:1.0 failed with error -22
[  764.602794] usbcore: registered new interface driver rtl8xxxu
`

use lastest openwrt , clone from master

George-RG, 2025-01-21:
No, Still no answer.  I even had another USB adapter officially supported by the driver. Same issue.

hauke, 2025-01-22:
I think you should report the problem upstream.

Maybe the USB Wifi card has problems with MIPS, but mt7621 is also little endian.
Maybe there is a bug in the mt7621 USB controller driver.

George-RG, 2025-01-22:
How can i report it there?

kzyapkov, 2025-12-29:
Experiencing the same issue with a Mercusys MW300UH. I've tried Openwrt versions 23.05.6, 24.10.5 and 25.12.0-rc1 with linux versions 5.15.189 thru 6.12.62 on an ASUS RT-AX53U (ramips/mt7621). 

On Openwrt, any of the tried versions

```
[   80.568609] usb 1-2: new high-speed USB device number 3 using xhci-mtk
[   80.722258] usb 1-2: This Realtek USB WiFi dongle (0x2c4e:0x0104) is untested!
[   80.729675] usb 1-2: Please report results to Jes.Sorensen@gmail.com
[   80.849130] usb 1-2: Fatal - failed to parse EFuse
[   80.853975] rtl8xxxu 1-2:1.0: probe with driver rtl8xxxu failed with error -22
```

but works fine on my laptop with the `rtl8xxxu` driver:

```
[88448.847690] usb 3-2: RTL8192EU rev B (SMIC) romver 0, 2T2R, TX queues 3, WiFi=1, BT=0, GPS=0, HI PA=0
[88448.847693] usb 3-2: RTL8192EU MAC: 38:6b:1c:a6:72:90
[88448.847695] usb 3-2: rtl8xxxu: Loading firmware rtlwifi/rtl8192eu_nic.bin
[88448.847879] usb 3-2: Firmware revision 35.7 (signature 0x92e1)
```

kzyapkov, 2025-12-30:
Follow-up: turns out the MW300UH (RTL8192EU) works just fine with OpenWRT on a TP-Link A7 (ath79/generic) with linux 5.15.189 and above, this only reproduces for me on the ramips/mt7621 target.