openwrt/openwrt #17592: attended sysupgrade does not show RC5 release, only 24.10.RC2

peterb99, 2025-01-13:
### Describe the bug

attended sysupgrade does not show RC5 release, only 24.10.RC2

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

 24.10.RC2

### OpenWrt target/subtarget

mediatek/filogic

### Device

openwrt one

### Image kind

Official downloaded image

### Steps to reproduce

start attended sysupgrade

### Actual behaviour

only shows 
[image]

### Expected behaviour

show RC5 release in list

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

peterb99, 2025-01-13:
complete info:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='24.10.0-rc2'
DISTRIB_REVISION='r28161-ea17e958b9'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 24.10.0-rc2 r28161-ea17e958b9'
DISTRIB_TAINTS=''

aimacintyre, 2025-01-14:
As you have opened another issue on the same topic, please close this duplicate.

peterb99, 2025-01-14:
Opened new issue as this one did not recognise the release info and was tagged as invalid