openwrt/openwrt #17595: attended sysupgrade does not show RC5 release, only 24.10.0-rc2

peterb99, 2025-01-13:
### Describe the bug

attended sysupgrade does not show RC5 release as option, only 24.10.0-rc2

Config info:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='24.10.0-rc2'
DISTRIB_REVISION='r28161-ea17e958b9'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 24.10.0-rc2 r28161-ea17e958b9'
DISTRIB_TAINTS=''

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

mediatek/filogic

### Device

openwrt one

### Image kind

Official downloaded image

### Steps to reproduce

Select attended sysupgrade from system menu

### Actual behaviour

[image]

### Expected behaviour

See an option to select 24.10.0-rc5 release

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

McGiverGim, 2025-01-14:
The attended-sysupgrade has not shown any of the RC versions for me neither. I don't know if it only works for stable ones and snapshots.
As workaround you can use `owut`, it works without problem.

efahl, 2025-01-15:
> As workaround you can use `owut`, it works without problem.

Also, Firmware Selector and `auc` use a different strategy from the LuCI app, so they work, too.  If you're coming from 23.05, this should show you the latest RC (or release, when that happens):
```
$ auc -b 24.10
```

(LuCI app should start working as soon as @aparcar merges my PR and restarts the ASU server.)

aparcar, 2025-01-18:
Should be fixed via https://github.com/openwrt/asu/commit/9fcf6b2ef1e18d1102ffa51a0570300ecd257617 can you please give it a try?

efahl, 2025-01-19:
Crashes when `upcoming_version` isn't present in `.versions.json`, that's why I used `get` in https://github.com/openwrt/asu/pull/1153

closed 2025-01-19 by aparcar