openwrt/openwrt #17599: Essential downgrade in wifi perfomance for TP-Link_TL-WA801NDv5 (MediaTek MT7628NN)

icegood, 2025-01-13:
### Describe the bug

Wifi is reachable only if device is close enough to a notebook only (<2 meters)

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ramips/mt76x

### Device

TP-Link TL-WA801ND v5

### Image kind

Official downloaded image

### Steps to reproduce

Try to connect to wifi once AP is properly installed.

### Actual behaviour

Accessible only if the device is near enough

### Expected behaviour

Accessible if the device is in another room 

### Additional info

Works well on 23.07:

root@owrt_TP-Link_TL-WA801NDv5:~#  echo $DISTRIB_REVISION
r24106-10cc5fcd00
root@owrt_TP-Link_TL-WA801NDv5:~# echo  $DISTRIB_RELEASE
23.05.5
root@owrt_TP-Link_TL-WA801NDv5:~# echo  $DISTRIB_TARGET
ramips/mt76x

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2025-01-14:
Probably the eeprom is not correctly loaded. Please post the output of dmesg from the device.

@DragonBluep  Do you have an idea?

icegood, 2025-01-15:
> Probably the eeprom is not correctly loaded. Please post the output of dmesg from the device.
Sure:
[fout.txt](https://github.com/user-attachments/files/18428527/fout.txt)

You said eeprom? Hmmm... Meanwhile, I'm trying to investigate changes in 
drivers/net/wireless/mediatek/mt76/mt7603
of kernel...

DragonBluep, 2025-01-16:
> > Probably the eeprom is not correctly loaded. Please post the output of dmesg from the device.
> > Sure:
> > [dmesg output](https://github.com/user-attachments/files/18428527/fout.txt)
> > [logtread output](https://github.com/user-attachments/files/18428588/logtread.txt)
> 
> You said eeprom? Hmmm... Meanwhile, I'm trying to investigate changes in drivers/net/wireless/mediatek/mt76/mt7603 of kernel...

Have you tried the snapshot version? Any output via `hexdump -C /dev/mtd6`?

icegood, 2025-01-16:
yes, tried snapshot as well. The same. Will add mtd6 a bit later

icegood, 2025-01-16:
`root@owrt_TP-Link_TL-WA801NDv5:~# hexdump -C /dev/mtd6
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
0000f100  b0 be 76 bb bf 66 ff ff  ff ff ff ff ff ff ff ff  |..v..f..........|
0000f110  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
0000f200  7f e8 4e 05 00 00 00 00  ff ff ff ff ff ff ff ff  |..N.............|
0000f210  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
0000f300  45 55 52 00 ff ff ff ff  ff ff ff ff ff ff ff ff  |EUR.............|
0000f310  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
0000f500  38 30 30 30 36 34 31 43  32 32 46 44 41 33 33 32  |8000641C22FDA332|
0000f510  41 36 41 37 41 36 41 46  42 34 38 35 33 41 32 36  |A6A7A6AFB4853A26|
0000f520  31 41 44 42 32 34 44 46  00 ff ff ff ff ff ff ff  |1ADB24DF........|
0000f530  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00020000  28 76 00 02 00 0c 43 e1  76 88 00 00 00 00 00 00  |(v....C.v.......|
00020010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00020020  00 00 00 00 20 00 00 00  00 0c 43 e1 76 29 00 0c  |.... .....C.v)..|
00020030  43 e1 76 2a 22 34 00 20  ff ff 00 01 00 00 00 00  |C.v*"4. ........|
00020040  00 00 22 00 00 00 00 00  30 00 00 00 00 00 00 00  |..".....0.......|
00020050  00 00 00 94 40 b8 c0 ca  24 c0 c0 c0 40 ca 24 c0  |....@...$...@.$.|
00020060  c0 c0 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00020070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000200a0  ca ca ca ca ca ca c0 ca  ca ca ca ca ca ca 00 00  |................|
000200b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000200c0  00 00 00 00 00 00 00 00  00 1a 22 2a 31 35 01 35  |.........."*15.5|
000200d0  39 40 46 4d 7f 7f 7f 00  00 00 00 00 00 00 00 00  |9@FM............|
000200e0  11 1d 11 1d 1c 35 1c 35  1e 35 1e 35 17 19 17 19  |.....5.5.5.5....|
000200f0  02 00 00 00 b7 80 9f 88  0a 00 00 00 00 00 00 00  |................|
00020100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00020120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 77 00  |..............w.|
00020130  11 1d 11 1d 15 7f 15 7f  17 7f 17 7f 10 3b 10 3b  |.............;.;|
00020140  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00020200  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00020400  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
0002fe00  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00030000`

DragonBluep, 2025-01-17:
> [mtd6.dump.txt](https://github.com/user-attachments/files/18445184/mtd6.dump.txt)
> 
> Have you noticed that it works well on 23.07? Wouldn't firmware damage both versions?

I don't have this signal strength issue on my device. And I think the recent device tree NVMEM eeprom conversion commits are correct. I noticed that you also reported https://github.com/openwrt/openwrt/issues/17542 and found you removed NVMEM driver on your repo https://github.com/icegood/openwrt/commit/13002de8911d8800844eb328028904dbfd52082b#diff-5cf154a6666e18e5bf55e892a9dbe14dfb58972d437ee4d419ddd46fe44d069aL141-L142. This driver is necessary to load the eeprom. So, are you using an official build?

icegood, 2025-01-17:
my bug is created against the official build. I don't create anything my custom one unless it should be specified. 
Besides, you see in logs that firmware was loaded

[   31.833809] mt76_wmac 10300000.wmac: ASIC revision: 76280001
[   32.092930] mt76_wmac 10300000.wmac: Firmware Version: 20151201
[   32.098982] mt76_wmac 10300000.wmac: Build Time: 20151201183641
[   32.140018] mt76_wmac 10300000.wmac: firmware init done

icegood, 2025-01-17:
Of course, my final goal is to work against my custom build. In this case I just downloaded 
official one to check if my issue is reproducible. Once I found it reproducible, I decided to raise a bug

icegood, 2025-01-18:
[eeprom.txt](https://github.com/user-attachments/files/18463614/eeprom.txt)
here you are. Will check patch later

icegood, 2025-01-18:
It seems to be exactly part of mtd6...

DragonBluep, 2025-01-18:
> It seems to be exactly part of mtd6...

Yes. That means the EEPROM has been loaded correctly on your device.

> ### Describe the bug
> Wifi is reachable only if device is close enough to a notebook only (<2 meters)

That's strange. I did not find any suspicious changes in the mt76 driver. If the EEPROM is correct, I think the signal strength should be consistent with OpenWrt v23. Is your device having a weak signal or just a low throughput?

icegood, 2026-02-13:
no such behaviour in latest openwrt 25.12