openwrt/openwrt #17601: Spurious resets on HPE 1920 (realtek)

svanheule, 2025-01-14:
### Describe the bug

As reported by @grobian, certain HPE 1920 devices (JG926A) are prone spurious reboots

### OpenWrt version

r28533-b511a513f63c

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

realtek/rtl838x

### Device

HPE 1920

### Image kind

Self-built image

### Steps to reproduce

Let the device running until it spontaneously reboots.

### Actual behaviour

Device does not spontaneously reboot.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

svanheule, 2025-01-14:
> It seems to happen once every few days, switches are used (traffic is flowing) not ginormous loads, but some ports constant, all the time (camera inputs). I had this happening with console switched on, but it did not show me anything but:
> 
> ```
> flow control rx/tx
> [266369.171672] switch: port 17(lan17) entered blocking state
> [266369.190064] switch: port 17(lan17) entered forwarding state
> w
>  08:48:18 up 3 days,  4:01,  0 user,  load average: 0.16, 0.18, 0.16
> USER     TTY        LOGIN@   IDLE   JCPU   PCPU WHAT
> root@switch-1.cheops.mgmt:~# ^@
> System is starting...
> Press Ctrl+D to access BASIC-BOOTWARE MENU
> Booting Normal Extend BootWare
> The Extend BootWare is self-decompressing......................Done!
> 
> ****************************************************************************
> *                                                                          *
> *     HPE 1920-24G-PoE+ (370W) Switch JG926A BootWare, Version 1.17        *
> *                                                                          *
> ****************************************************************************
> ```
> 
> So the `^@` likely is the some init or so, I'd presume if the watchdog would timeout it would leave a message?

The watchdog on these devices is a very simple one, as the SoC watchdog has been disabled. The active watchdog does not provide a warning when it is about to reset the device. So this looks does look like a watchdog reset.

I can't imagine the device being so busy it can't kick the watchdog a few times per second, so I'm not sure what's causing this yet.

grobian, 2025-01-14:
I also saw a ZyXEL GS1900-8 do the same spontaneous reboot :(

grobian, 2025-01-14:
> What builds were you running before these reboots appeared?
23.05.5

I've reverted the JG926A back to 23.05.5 because it's my main switch and it takes a few minutes before it come back in service.  So I'm waiting now, if it resets again, I'll swap it out for a JG925A I've got as a spare here, and see what happens with that device.

Neustradamus, 2025-01-14:
Thanks to @grobian for this test!

grobian, 2025-01-16:
I'm thinking, could these resets perhaps be caused by reasons as mentioned in the debounce commits such as https://github.com/openwrt/openwrt/commit/68bf4844a1cbc9f404f6e93b70a2657e74f1dce9?

Fact that my GS1900-8 did the same, makes me think, perhaps a similar thing happens on the HPE (not that I can find a reset button on the outside of the devices), which would explain why 23.05.5 doesn't seem to exhibit the problem.  Just thinking out loud here.

svanheule, 2025-01-16:
In that case the console would print `REBOOT` as it detected a spurious restart key press, and go through a proper shutdown and reboot cycle.

I rather wonder if the watchdog timeouts would happen more frequently with higher load, or something similar.

janh, 2025-01-18:
My JG924A running 24.10.0-rc5 randomly rebooted today. So it seems like this issue may be unrelated to the recent changes?

svanheule, 2025-01-18:
@janh thanks for the report! I seemed to remember other reports of random reboots before I started merging the new GPIO expander driver, but I couldn't find them.

In case this is related to the external watchdog not being serviced in time, leaving the sys_led peripheral enabled (which by default toggles the output (system LED) a few times per second) should help prevent this. But it seems this issue is happening rather infrequently, so that's making it a lot harder to reproduce...

grobian, 2025-01-18:
I think so yes, it's something that started at 24.10.0-rc2. (basically the first 24.10.0)
23.05.5 really appears to be stable(r).

I want to try @svanheule's RTL changes (https://github.com/openwrt/openwrt/pull/17613) but before I do I want to make sure 23.05.5 really doesn't exhibit the same problem.  Thanks for confirming another reboot, makes it much less likely my device is broken or something.

As for the system LED, there is none on the HPEs, only a power LED which only bootware/bootloader seems to be able to control.  (A bit like all the other LEDs.)

svanheule, 2025-01-18:
For the system LED, I wasn't referring to a physical LED on these devices, but rather the HW peripheral on the SoC intended to control a system LED if there is one. This peripheral is muxed on the SoC's GPIO0 line.

The external watchdog has a [pinctrl property](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/realtek/dts/rtl83xx_hpe_1920.dtsi;h=c7b601311751c78d3630ff559d877f1551c6ff09;hb=HEAD#l16) to disable the sys_led peripheral (inside the SoC) and allow that pin to be used as GPIO0. That way the kernel can periodically toggle that line to keep the watchdog happy.

If the pinctrl property were to be removed, the sys_led peripheral would remain muxed on the pin going to the external watchdog. Since this device doesn't use the sys_led peripheral, I assume the bootloader also doesn't change it's settings. In which case it will default to toggling at something like 4Hz. This should also be sufficient to keep the watchdog happy, but sort of beats the purpose of having a watchdog...

janh, 2025-01-18:
If I remember correctly, the watchdog is the only reliable way to reboot these devices, though. So just configuring the system LED to blink to satisfy the watchdog is probably not an option.

Also, the issue really seems to be happening only infrequently. I started using 24.10 since the first RC, and this is the first time I noticed a reboot (after 11 days of uptime).

grobian, 2025-01-19:
Nor sure this helps, but I see the same spurious reboots on ZyXEL GS1900-8.  Also running 24.10.0.  Right before the reset nothing spectacular or different was happening, I saw the same on the HPE, no special condition or anything that would explain the reset.
[image]
[image]
[image]
[image]

trenchcoatjedi, 2025-01-19:
> I have the same switch and I'm able to test as well

I pulled my JG926A out of storage this weekend and installed 24.10.0-rc5 on it. Booting the kernel over tftp, sysupgrade and setting the OpenWRT image as default worked perfectly. 

Are there any specific tests you need run?

svanheule, 2025-01-20:
Updated the ticket title and description.

> Edit: In storage I also have a HPE 1910-16G JE005A that I could test if there is forthcoming support for this device.

Only the fast ethernet devices of the [HPE 1910 series](https://svanheule.net/switches/hpe_1910_series) could potentially be supported, but I wouldn't wait for that to happen. The HPE 1910 gigabit devices use a different SoC.

grobian, 2025-01-27:
[image]
Update: I've started an idle JG925A on 24.10.0-rc5 which has been running sofar without a reboot.  The GS1900-8 which has constant (but lowish) traffic has rebooted a lot over the past week, and I've upgraded it to rc6 (at the dotted line). 

I consider somehow getting some traffic load on the JG925A to see if that makes a difference.  I also want to run a second GS1900-8 without load, to see if reboots then too.  What is less convenient is that all sofar seem to not reboot.  The JG926A (23.05.5) is there only to confirm it is more stable.

grobian, 2025-02-05:
For some reason the rebooting has stopped with the later -rc's.  I've upgraded to the 24.10.0 release across the board, and will keep an eye on it, but it looks to me like whatever it was is now gone.

grobian, 2025-03-02:
I have not experienced any involuntary reboots on 24.10.0 release sofar, so I think it's time to close this issue.

svanheule, 2025-03-05:
Good that it's gone, but maybe not great that we don't know what was causing it. We can re-open later if needed.

closed 2025-03-05 by svanheule

grobian, 2025-05-03:
I just experienced this out of the blue reboot using `OpenWrt 24.10.1` again on the same JG926A, it was up 2.56 weeks.  From whatever was sent to the logserver, nothing that seems to indicate anything out of the ordinary.

grobian, 2025-06-25:
I took another JG926A switch with bootware 1.22 and I have the impression that makes the difference.

plappermaul, 2025-07-21:
Please check here: https://github.com/openwrt/openwrt/pull/19468

grobian, 2025-10-16:
Final and last comment here.  Running the patches from @plappermaul on a snapshot build for almost 11 weeks without any itch whatsoever.