openwrt/openwrt #17603: dhcpv6.sh offlink flag Always set 1，cause pppoe v6 network unreachable

yangdaxia6666, 2025-01-14:
### Describe the bug

in package/network/ipv6/odhcp6c/files/dhcpv6.script line 83
proto_add_ipv6_address "$addr" "$mask" "$preferred" "$valid" 1   offlink flag always set 1

cause interface-ip.c  interface_ip_set_enabled function add a null route，cause pppoe v6 network unreachable
				/*
				 * In case off link is specifed as address property
				 * add null-route to avoid routing loops
				 */

### OpenWrt version

22.03

### OpenWrt release

all

### OpenWrt target/subtarget

generic

### Device

generic

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2025-01-16:
@yangdaxia6666: Can you check with a more recent OpenWrt?

brada4, 2025-01-16:
Code has not changed since?
https://github.com/openwrt/openwrt/blob/0a7c8ed9d94930ba062c71df79f63c06eeab4543/package/network/ipv6/odhcp6c/files/dhcpv6.script#L83

Can you state exactly what does not work, since having a line in a script by itself is not any kind of problem.

yangdaxia6666, 2025-01-16:
When offlink is set to 1, an unreachable route is added in interface-ip.c code。As a result  the default route does not take effect.