openwrt/openwrt #17611: [ipq40xx] Low throughput after upgrading to 24.10.0-rc5

jamesdoherty, 2025-01-15:
### Describe the bug

After upgrading from OpenWrt 23.05.5 to 24.10.0-rc5 (fresh install), I am no longer able to achieve full download speed of 325Mbps over a 1gbit wired LAN connection. Running  24.10.0-rc5, a speedtest will struggle to reach 200Mbps and trends down for the duration of the speedtest. Downgrading OpenWrt to 23.05.5 fixes the issue and a speedtest confirms 325Mbps again.

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ASUS RT-AC58U

### Image kind

Official downloaded image

### Steps to reproduce

- Install 23.05.5 (fresh install) and run a speedtest from laptop connected via wired LAN (1gbit)
- Upgrade to 24.10.0-rc5 (removing existing config) and run speedtest again
- Downgrade to 23.05.5  and run speedtest again

### Actual behaviour

When running 24.10.0-rc5, speedtests are much slower (often 100Mbps slower)

### Expected behaviour

Speedtest results should be very similar between OpenWrt releases

### Additional info

I have been following https://github.com/openwrt/openwrt/issues/15445 as this sounds like the same issue (packet steering). I have been able to confirm that when running a speedtest on 24.10.0-rc5, all the load goes onto a single CPU, while on 23.05.5 the load is shared across CPU's.

The default RPS settings differ between releases, potentially due to changes in the packet steering scripts last year: https://github.com/openwrt/openwrt/issues/15445#issuecomment-2109351247

root@OpenWrt:~# . /etc/openwrt_release && echo $DISTRIB_RELEASE
23.05.5
root@OpenWrt:~# cat /sys/class/net/eth0/queues/rx-*/rps_cpus
0
0
0
0

root@OpenWrt:~# . /etc/openwrt_release && echo $DISTRIB_RELEASE
24.10.0-rc5
root@OpenWrt:~# cat /sys/class/net/eth0/queues/rx-*/rps_cpus
1
1
1
1

Setting the RPS settings back to 0 resolves the issue.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

nbd168, 2025-01-15:
Should be fixed in 67a4aeef6c63e5836bd6b82faf6cc15934869efb, please test

nbd168, 2025-01-16:
@jamesdoherty, thanks for testing. I looked into this more closely on an ipq40xx device and found a better way to steer packets.
The hardware has 4 rx rings for handling incoming packets and the script's behavior of keeping all eth0 napi tasks on the same CPU was hurting performance.
Please try this updated version: https://nbd.name/packet-steering.uc
It improves performance by assigning each rx napi task to its own CPU and steers rx processing from there to the next CPU to distribute the load.

When routing ~930 Mbit/s through the device, I get 36% idle with steering enabled, and 13% idle with steering disabled.

jamesdoherty, 2025-01-17:
@nbd168 I have tested with https://nbd.name/packet-steering.uc and it has fixed the performance issue I was encountering with slow packet steering. Many thanks!

htop with 24.10.0-rc5 packet steering and speedtest running
[image]

htop with https://nbd.name/packet-steering.uc packet steering and speedtest running
[image]

Regarding PR https://github.com/openwrt/openwrt/pull/17612 from @aimacintyre to allow packet steering to be disabled, is that better dealt with separately or should we keep this issue open until there is an agreed fix?

closed 2025-01-17 by openwrt-bot