openwrt/openwrt #17614: TP-Link Archer C2600 (ipq806x) stops responding (24.10-SNAPSHOT)

funnydog, 2025-01-15:
### Describe the bug

After a few minutes the router stops responding on the LAN and on the WIFI.

The led lights still blink but the router isn't reachable any more.

Sometimes after a while it recovers and in one of those occasions I could grab the OOPS stored in /sys/fs/pstore (attached).

[console-ramoops-0.txt](https://github.com/user-attachments/files/18424913/console-ramoops-0.txt)
[dmesg-ramoops-0.txt](https://github.com/user-attachments/files/18424918/dmesg-ramoops-0.txt)
[dmesg-ramoops-1.txt](https://github.com/user-attachments/files/18424917/dmesg-ramoops-1.txt)

### OpenWrt version

r28337-7d597f8709

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Archer C2600

### Image kind

Self-built image

### Steps to reproduce

Power on the router, let it connect to the the internet over PPPoE and just use the WIFI for a few minutes.

### Actual behaviour

The router stops responding on the LAN or wireless interfaces, producing OOPS and sometimes recovering after a few minutes. 

Network traffic is halted but apparently the WAN interface is still connected since the WAN led keeps blinking.

### Expected behaviour

No network interruption and no OOOPSes.

### Additional info

This router is rock solid with OpenWRT 23.05.

### Diffconfig

```text
CONFIG_TARGET_ipq806x=y
CONFIG_TARGET_ipq806x_generic=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ipq806x_generic_DEVICE_netgear_r7800=y
CONFIG_TARGET_DEVICE_ipq806x_generic_DEVICE_tplink_c2600=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
# CONFIG_BINUTILS_USE_VERSION_2_42 is not set
CONFIG_BINUTILS_USE_VERSION_2_43=y
CONFIG_BINUTILS_VERSION="2.43.1"
CONFIG_BINUTILS_VERSION_2_43=y
CONFIG_CCACHE=y
CONFIG_EXPERIMENTAL=y
# CONFIG_FEED_luci is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_GCC_USE_VERSION_13 is not set
CONFIG_GCC_USE_VERSION_14=y
CONFIG_GCC_VERSION="14.2.0"
CONFIG_GCC_VERSION_14=y
# CONFIG_GDB is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_WERROR is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
# CONFIG_PACKAGE_dnsmasq is not set
CONFIG_PACKAGE_dnsmasq-full=y
CONFIG_PACKAGE_dnsmasq_full_dhcp=y
CONFIG_PACKAGE_dnsmasq_full_dnssec=y
CONFIG_PACKAGE_dnsmasq_full_noid=y
CONFIG_PACKAGE_dnsmasq_full_tftp=y
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_iftop=y
CONFIG_PACKAGE_ip-bridge=y
CONFIG_PACKAGE_ip-full=y
# CONFIG_PACKAGE_jansson is not set
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-lib-textsearch=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-nathelper=y
# CONFIG_PACKAGE_kmod-nft-fib is not set
CONFIG_PACKAGE_kmod-sched=y
CONFIG_PACKAGE_kmod-sched-cake=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-sched-drr=y
CONFIG_PACKAGE_kmod-sched-flower=y
CONFIG_PACKAGE_kmod-sched-fq-pie=y
CONFIG_PACKAGE_kmod-sched-mqprio=y
CONFIG_PACKAGE_kmod-sched-mqprio-common=y
CONFIG_PACKAGE_kmod-sched-pie=y
CONFIG_PACKAGE_kmod-sched-prio=y
CONFIG_PACKAGE_kmod-sched-red=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libgmp=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnettle=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_mini_snmpd=y
# CONFIG_PACKAGE_nftables-json is not set
CONFIG_PACKAGE_nftables-nojson=y
CONFIG_PACKAGE_tc-full=y
CONFIG_PACKAGE_tcpdump-mini=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_zlib=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_USE_FS_ACL_ATTR is not set
# CONFIG_PACKAGE_dnsmasq_full_auth is not set
# CONFIG_PACKAGE_dnsmasq_full_conntrack is not set
# CONFIG_PACKAGE_dnsmasq_full_dhcpv6 is not set
# CONFIG_PACKAGE_dnsmasq_full_nftset is not set
# CONFIG_PACKAGE_jansson is not set
# CONFIG_PACKAGE_kmod-nf-conntrack-netlink is not set
# CONFIG_PACKAGE_libnetfilter-conntrack is not set
# CONFIG_PACKAGE_libnfnetlink is not set
# CONFIG_PACKAGE_nftables-json is not set
CONFIG_PACKAGE_nftables-nojson=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

funnydog, 2025-01-15:
Some more information which may help to debug the issue.

I started the debugger and looked at the location that dereference the NULL  pointer:

```
(gdb) l *qca8k_mib_autocast_handler+0xb0
0xc078fcb4 is in qca8k_mib_autocast_handler (drivers/net/dsa/qca/qca8k-8xxx.c:1660).
1655            for (i = 0; i < priv->info->mib_count; i++) {
1656                    mib = &ar8327_mib[i];
1657
1658                    /* First 3 mib are present in the skb head */
1659                    if (i < 3) {
1660                            mib_eth_data->data[i] = get_unaligned_le32(mib_ethhdr->data + i);
1661                            continue;
1662                    }
1663
1664                    /* Some mib are 64 bit wide */
```

mib_eth_data is assigned at line:

`1644	mib_eth_data = &priv->mib_eth_data;`

Could it be that priv->mib_eth_data is NULL?

funnydog, 2025-01-16:
I've applied this patch to see if it changes anything:

```
--- a/drivers/net/dsa/qca/qca8k-8xxx.c
+++ b/drivers/net/dsa/qca/qca8k-8xxx.c
@@ -1643,6 +1643,12 @@ static void qca8k_mib_autocast_handler(s
        mib_ethhdr = (struct mib_ethhdr *)skb_mac_header(skb);
        mib_eth_data = &priv->mib_eth_data;

+       /* just skip if there is no port to parse */
+       if (!refcount_read(&mib_eth_data->port_parsed))
+       {
+               return;
+       }
+
        /* The switch autocast every port. Ignore other packet and
         * parse only the requested one.
         */
```

As expected it cures the effects, i.e. the Oops, but unfortunately not the cause: the switch gets confused and disconnects the wired interfaces.

The wireless interfaces continue to work without issues, I can connect to the router via ssh and look around: no Oopses but some warnings from the ath10k wireless driver, and a message about getting a DHCP request from an interface without any IP assigned (switch.3). 

No internet though, because the wired interfaces are not working anymore.

So I suspect I'm doing something in the configuration that the switch doesn't like. 

There are two internet connections,  the primary on the WAN port tagged with vid 835 and the backup on the LAN4 port untagged. When pushing the front button the user can switch from the primary to the other and back.

The problem occurs when switching from the primary to the backup internet connection either manually or with the script.

The script to switch the internet lines (/etc/rc.button/lights_toggle) is the following: 

```
#!/bin/sh
[ "${ACTION}" = "released" ] || exit 0

ifdown wan

IFACE=$(uci get network.wan.device)
if [ "$IFACE" = "switch.3" ]; then
    # Fiber
    uci set network.wan.device="switch.835"
    echo 'none' > /sys/class/leds/white:general/trigger
else
    # Radio bridge
    uci set network.wan.device="switch.3"
    echo 'default-on' > /sys/class/leds/white:general/trigger
fi

ifup wan

return 0
```

The network configuration is the following:

```
config device
        option name 'switch'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'wan'

config bridge-vlan
        option device 'switch'
        option vlan '1'
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:t'

config bridge-vlan
        option device 'switch'
        option vlan '2'
        list ports 'lan3:t'

config bridge-vlan
        option device 'switch'
        option vlan '3'
        list ports 'lan4:u*'

config bridge-vlan
        option device 'switch'
        option vlan '835'
        list ports 'wan:t'

config interface 'lan'
        option device 'switch.1'
        option proto 'static'
        option ipaddr '192.168.1.255'
        option netmask '255.255.255.0'
        option ip6assign '64'

config interface 'guest'
        option device 'switch.2'
        option proto 'static'
        option ipaddr '192.168.2.254'
        option netmask '255.255.255.0'
        option ip6assign '64'

config interface 'wan'
        option device 'switch.835'
        option proto 'pppoe'
        option username 'REDACTED'
        option password 'REDACTED'
        option ipv6 '0'
        option peerdns '0'
        list dns '8.8.8.8'
        list dns '8.8.4.4'
```

aimacintyre, 2025-01-21:
Particularly as it seems to be related to the switch, just a reminder that ipq806x has migrated from swconfig to DSA as of 24.10.  Given you appear to have configuration baked into your build have you confirmed that your switch configuration is DSA valid?  (in other words, if you recreate your switch configuration via 24.10 LuCI does it match your current configuration?)

You might wish to try installing a downloaded 24.10-rc5 image and setting that up from scratch to get a good switch configuration.

BIGFAAT, 2025-02-08:
The issue seems still to exist and happen randomly after a few days running the latest RC. As for me, both my c2600 become unreachable but still passes L2 towards other devices (sometimes with issues but they recover after a few min).
Beside both devices switches crapping out it seems also that 5GHZ wifi needs several restarts (software or cold) to work at all, just until it crashes again together with the switch.

I have now updated to the stable release. At least the wifi issue stills exist.

funnydog, 2025-02-09:
I'm glad I'm not the only one seeing this problem, maybe someone hit by the bug will be able to fix it.

Unfortunately I can reproduce the hangs only on the field and not in a testing environment, so my options to analyze it further are limited.

I think the problem is connected to VLAN tagging and the bug could be triggered by some kind of packet coming from the wan interface.

Unfortunately when I tried to capture the packets from the wan interface tcpdump didn't work and exited with an error (couldn't listen on the interface).

However I think that the issue could be either:

  1. a real packet is misidentified as a switch autocast packet (could it be a layering violation between VLAN tags and the switch header?)
  2. a memory corruption somewhere in the dsa code that triggers the emission of autocast packets.

So I went back to 23.05 because it is rock solid.

Besides, when testing 24.10 for the little while the router could stay alive, I saw a hit of about 50% of the routing bandwidth.

This is more than I can live with.

I think the DSA code should be fine tuned and the lurking fixed before 24.10 can be used on this router.

TeutonJon78, 2025-04-12:
I started running into this as well on my C2600 (i have two, one my house and one at my parents). Everything seemed solid on 23.05.x.

I've been running 24.10.0 since it came out on mine, and it has been fine, but at my parent's house, they've been experiencing similar issues to you where it just stops responding. Sometimes it won't even respond over LAN, and sometimes it's just the internet and wifi that are down. I'm using some complicated stuff like IOT and Guest firewall zones along with wireguard, but not using any VLAN stuff. 

And I checked my log, and it's showing some weird stuff on my end. Both seem to be having one of the problems mentioned here.

C2600 1: no real issues, but seeing a lot of this in the kernel log:
`[168905.478741] qca8k 37000000.mdio-mii:10 lan2: failed (err=-22) to del object (id=2)

C2600 2: issues where it's dropping off the WAN and wifi, and sometimes LAN. I only have one log from the most recent incident where LAN was still up, and the only odd things are:
system log: nothing weird, but I start seeing this around when the problem starts:
`Sat Apr 12 13:35:06 2025 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)

kernel log:
`[153326.354359]` ath10k_pci 0001:01:00.0: mac flush vdev 1 drop 0 queues 0x2 ar->paused: 0x0  arvif->paused: 0x0

TeutonJon78, 2025-04-17:
The only other thought I had was if maybe something on the network was behaving badly like Chromecasts used to do with flooding networks with requests and crashing the router. The only majorly different devices on the problematic C2600 are 2 Nest G4 thermostats and a Ting power line montior (both on the guest network). @funnydog and @BIGFAAT , do those devices match your setup at all?

funnydog, 2025-04-17:
No, not in my case, it was an office network with two computers and a printer.

It's unfortunate but 24.10 crippled these routers with a premature DSA migration.

funnydog, 2025-04-18:
I can't check the wireless power now because I reverted back to 23.05.

But I don't think it was wireless related because after applying the patch above, the wireless interface was still working even after losing the wired network.

I think your WIFI interface stops working because of a previous kernel Ooops, crippling the kernel.

Infact this whole thing is a Linux kernel bug and must be solved with a kernel patch and not by changing some parameters here and there and hoping for the best.

Yes it was a very simple setup and the only "strange" thing was the WAN connection through VLAN VID 835.

The Ooops was triggered randomly not by something on the network but by some packet received on the WAN cable (probably VLAN tagged with some VID != 835).

If you look at the debug traces above (that I could extract by luck), the switch executed the qca8k_mib_autocast_handler without any reason.

Infact it should be triggered only after a call qca8k_get_ethtool_stats_eth() and that function wasn't called at all otherwise it would have assigned a non NULL value to mib_eth_data->data.

When qca8k_mib_autocast_handler() is called it dereferences mib_eth_data->data and if it's NULL that causes a kernel Ooops.

The patch I published above avoids to dereference the NULL pointer in qca8k_mib_autocast_handler() and therefore avoid the Ooops, but it's only curing some (fatal) side effects and not the cause of the problem.

Looking at the DSA code there is a call to qca8k_mib_autocast_handler() that is suspect because no other DSA driver checks every packet and calls the autocast handler that way. What in the code is preventing to misclassify a raw received packet as an autocast mib packet and call qca8k_mib_autocast_handler()?

I admit I didn't dig deeper and maybe this isn't even the right place to discuss the matter, I think a proper bugreport to netdev would be more successful.

But to do a proper report I should have had at least a reproduceable case and I couldn't find because:

1) I couldn't dump the raw packets on the WAN cable (tcpdump didn't work because it didn't recognize the DSA interface).

2) I couldn't reproduce the problem in a controlled testing environment (switch with VLAN 835 + simulated PPPOE) but only on the field where I can't afford any downtime.

Hey but do not dispair, at least 23.05 works!!! 😃