openwrt/openwrt #17621: GL-MT6000 installation has `/overlay` way bigger than RAM size

adelton, 2025-01-15:
### Describe the bug

The GL.iNet GL-MT6000 has 8 GB of NAND / eMMC and 1 GB of RAM (https://openwrt.org/toh/gl.inet/gl-mt6000). So there is enough space to make for example `/var` persistent and not a symlink to `tmp`, to store things like container images, database data, or logs.

Unfortunately, merely changing the `/var` symlink to be a directory does not work for example for podman
```
Error: configure storage: 'overlay' is not supported over overlayfs, a mount_program is required: backing file system is unsupported for this graph driver
```

In addition, per https://openwrt.org/docs/techref/sysupgrade the rootfs_data partition content is stored in `/tmp/sysupgrade.tgz` to be restored after sysupgrade, which means the fate of that content depends on surviving in RAM. Also https://forum.banana-pi.org/t/bpi-r64-tip-how-to-make-full-use-of-emmc-in-openwrt/12651/6 confirms that having overlay larger than RAM doesn’t make sense (the device has 1 GB of RAM).

In a great discussion in https://github.com/openwrt/openwrt/issues/9550 and specifically in https://github.com/openwrt/openwrt/issues/9550#issuecomment-1085285509 @dangowrt explained that ideally kernel+rootfs+rootfs_data should be small and separate devices used for additional storage needs. The GL-MT6000 image doesn't seem to respect that guidance.

There seem to be two mechanisms described in https://openwrt.org/docs/techref/flash.layout how to restrict the size of rootfs_data and thus free space for custom partition: `fw_setenv rootfs_data_max <VALUE>` and `uvol`. But I assume neither of them should be used on GL-MT6000. The device has
```
# lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
loop0          7:0    0  7.2G  0 loop /overlay
mmcblk0      179:0    0  7.3G  0 disk 
├─mmcblk0p1  179:1    0  512K  0 part 
├─mmcblk0p2  179:2    0    2M  0 part 
├─mmcblk0p3  179:3    0    2M  0 part 
├─mmcblk0p4  179:4    0    2M  0 part 
├─mmcblk0p5  179:5    0    2M  0 part 
├─mmcblk0p6  179:6    0   32M  0 part 
└─mmcblk0p7  179:7    0  7.2G  0 part /rom
mmcblk0boot0 179:8    0    4M  1 disk 
mmcblk0boot1 179:16   0    4M  1 disk 
```

Checking the `rootfs_data_max`
```
# fw_printenv -n rootfs_data_max
Warning: Bad CRC, using default environment
## Error: "rootfs_data_max" not defined
```
scares me with that bad CRC message. And with `uvol` I see
```
Configuring uvol.
No backend available. (tried: LVM UBI)
To setup devices with block storage install 'autopart'.
//usr/lib/opkg/info/uvol.postinst: /etc/uci-defaults/90-uvol-init: line 8: arithmetic syntax error
```
while installing the package and then
```
# uvol list
No backend available. (tried: LVM UBI)
To setup devices with block storage install 'autopart'.
```

Alright, so if my device is of the category devices with block storage, let's go the `autopart` route. When I install that package, it seems to pull in multiple dependencies including `lvm2` but the package itself has
```
# opkg files autopart
Package autopart (0.9-r3) is installed on root and has the following files:
/etc/uci-defaults/30-autopart
# cat /etc/uci-defaults/30-autopart
cat: can't open '/etc/uci-defaults/30-autopart': No such file or directory
```

Searching for quite some time I don't see a canonical documentation about how autopart should be configured to achieve making the `/overlay` smaller. I don't need LVM, a single ext4 partition would be enough.

There is of course a more general question -- shouldn't OpenWrt on GL-MT6000 make the `/overlay` reasonably small to make space for the extra partition. Or even create the extra partition automatically?

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

1. Look at the default sizing of OpenWrt installation on GL-MT6000.
2. Wonder why the `/overlay` is so big when it cannot survive sysupgrade, while wasting space that could be used for another partition.
3. Get the idea of only making the `/overlay` partition 500 MB and using the rest for separate partition which would get mounted at `/var` or `/data`.
4. Search around for a working guidance.

### Actual behaviour

Not sure how to proceed.

### Expected behaviour

Either by default the `/overlay` should not be bigger than 1 GB (the RAM size) and ideally smaller.

Or https://openwrt.org/docs/techref/flash.layout has a link to a document which describes what should be done with block devices, if that document is not applicable.

### Additional info

I face the same issue / dilemma on x86_64 VM where the layout is
```
# lsblk
NAME     MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0      7:0    0 491.2M  0 loop /overlay
vda      254:0    0   512M  0 disk 
├─vda1   254:1    0    16M  0 part /boot
│                                  /boot
├─vda2   254:2    0 495.7M  0 part /rom
└─vda128 259:0    0   239K  0 part 
```
so the use-case is not strictly limited to GL-MT6000.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dangowrt, 2025-01-15:
In order for `autopart` to work there needs to be free space available. OpenWrt support for the GL-MT6000 was provided by GL-iNet themselves and the partitioning scheme on the device intends the whole remaining space on the eMMC to be used as rootfs+rootfs_data, which hence results in a huge overlayfs which isn't that useful after all.

The best fix is to resize the `mmcblk0p7` partition to something around 1 GiB, so after a subsequent sysupgrade (which should include `autopart`) the now remaining free space will be allocated for use with `uvol`.

adelton, 2025-01-15:
Thanks for the superquick reply.

What is the best mechanism of resizing that `mmcblk0p7`? The cleanest way would be to boot something from ramdisk so that the `/dev/mmcblk0` wouldn't be mounted ... do we have something like that?

Alternatively I can likely force the partition end with `fdisk` and quickly do sysupgrade while having the filesystem inconsistently reach beyond the end of the partition, with fingers crossed.

BTW, I see
```
# fdisk /dev/mmcblk0

Welcome to fdisk (util-linux 2.40.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

The backup GPT table is not on the end of the device. This problem will be corrected by write.
This disk is currently in use - repartitioning is probably a bad idea.
It's recommended to umount all file systems, and swapoff all swap
partitions on this disk.
```
Those
```
mmcblk0boot0 179:8    0    4M  1 disk 
mmcblk0boot1 179:16   0    4M  1 disk 
```
that `lsblk` shows, is that some content on that `/dev/mmcblk0` that `fdisk`'s GPT write could destroy, or is it safe to let `fdisk` make that backup GPT?

dangowrt, 2025-01-15:
`mmcblk0bootX` are special *hardware* partitions used to store ARM TrustedFirmware-A `bl2`. Ie. you shouldn't touch them and their existence is not related to (GPT, *software*) partitions `mmcblk0pX`.

Doing fdisk on the life system just before sysupgrade can (and should) work fine, especially if the rootfs_data overlay is mostly empty.

adelton, 2025-01-16:
I confirm that remount to read-only + resize in `fdisk` + `sysupgrade` gave me a working setup with a smaller `/overlay`
```
# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 5.8M      5.8M         0 100% /rom
tmpfs                   493.6M      1.3M    492.3M   0% /tmp
/dev/loop0              742.4M     62.6M    679.8M   8% /overlay
overlayfs:/overlay      742.4M     62.6M    679.8M   8% /
tmpfs                   512.0K         0    512.0K   0% /dev
```

Specifically the steps I've done were
```
# opkg update
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/targets/mediatek/filogic/packages/Packages.gz
[...]
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
# opkg install fdisk
Installing fdisk (2.40.2-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/aarch64_cortex-a53/base/fdisk_2.40.2-r1_aarch64_cortex-a53.ipk
Installing libsmartcols1 (2.40.2-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/aarch64_cortex-a53/base/libsmartcols1_2.40.2-r1_aarch64_cortex-a53.ipk
Installing libfdisk1 (2.40.2-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/aarch64_cortex-a53/base/libfdisk1_2.40.2-r1_aarch64_cortex-a53.ipk
Installing terminfo (6.4-r2) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/aarch64_cortex-a53/base/terminfo_6.4-r2_aarch64_cortex-a53.ipk
Installing libncurses6 (6.4-r2) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/aarch64_cortex-a53/base/libncurses6_6.4-r2_aarch64_cortex-a53.ipk
Configuring terminfo.
Configuring libsmartcols1.
Configuring libfdisk1.
Configuring libncurses6.
Configuring fdisk.

# mount -o remount,ro /overlay
# mount -o remount,ro /
# wget -P /tmp https://downloads.openwrt.org/releases/24.10.0-rc5/targets/mediatek/filogic/openwrt-24.10.0-rc5-mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin
Downloading 'https://downloads.openwrt.org/releases/24.10.0-rc5/targets/mediatek/filogic/openwrt-24.10.0-rc5-mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin'
Connecting to 151.101.130.132:443
Writing to 'openwrt-24.10.0-rc5-mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin'
openwrt-24.10.0-rc5- 100% |*******************************|  9790k  0:00:00 ETA
Download completed (10025322 bytes)

# fdisk /dev/mmcblk0
Welcome to fdisk (util-linux 2.40.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

The backup GPT table is not on the end of the device. This problem will be corrected by write.
This disk is currently in use - repartitioning is probably a bad idea.
It's recommended to umount all file systems, and swapoff all swap
partitions on this disk.

Command (m for help): p

Disk /dev/mmcblk0: 7.28 GiB, 7818182656 bytes, 15269888 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 2BD17853-102B-4500-AA1A-8A21D4D7984D

Device         Start      End  Sectors  Size Type
/dev/mmcblk0p1  8192     9215     1024  512K Linux filesystem
/dev/mmcblk0p2  9216    13311     4096    2M Linux filesystem
/dev/mmcblk0p3 13312    17407     4096    2M Linux filesystem
/dev/mmcblk0p4 17408    21503     4096    2M Linux filesystem
/dev/mmcblk0p5 21504    25599     4096    2M Linux filesystem
/dev/mmcblk0p6 25600    91135    65536   32M Linux filesystem
/dev/mmcblk0p7 91136 15269887 15178752  7.2G Linux filesystem

Command (m for help): e
Partition number (1-7, default 7): 7

New <size>{K,M,G,T,P} in bytes or <size>S in sectors (default 7.2G): 750M

Partition 7 has been resized.

Command (m for help): w
The partition table has been altered.
Syncing disks.

# sysupgrade /tmp/openwrt-24.10.0-rc5-mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin 
verifying sysupgrade tar file integrity
Thu Jan 16 09:46:38 UTC 2025 upgrade: Saving config files...
Thu Jan 16 09:46:38 UTC 2025 upgrade: Commencing upgrade. Closing all shell sessions.
Command failed: Connection failed
```

adelton, 2025-01-16:
The next step was to use `fdisk` to create a `/dev/mmcblk0p8` partition to fill the rest of the space, run
```
# mkfs.ext4 -m 0 -L data /dev/mmcblk0p8
```
to create a filesystem on it, add
```
config 'mount'
	option	target	'/data'
	option	label	'data'
	option	enabled	'1'
```
to `/etc/config/fstab`, and `opkg install block-mount`. After a reboot, I now have
```
# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 5.8M      5.8M         0 100% /rom
tmpfs                   493.6M    224.0K    493.4M   0% /tmp
/dev/loop0              742.4M     62.9M    679.5M   8% /overlay
overlayfs:/overlay      742.4M     62.9M    679.5M   8% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mmcblk0p8            6.3G      1.7M      6.3G   0% /data
```

I was surprised I did not need to explicitly install `kmod-fs-ext4`.

Would you be open to accepting a PR which would add `block-mount` into `Device/glinet_gl-mt6000`'s `DEVICE_PACKAGES`? Running `git grep block-mount` I can see that there already are `*.mk`s that do that for some boards. That would make it easier for this more useful configuration to survive `sysupgrades` without additional package handling setup/steps.

dangowrt, 2025-01-16:
In a way I don't think `block-mount` (or `blockd`) should be added to any device by default. It is not needed to use the device with the default configuration and it can easily be installed by anyone who needs it, and they may use `owut` or `luci-app-attendedsysupgrade` to retain that package accross upgrades.
People may obviously also generate their own custom images including that using the [firmware selector](https://firmware-selector.openwrt.org/).
I'd rather consider removing that package from the default selection on devices which do not strictly need to to boot and run the default configuration.