openwrt/openwrt #17624: GL-MT6000 WAN random crash - only recoverable by rebooting

PanosChtz, 2025-01-16:
### Describe the bug

Router randomly loses WAN connectivity (happens every few days or so).
Restarting wan, wan6 or eth1 interfaces has no effect, can only recover connection by rebooting router.
Very problematic when not on-site and running services. Can only wait until watchcat kicks in and restarts.
Logs when WAN connection is lost as follows:

```
Sat Jan 11 20:34:09 2025 kern.warn kernel: [3025370.094611] Ignoring NSS change in VHT Operating Mode Notification from ac:50:de:1e:41:e9 with invalid nss 2
Sat Jan 11 20:34:09 2025 kern.warn kernel: [3190778.177275] ------------[ cut here ]------------
Sat Jan 11 20:34:09 2025 kern.info kernel: [3190778.192040] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 5 timed out
Sat Jan 11 20:34:09 2025 kern.warn kernel: [3190778.199186] WARNING: CPU: 3 PID: 0 at 0xffffffc0086c6464
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204656] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath10k_pci ath10k_core ath xt_state xt_conntrack xt_CT wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 iptable_mangle iptable_filter ipt_REJECT ip_tables chacha_neon cfg80211 xt_time xt_tcpudp xt_physdev xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables usbhid slhc poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libchacha hid_generic compat br_netfilter crypto_safexcel hid evdev input_core ledtrig_usbport ifb ip6_udp_tunnel udp_tunnel veth xxhash_generic sha1_generic seqiv
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204800]  md5 des_generic libdes blake2b_generic authencesn authenc uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd dwc3 ohci_platform ohci_hcd ahci libahci fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug exfat btrfs xor xor_neon raid6_pq libcrc32c usbcore usb_common aquantia crc_ccitt
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204859] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 5.15.167 #0
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204865] Hardware name: GL.iNet GL-MT6000 (DT)
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204868] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204872] pc : 0xffffffc0086c6464
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204876] lr : 0xffffffc0086c6464
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204878] sp : ffffffc008b83db0
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204880] x29: ffffffc008b83db0 x28: 0000000000000140 x27: 00000000ffffffff
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204887] x26: 0000000000000000 x25: 0000000000000003 x24: ffffff80003c54c0
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204893] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008a46000
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204899] x20: ffffff80003c5000 x19: 0000000000000005 x18: ffffffc008a5a2a8
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204905] x17: 353a6361206d6f72 x16: 66206e6f69746163 x15: 00000000000004ad
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204912] x14: 000000000000018f x13: ffffffc008b83ae8 x12: ffffffc008ab22a8
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204918] x11: 712074696d736e61 x10: ffffffc008ab22a8 x9 : 0000000000000000
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204924] x8 : ffffffc008a5a258 x7 : ffffffc008a5a2a8 x6 : 0000000000000001
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204931] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204937] x2 : ffffff803fdc3090 x1 : ffffffc037395000 x0 : 000000000000003f
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204944] Call trace:
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204946]  0xffffffc0086c6464
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204949]  0xffffffc0080a5b50
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204951]  0xffffffc0080a63c8
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204954]  0xffffffc0080a6918
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204956]  0xffffffc00803b04c
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204959]  0xffffffc00803b404
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204961]  0xffffffc00803b5d0
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204964]  0xffffffc008089454
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204966]  0xffffffc0080100e0
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204969]  0xffffffc00801584c
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204971]  0xffffffc00801609c
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204974]  0xffffffc008835d60
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204976]  0xffffffc008836514
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204979]  0xffffffc008011374
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204981]  0xffffffc008836af4
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204984]  0xffffffc008069350
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204986]  0xffffffc008069560
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204988]  0xffffffc00801ebe0
Sat Jan 11 20:34:09 2025 kern.debug kernel: [3190778.204991]  0xffffffc00883f2cc
Sat Jan 11 20:34:09 2025 kern.warn kernel: [3190778.204994] ---[ end trace 14be6397fbc80364 ]---
Sat Jan 11 20:34:09 2025 kern.err kernel: [3190778.209783] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
Sat Jan 11 20:34:09 2025 kern.info kernel: [3190778.348131] mtk_soc_eth 15100000.ethernet eth0: Link is Down
Sat Jan 11 20:34:09 2025 kern.info kernel: [3190778.353990] mtk_soc_eth 15100000.ethernet eth1: Link is Down
Sat Jan 11 20:34:10 2025 kern.err kernel: [3190779.017390] mtk_soc_eth 15100000.ethernet: warm reset failed
Sat Jan 11 20:34:10 2025 kern.info kernel: [3190779.038344] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
Sat Jan 11 20:34:10 2025 kern.info kernel: [3190779.046927] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
Sat Jan 11 20:34:10 2025 kern.info kernel: [3190779.336086] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=62)
Sat Jan 11 20:34:10 2025 kern.info kernel: [3190779.346509] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/sgmii link mode
Sat Jan 11 20:34:10 2025 daemon.notice netifd: Network device 'eth1' link is down
Sat Jan 11 20:34:10 2025 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sat Jan 11 20:34:10 2025 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Sat Jan 11 20:34:10 2025 daemon.notice netifd: wan6 (3716): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan6" } (Permission denied)
Sat Jan 11 20:34:10 2025 daemon.notice netifd: wan (3718): udhcpc: received SIGTERM
Sat Jan 11 20:34:10 2025 daemon.notice netifd: wan (3718): udhcpc: unicasting a release of x.x.x.223 to x.x.x.1
Sat Jan 11 20:34:10 2025 daemon.notice netifd: wan (3718): udhcpc: sending release
Sat Jan 11 20:34:10 2025 daemon.notice netifd: wan (3718): udhcpc: entering released state
Sat Jan 11 20:34:10 2025 daemon.notice netifd: wan (3718): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Sat Jan 11 20:34:10 2025 daemon.notice netifd: Interface 'wan' is now down
Sat Jan 11 20:34:10 2025 daemon.info avahi-daemon[2219]: Withdrawing address record for x.x.x.223 on eth1.
[... 26 lines cut ...]
```

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

castiel652, 2025-01-16:
Try upgrading to 24.10

mrkiko, 2025-01-16:
Hi!

is this problem reproducible on snapshot as well?
In case you are going to try that out, I would recommend to not keep settings.

PanosChtz, 2025-01-16:
@castiel652 @mrkiko I would rather stay in a stable release unless I'm sure that these releases actually solve the problem. I am running a number of services on my network and I want to minimize experimentation and/or downtime.

I see there are similar issues opened,
https://github.com/openwrt/openwrt/issues/13262
https://github.com/openwrt/openwrt/issues/13122
This PR https://github.com/openwrt/openwrt/pull/16419 was supposed to solve it but apparently the issue still occurs.

everything411, 2025-04-05:
same problem on jcg q30 pro, mt7981
```
[91140.606887] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 5020 ms
[91140.616639] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[91142.306076] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[91142.847608] mtk_soc_eth 15100000.ethernet: warm reset failed
[91142.867713] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[91142.876155] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[91143.726085] NOHZ tick-stop error: local softirq work is pending, handler #08!!!
[91144.765813] NOHZ tick-stop error: local softirq work is pending, handler #08!!!
[91145.805549] NOHZ tick-stop error: local softirq work is pending, handler #08!!!
[91147.885014] NOHZ tick-stop error: local softirq work is pending, handler #08!!!
[91148.844745] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 5950 ms
[91148.854490] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[91148.924746] NOHZ tick-stop error: local softirq work is pending, handler #08!!!
[91149.033806] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[91149.575854] mtk_soc_eth 15100000.ethernet: warm reset failed
```

kevin-david, 2025-04-15:
I'm having the same, or at least a similar, problem with my MikroTik RouterBOARD 750Gr3 (MediaTek MT7621). Seems like high throughput (e.g. video streaming) exacerbates the issue.

This seems to be a regression with 24.10 personally, 23.05 was rock solid (6+ months stability)

Found this old thread: https://forum.openwrt.org/t/issue-with-mtk-soc-eth-on-mt7621/78263/15

Log below

```
Sat Apr  5 01:11:09 2025 kern.crit kernel: [1712697.789945] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 7 timed out 5340 ms
Sat Apr  5 01:11:14 2025 kern.crit kernel: [1712702.830084] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 7 timed out 10380 ms
Sat Apr  5 01:11:15 2025 kern.crit kernel: [1712703.789889] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 5400 ms
Sat Apr  5 01:11:20 2025 kern.crit kernel: [1712708.829848] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 10440 ms
Sat Apr  5 01:11:25 2025 kern.crit kernel: [1712713.789823] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 15400 ms
Sat Apr  5 01:11:28 2025 kern.crit kernel: [1712716.749785] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 5440 ms
Sat Apr  5 01:11:30 2025 kern.crit kernel: [1712718.830119] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 20440 ms
Sat Apr  5 01:11:33 2025 kern.crit kernel: [1712721.709750] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 10400 ms
Sat Apr  5 01:11:35 2025 kern.crit kernel: [1712723.789845] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 25400 ms
Sat Apr  5 01:11:38 2025 kern.crit kernel: [1712726.749708] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 15440 ms
Sat Apr  5 01:11:40 2025 kern.crit kernel: [1712728.829775] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 30440 ms
Sat Apr  5 01:11:43 2025 kern.crit kernel: [1712731.709683] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 20400 ms
Sat Apr  5 01:11:45 2025 kern.crit kernel: [1712733.789658] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 35400 ms
Sat Apr  5 01:11:48 2025 kern.crit kernel: [1712736.749640] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 25440 ms
Sat Apr  5 01:11:50 2025 kern.crit kernel: [1712738.829789] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 40440 ms
Sat Apr  5 01:11:53 2025 kern.crit kernel: [1712741.709604] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 30400 ms
Sat Apr  5 01:11:55 2025 kern.crit kernel: [1712743.789585] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 45400 ms
Sat Apr  5 01:11:58 2025 kern.crit kernel: [1712746.749556] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 35440 ms
Sat Apr  5 01:12:00 2025 kern.crit kernel: [1712748.829552] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 50440 ms
Sat Apr  5 01:12:03 2025 kern.crit kernel: [1712751.709519] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 40400 ms
Sat Apr  5 01:12:05 2025 kern.crit kernel: [1712753.789508] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 55400 ms
Sat Apr  5 01:12:08 2025 kern.crit kernel: [1712756.749494] mtk_soc_eth 1e100000.ethernet wan: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 45440 ms
Sat Apr  5 01:12:10 2025 kern.crit kernel: [1712758.829466] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 5 timed out 60440 ms
````

starworld99, 2026-05-16:
The same issue MTK + ImmortalWrt 24.10.5 r33805-7c4e882aaf6f + ASIX adapter ,disabled all offloading but no luck for me, the wan crashed all the time. Disconnect ASIX adapter and use wifi instead work for me now.
Sun May 17 03:13:11 2026 kern.crit kernel: [434935.453202] mtk_soc_eth 15100000.ethernet eth1: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 410290 ms
Sun May 17 03:13:13 2026 kern.crit kernel: [434937.452203] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 415670 ms
Sun May 17 03:13:16 2026 kern.crit kernel: [434940.420719] mtk_soc_eth 15100000.ethernet eth1: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 415260 ms
Sun May 17 03:13:18 2026 kern.crit kernel: [434942.419718] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 420640 ms
Sun May 17 03:13:21 2026 kern.crit kernel: [434945.448208] mtk_soc_eth 15100000.ethernet eth1: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 420290 ms
Sun May 17 03:13:23 2026 kern.crit kernel: [434947.447203] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 425670 ms
Sun May 17 03:13:26 2026 kern.crit kernel: [434950.405728] mtk_soc_eth 15100000.ethernet eth1: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 425250 ms
Sun May 17 03:13:28 2026 kern.crit kernel: [434952.414723] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 430640 ms
Sun May 17 03:13:31 2026 kern.crit kernel: [434955.453201] mtk_soc_eth 15100000.ethernet eth1: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 430300 ms
Sun May 17 03:13:33 2026 kern.crit kernel: [434957.442206] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 435670 ms
Sun May 17 03:13:36 2026 kern.crit kernel: [434960.400733] mtk_soc_eth 15100000.ethernet eth1: NETDEV WATCHDOG: CPU: 3: transmit queue 1 timed out 435250 ms

TerryBarnaby, 2026-10-16:
I think I am seeing this issue as well with OpenWrt 25.12.5 r33051-f5dae5ece4 on a MT6000 router, where the Ethernet interfaces hang.

We recently replaced an old TP TL-WDR3600 router on one of our FTTP (Openreach) lines and since then every 4 days or so the Ethernet stops working on both WAN and LAN* ports. Wifi is still functional. We have an MT6000 on another, more lightly used, Openreach FTTP connection and so far that one has been fine (about 3 weeks).

The router has  the WAN connection connected to a 1 GBit/s Openreach ONT, the LAN1 connection to a Linux box (1 GBits/s) acting as main firewall/web servers etc. and LAN5 connected to a Cisco SPA112 VOIP telephone adapter.
Config is basic, but with a br_global bridge and global interface to allow the pass through of our 8 x IP address range to the linux box and we have Flow offloading set to "Hardware flow offloading".

When the issue occurs we cannot ping the router from the Linux box and when looking at the router Network/Interfaces page our "global" interface (the 8 x IP address bridge) shows "Carrier Absent" and the "wan" interface shows "Connection attempt failed".

Pluging in a Laptop to LAN4 shows no Ethernet lights on the laptop (seems like the router's Ethernet PHY is down). If I restart the "lan" interface, the Laptop's Ethernet port lights come on, but it does not get a DHCP IP address. If I restart the "global" interface the Carrier turns to Present, but it still does not function.

In the kernels log I see messages like:
[242471.674764] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[242471.674788] br-global: port 1(lan1) entered blocking state
[242471.688328] br-global: port 1(lan1) entered forwarding state
[243145.429713] br-global: port 1(lan1) entered disabled state
[243145.429826] mt7530-mdio mdio-bus:1f lan1: Link is Down

In the system log i see messages like:
Fri Oct 16 07:19:57 2026 daemon.notice netifd: Interface 'wan' is now down
Fri Oct 16 07:19:57 2026 daemon.notice netifd: Interface 'wan' is setting up now
Fri Oct 16 07:19:57 2026 daemon.info pppd[14237]: Plugin pppoe.so loaded.
Fri Oct 16 07:19:57 2026 daemon.info pppd[14237]: PPPoE plugin from pppd 2.5.2
Fri Oct 16 07:19:58 2026 daemon.notice pppd[14237]: pppd 2.5.2 started by root, uid 0
Fri Oct 16 07:20:33 2026 daemon.warn pppd[14237]: Timeout waiting for PADO packets
Fri Oct 16 07:20:33 2026 daemon.err pppd[14237]: Unable to complete PPPoE Discovery phase 1
Fri Oct 16 07:20:33 2026 daemon.info pppd[14237]: Exit.

So it looks like an intermittent (race ?) low level driver bug in the Ethernet drivers that affects both Ethernet interfaces (eth0 and eth1).

ifconfig down/up on the eth0 or eth1 interfaces does not clear the issue.
I will enclose the system and kernel logs and a picture of the Network/Interfaces page.

I have had to go back to the TP TL-WDR3600 router for now.

TerryBarnaby, 2026-10-16:
[kernellog.txt](https://github.com/user-attachments/files/33536919/kernellog.txt)
[systemlog.txt](https://github.com/user-attachments/files/33536920/systemlog.txt)
[image]

TerryBarnaby, 2026-10-16:
This may also be of interest, the output of ifconfig showing packets dropped.

[ifconfig.txt](https://github.com/user-attachments/files/33539683/ifconfig.txt)

bulldozer2003, 2026-10-19:
Same hardware (GL-MT6000), same pattern: ethernet dies, WiFi keeps working. But the failure mode is different from the classic transmit queue timeout reported here.

**Setup**
OpenWrt 24.10.8 (r29233-443ec4032a), kernel 6.6.144. Collected live data during the wedge using a capture script that runs from a WiFi laptop over SSH. The data captures dmesg, logread, interface counters, conntrack, and everything else while the ethernet is dead but the router is still reachable.

**The wedge, by the numbers**
- Uptime at capture: 2 days 21 hours
- eth0: carrier=1, speed=2500, UP LOWER_UP
- eth1: carrier=1, speed=1000, UP LOWER_UP
- All DSA ports (lan2-lan5): UP LOWER_UP, attached to br-lan
- br-lan: UP, 24786 RX drops
- WiFi radios (phy0-ap0, phy1-ap0): UP, serving clients normally
- CPU load: 0.00, memory: 19%
- Health checks every 5 minutes show normal load/memory/carrier right up to the capture
- All cron jobs running (host_check, lan-watchdog, health, ddns)
- Neighbor table: 192.168.1.2, 192.168.1.3, 192.168.1.70 are INCOMPLETE or FAILED

Pings from the router: all DEAD. WAN (4.2.2.1, 1.1.1.1) and LAN wired hosts (secondAP, waran, storaje) all DEAD.

**What makes this interesting**
- No mtk_soc_eth NETDEV WATCHDOG messages in dmesg. No "transmit timed out". No "warm reset failed". The classic signature from this issue is absent.
- 17 instances of "br-lan: received packet on lan2 with own address as source address" in the kernel ring buffer, the most recent at uptime +244682s (~2.8 days). That means something on lan2 is sending packets with the router MAC as the source.
- massive flood of "hostapd: IEEE 802.11: Failed to send Action frame" around the time the wedge started (thousands in a few minutes, then tapering off). This is the FT/802.11r action frame failure when a client roams between radios and the kernel cant deliver the frame over the air.
- We had a previous hypothesis about nlbwmon netlink buffer exhaustion (rmem_max was capping SO_RCVBUF at 212992 while nlbwmon requested 524288, running for 2.5 days with ENOBUFS every 30s). That fix was applied: rmem_max=4M, nlbwmon -b 4194304. It held. Zero netlink errors in the new capture. So that theory is ruled out for this occurrence.

**What I think is happening**
The ethernet MAC (or the DSA switch) enters a state where the carrier stays up and the interface looks healthy, but packet forwarding stops. No crash, no warning, no dmesg event. The bridge keeps accumulating RX drops because it is receiving packets from the WiFi side but cannot deliver them to the wired ports. The "own address as source" messages suggest a layer-2 forwarding loop or a DSA bridge forwarding table issue.

This is on 24.10.8, which was supposed to have the warm-reset recovery from #19768. But the recovery doesnt trigger because there is no warm reset event. The MAC just quietly stops working.

I have the full live capture data (dmesg, logread, ip -s link, ethtool counters, conntrack, process list, config, and the on-device persistent log spanning several days before the wedge). Happy to upload anything that would help.