openwrt/openwrt #17625: mac80211 kernel stacktrace on boot

rblumel, 2025-01-16:
### Describe the bug

Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.519315] ------------[ cut here ]------------
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.524102] WARNING: CPU: 0 PID: 1872 at backports-6.12.6/net/mac80211/main.c:272 0x82a2089c [mac80211@dba0a61a+0x8a000]
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.535431] Modules linked in: ath9k(O) ath9k_common(O) pppoe ppp_async nft_fib_inet nf_flow_table_inet ath9k_hw(O) ath10k_pci(O) ath10k_core(O) ath(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) lzo cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcrc32c hwmon crc_ccitt compat(O) ledtrig_usbport sha512_generic seqiv sha3_generic jitterentropy_rng drbg hmac geniv rng cmac crypto_acompress fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug(O) phy_ar7200_usb phy_ar7100_usb usbcore nls_base usb_common crc32c_generic
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.613495] CPU: 0 PID: 1872 Comm: hostapd Tainted: G           O       6.6.69 #0
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.621245] Stack : 00000000 80758470 00000000 82f1eea8 00000000 00000000 00000000 00000000
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.629917]         00000000 00000000 00000000 00000000 00000000 00000001 81aafb10 faaea9d2
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.638593]         81aafba8 00000000 00000000 81aafa58 00000038 80701564 00000000 ffffffea
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.647267]         000000c5 81aafa64 000000c5 807f9720 8071aad0 81aafaf0 00000000 82a2089c
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.655943]         00000009 82f1eea8 00000000 82ff8a00 00000018 803d0bfc 00000000 809b0000
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.664617]         ...
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.667151] Call Trace:
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.667308] [<80701564>] 0x80701564
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.673534] [<82a2089c>] 0x82a2089c [mac80211@dba0a61a+0x8a000]
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.679765] [<803d0bfc>] 0x803d0bfc
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.683482] [<800661e8>] 0x800661e8
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.687098] [<800661f0>] 0x800661f0
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.690707] [<806d7ee4>] 0x806d7ee4
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.694352] [<800849b0>] 0x800849b0
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.697995] [<82a2089c>] 0x82a2089c [mac80211@dba0a61a+0x8a000]
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.704176] [<80084a40>] 0x80084a40
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.707839] [<82a02394>] 0x82a02394 [mac80211@dba0a61a+0x8a000]
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.714156] [<82a2089c>] 0x82a2089c [mac80211@dba0a61a+0x8a000]
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.720519] [<82a20a60>] 0x82a20a60 [mac80211@dba0a61a+0x8a000]
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.726785] [<804bca84>] 0x804bca84
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.730399] [<801925ec>] 0x801925ec
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.734115] [<804bcf9c>] 0x804bcf9c
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.737727] [<80192cd8>] 0x80192cd8
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.741410] [<804bd048>] 0x804bd048
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.745089] [<804b0eb8>] 0x804b0eb8
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.748772] [<805a85e4>] 0x805a85e4
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.752382] [<801ce230>] 0x801ce230
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.756171] [<805ab320>] 0x805ab320
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.759783] [<801bff78>] 0x801bff78
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.763429] [<804ee97c>] 0x804ee97c
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.767144] [<804ef764>] 0x804ef764
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.770829] [<8048af8c>] 0x8048af8c
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.774584] [<802216cc>] 0x802216cc
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.778301] [<801cbaa4>] 0x801cbaa4
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.782054] [<8006d420>] 0x8006d420
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.785745]
Wed Jan 15 20:08:39 2025 kern.warn kernel: [   99.787609] ---[ end trace 0000000000000000 ]---

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer A9 v6

### Image kind

Official downloaded image

### Steps to reproduce

boot device

### Expected behaviour

no stack

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

castiel652, 2025-01-16:
https://github.com/openwrt/openwrt/issues/15959#issuecomment-2259603978

DragonBluep, 2025-04-01:
Fixed by 88234a03bc645d327016cc807c35972fcb6834eb. Please try snapshot version. And this ticket can be closed.

thomascrisan, 2025-04-02:
I am also experiencing this mac80211 kernel stacktrace on boot with the non-CT versions of ath10k.  Before I try the snapshot, will your patch also fix ath10k (non-CT version)?

DragonBluep, 2025-04-13:
> I am also experiencing this mac80211 kernel stacktrace on boot with the non-CT versions of ath10k. Before I try the snapshot, will your patch also fix ath10k (non-CT version)?

No. It's a different issue.

thomascrisan, 2025-04-14:
> > I am also experiencing this mac80211 kernel stacktrace on boot with the non-CT versions of ath10k. Before I try the snapshot, will your patch also fix ath10k (non-CT version)?
>
> No. It's a different issue.

My mistake, I guess it just looks the same.

```
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.116844] ------------[ cut here ]------------
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.121764] WARNING: CPU: 0 PID: 13 at backports-6.12.6/net/mac80211/main.c:408 0x83502a50 [mac80211@7ddd411f+0x8a000]
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.132983] Modules linked in: ath9k(O) ath9k_common(O) pppoe ppp_async nft_fib_inet nf_flow_table_inet ath9k_hw(O) ath10k_pci(O) ath10k_core(O) ath(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat(O) fuse ledtrig_usbport nfsv4 nfsd nfs msdos rpcsec_gss_krb5 auth_rpcgss oid_registry ntfs lockd sunrpc grace dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha1_generic seqiv sha3_generic jitterentropy_rng drbg md5 hmac geniv rng ecb des_generic libdes cts cmac arc4 uas usb_storage uhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.133532]  gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2 dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax phy_ar7200_usb phy_ar7100_usb usbcore nls_base usb_common crc16 tpm crc32c_generic cbc sha256_generic encrypted_keys trusted
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.247252] CPU: 0 PID: 13 Comm: kworker/u2:1 Tainted: G           O       6.6.73 #0
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.255130] Workqueue: events_unbound 0x83400338 [cfg80211@e5f5e465+0x46000]
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.262334] Stack : 00000000 80758350 81887d2c 834501f0 00000000 00000000 00000000 00000000
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.270843]         00000000 00000000 00000000 00000000 00000000 00000001 81887ce8 be8c5264
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.279343]         81887d80 00000000 00000000 81887c10 00000038 807014a4 00000000 ffffffea
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.287844]         0000011c 81887c1c 0000011c 807f9720 8071a9b8 81887cc8 00000000 83502a50
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.296345]         00000009 834501f0 81883c50 61c88647 00000018 803d09c8 00000000 809b0000
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.304845]         ...
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.307328] Call Trace:
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.307409] [<807014a4>] 0x807014a4
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.313448] [<83502a50>] 0x83502a50 [mac80211@7ddd411f+0x8a000]
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.319552] [<803d09c8>] 0x803d09c8
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.323089] [<800661e8>] 0x800661e8
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.326627] [<800661f0>] 0x800661f0
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.330164] [<806d7e24>] 0x806d7e24
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.333768] [<800849b0>] 0x800849b0
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.337346] [<83502a50>] 0x83502a50 [mac80211@7ddd411f+0x8a000]
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.343406] [<80084a40>] 0x80084a40
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.347029] [<83502a50>] 0x83502a50 [mac80211@7ddd411f+0x8a000]
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.353085] [<8357d2e4>] 0x8357d2e4 [mac80211@7ddd411f+0x8a000]
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.359529] [<834003ec>] 0x834003ec [cfg80211@e5f5e465+0x46000]
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.365832] [<8009ed8c>] 0x8009ed8c
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.369501] [<8009f34c>] 0x8009f34c
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.373037] [<8009efb4>] 0x8009efb4
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.376574] [<800a660c>] 0x800a660c
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.380114] [<8009efb4>] 0x8009efb4
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.383650] [<800a660c>] 0x800a660c
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.387189] [<800a6708>] 0x800a6708
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.390726] [<800a660c>] 0x800a660c
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.394264] [<800a660c>] 0x800a660c
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.397802] [<800619f8>] 0x800619f8
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.401349]
Sun Apr 13 19:55:58 2025 kern.warn kernel: [   98.403959] ---[ end trace 0000000000000000 ]---
```

pabl0, 2025-04-22:
Seeing this on eap225-v3 with 24.10.0 every now and then, is it related and does 88234a0 actually affect this behavior? To me it seems that the radio crashes/goes down for few seconds.

```
ue Apr 22 10:42:54 2025 kern.warn kernel: [35596.824198] ath10k_pci 0000:00:00.0: peer-unmap-event: unknown peer id 0
Tue Apr 22 10:42:54 2025 kern.warn kernel: [35596.831200] ath10k_pci 0000:00:00.0: peer-unmap-event: unknown peer id 0
Tue Apr 22 10:42:54 2025 daemon.notice hostapd: Add bss phy0-ap0 on phy phy0
Tue Apr 22 10:42:54 2025 daemon.err hostapd: Could not unlink existing ctrl_iface socket '/var/run/hostapd/': Is a directory
Tue Apr 22 10:42:54 2025 daemon.err hostapd: hostapd_free_hapd_data: Interface phy0-ap0 wasn't started
Tue Apr 22 10:42:54 2025 daemon.notice hostapd: Failed to add new bss phy0-ap0 on phy phy0
Tue Apr 22 10:42:54 2025 daemon.notice hostapd: Restart interface for phy phy0
Tue Apr 22 10:42:54 2025 daemon.notice hostapd: Remove interface 'phy0'
Tue Apr 22 10:42:54 2025 daemon.notice hostapd: phy0-ap1: interface state ENABLED->DISABLED
Tue Apr 22 10:42:54 2025 daemon.notice hostapd: phy0-ap1: AP-DISABLED
Tue Apr 22 10:42:54 2025 daemon.notice hostapd: phy0-ap1: CTRL-EVENT-TERMINATING
Tue Apr 22 10:42:54 2025 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Tue Apr 22 10:42:54 2025 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap1 disabled_11b_rates=0
Tue Apr 22 10:42:54 2025 kern.info kernel: [35596.889391] ath10k_pci 0000:00:00.0: mac flush null vif, drop 0 queues 0xffff
Tue Apr 22 10:42:54 2025 kern.warn kernel: [35596.901604] ath10k_pci 0000:00:00.0: peer-unmap-event: unknown peer id 1
Tue Apr 22 10:42:54 2025 kern.warn kernel: [35596.908664] ath10k_pci 0000:00:00.0: peer-unmap-event: unknown peer id 1
Tue Apr 22 10:42:54 2025 kern.info kernel: [35597.008795] br-lan: port 3(phy0-ap1) entered disabled state
Tue Apr 22 10:42:54 2025 kern.info kernel: [35597.016098] ath10k_pci 0000:00:00.0 phy0-ap1: left allmulticast mode
Tue Apr 22 10:42:54 2025 kern.info kernel: [35597.022694] ath10k_pci 0000:00:00.0 phy0-ap1: left promiscuous mode
Tue Apr 22 10:42:54 2025 kern.info kernel: [35597.029317] br-lan: port 3(phy0-ap1) entered disabled state
Tue Apr 22 10:42:54 2025 daemon.notice netifd: Network device 'phy0-ap1' link is down
Tue Apr 22 10:42:55 2025 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=FI ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 stationary_ap=1 chanlist=100 tx_queue_data2_burst=2.0 #num_global_macaddr=1 #macaddr_base= ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=106 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-2][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] channel=100  interface=phy0-ap0 bssid=28:ee:52:9b:e1:d9 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa
Tue Apr 22 10:42:57 2025 kern.warn kernel: [35599.969256] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Tue Apr 22 10:42:57 2025 kern.warn kernel: [35599.977069] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
Tue Apr 22 10:42:57 2025 kern.info kernel: [35600.026865] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Tue Apr 22 10:42:57 2025 kern.info kernel: [35600.038426] ath10k_pci 0000:00:00.0: wmi print 'free: 114572 iram: 12644 sram: 29508'
Tue Apr 22 10:42:58 2025 kern.info kernel: [35600.363262] ath10k_pci 0000:00:00.0: rts threshold -1
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.368954] ath10k_pci 0000:00:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.380421] ------------[ cut here ]------------
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.385198] WARNING: CPU: 0 PID: 1375 at backports-6.12.6/net/mac80211/main.c:272 0x8292089c [mac80211@2a0f6b08+0x8a000]
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.396487] Modules linked in: ath9k(O) ath9k_common(O) pppoe ppp_async nft_fib_inet nf_flow_table_inet ath9k_hw(O) ath10k_pci(O) ath10k_core(O) ath(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) lzo cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcrc32c hwmon crc_ccitt compat(O) sha512_generic seqiv sha3_generic drbg hmac geniv rng cmac crypto_acompress gpio_button_hotplug(O) crc32c_generic
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.461964] CPU: 0 PID: 1375 Comm: hostapd Tainted: G        W  O       6.6.73 #0
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.469699] Stack : 00000000 80758350 00000000 82d26ea8 00000000 00000000 00000000 00000000
[... cut ...]
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.618985] [<80221310>] 0x80221310
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.622677] [<80488e54>] 0x80488e54
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.626340] [<8048cbd0>] 0x8048cbd0
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.629946] [<80489024>] 0x80489024
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.633611] [<8006d420>] 0x8006d420
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.637252]
Tue Apr 22 10:42:58 2025 kern.warn kernel: [35600.639172] ---[ end trace 0000000000000000 ]---
```

pabl0, 2025-04-22:
Oh, looks like 24.10.1 does indeed fix this issue.