openwrt/openwrt #17631: [ipq806x] Arris TR4400 v2 (RAC2V1A) no network access with 24.10.0-rc5 nor 14jan'25 snapshot

jeremywh7, 2025-01-16:
### Describe the bug

I tried both dhcp and static IP, and both 24.10.0-rc5 and a snapshot from 14jan'25.  No network access after flash (ssh, ping, webGUI, no dhcp IP, et cetera).  The serial boot log from this appears ok [attached in _Add'l info_] from what I can tell.  Recovery in uboot back to 23.05.5 works fine.  I spoke to @Lanchon and also tried a 14nov'24 build (r28073+13-93d005e6bc) and it exhibited the same issue.  He said to @Ansuel as well.  :-)

Note: 23.05 -> 24.10 [converts from swconfig to DSA from 30sep'23](https://github.com/openwrt/openwrt/commit/337e36e0ef988d).

I have left this running 24.10.0-rc5 in case a dev wants me to check/try anything.

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ipq806x/generic

### Device

Arris TR4400 v2

### Image kind

Official downloaded image

### Steps to reproduce

- Go to webGUI firmware flash page to upload 24.10.0-rc5, not saving settings/configs due to swconfig->DSA conversion.
- Forced upgrade required:
[image]
- After the serial output shows boot is complete, check that serial access is working (ifconfig, ps w, et cetera)
- No network access available: ssh, ping, webGUI, no dhcp IP, static IP also doesn't work
- Reboot, break to uboot via serial, boot recovery, access webGUI, flash 23.05.5, everything works again

### Actual behaviour

No network access after upgrade, but serial access still works fine

### Expected behaviour

Working network after upgrade

### Additional info

See attached boot log which goes through these steps:
- Cold boot to break into uboot - (IPQ) prompt and checked `printenv`
- `run boot_recovery` for 23.05.5 recovery [and then I checked that the webGUI was working]
- `reboot` into normal 23.05.5 [squashfw and checked ifconfig, etc.]
- **Flashed 24.10-rc5 from normal webGUI**, without saving settings/configs [due to swconfig -> DSA change]
- Boot log looked fine, but no network access
- `reboot`, break into uboot, and `run boot_recovery` for 23.05.5 recovery again
- **Flashed 14jan'25 snapshot from recovery webGUI**, without saving settings/config
- Boot log looked fine, but no network access again
- `reboot`, break into uboot, and `run boot_recovery` for 23.05.5 recovery again
- **Flashed normal 23.05.5 from recovery webGUI**, without saving settings/config
- Booted up fine, and network/webGUI working normal again

[Arris TR100v2 - 24.10 and snapshot flash fail (no network).txt](https://github.com/user-attachments/files/18443330/Arris.TR100v2.-.24.10.and.snapshot.flash.fail.no.network.txt)

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Lanchon, 2025-01-17:
thanks!

@Ansuel, could this be the result of the migration to DSA?

unfortunately a friend needed my router so i can't test anything, but it will be back sometime in february.

Lanchon, 2025-01-17:
@jeremywh7,

i suggest you flash a modded rc5 so that you try to have wifi access to the device while the ethernet ports are dead. it might be helpful to debug issues. it's simple, you don't need to do a custom build.

steps:
- go [here](https://firmware-selector.openwrt.org/?version=24.10.0-rc5&target=ipq806x/generic&id=arris_tr4400-v2).
- click on "Customize installed packages and/or first boot script".
- in "Script to run on first boot (uci-defaults)", click the gear at the lower right corner.
- uncomment (and customize if desired) the following lines to enable wifi on boot:
```
# wlan_name="OpenWrt"
# wlan_password="12345678"
```
- click on "REQUEST BUILD".
- download and flash the resulting sysupgrade image (wipe settings).

after reboot, if wifi actually works, you should see an AP and be able to connect to it.

jeremywh7, 2025-01-17:
> i suggest you flash a modded rc5 so that you try to have wifi access to the device while the ethernet ports are dead.

👍🏼  That gave me an idea... I compared /etc/config/{network,wireless} of this device versus my primary gateway (ramips/mt7621) and wap (ipq40xx/generic) and noticed a couple significant differences:

RAC2V1A _/etc/config/network_ default:
```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd6a:78b4:743c::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option device 'eth2'
        option proto 'dhcp'

config interface 'wan6'
        option device 'eth2'
        option proto 'dhcpv6'
```

My gateway uses 'wan' instead of 'eth0', but maybe the interface/device nomenclature is not important?  The key diff b/t my other devices is that they have a 'wan' device while the RAC2V1A instead has a 'eth2' device.  Anyway, it is setup like this:
```
config interface 'wan'
        option device 'wan'
        option proto 'dhcp'
[...]
config device
        option name 'wan'
        option ipv6 '0'
```

More significant; both other devices have _eth0_ and _lan[1-4]_ devices defined, which the RAC2V1A **does not** have:
```
config device
        option name 'eth0'
        option ipv6 '0'

config device
        option name 'lan1'
        option ipv6 '0'

config device
        option name 'lan2'
        option ipv6 '0'

config device
        option name 'lan3'
        option ipv6 '0'

config device
        option name 'lan4'
        option ipv6 '0'

```

I merged this last part into the RAC2V1A, along with un-disabling the radios in the wireless config.  No help on the LAN side (as I'm sure I didn't do it right), but - I was able to connect to the wireless and... progress!
[image]
So it indeed looks like the DSA setup needs adjusted?  Hopefully the high load average is just due to my incorrect _/network_ hackery; it did not come down after another 10min.  `top` showed 98-99% idle though.

jeremywh7, 2025-01-17:
I found a default backup of my _ipq40xx/generic_ device to compare the configs (dhcp, firewall, network, wireless).  I figured it might be more similar than my _mt7621_ but didn't really see anything (besides the _wan_ vs _eth2_ device diff).  I changed my lan IP so I can connect the wan upstream, and confirmed I get internet through the wireless - lan is still dead though, and the load average is still hovering around 2.40-2.70.  The 'Port status' for eth2 does show '1 GbE' while the connected 'lan3' shows traffic but says 'no link' (in blue below).  In case any of this is useful:
[image]

Lanchon, 2025-01-21:
hey @hauke @robimarko

we are at rc5, close to a release that will apparently soft-brick a supported device.
what do you want to do about it?

PS.
i've lent my device so i can't test nor verify. i will recover it in february.

robimarko, 2025-01-21:
The information here is kind of confusing, you are saying no link but for example in the log you provide lan1 is clearly seeing both RX and TX traffic

jeremywh7, 2025-01-21:
> you are saying no link

The GUI says 'no link', but the key issue is that there is no actual Ethernet connection.  DHCP gets no address, and a static IP also will not connect; ping, ssh, internet, et cetera.  The missing _config device_'s in DSA seems a key issue (at least part of it), but adding it in doesn't fix the dead Ethernet.

> but for example in the log you provide lan1 is clearly seeing both RX and TX traffic

The log is from the initial 23.05.5 -> 24.10-rc5 -> recovery -> snapshot -> recovery -> 23.05.05 testing, when I was connected to lan1.  I tested all the ports to verify nothing worked, and then once I got wireless working (per Lanchon's suggestion), the Ethernet happened to be on lan3 (as seen in the ports' screenshot).  The shown traffic is probably just from the failed negotiations?  If I had working Ethernet, the `ifconfig` RX/TX would be much higher, like in _phy1-ap0_.

Lanchon, 2025-01-24:
### cannot reproduce

i recovered my device.

i tested rc6 and today's snapshot and they both work perfectly !!

email me you OpenWrt.mtd21.fw-env.bin partition and the result of fw_printenv
lanchon at gmail, ill try see whats wrong on your side

Lanchon, 2025-01-24:
@jeremywh7 please always write text in the open, unless it is sensitive.

there are a few missing entries in your uboot env, idk why:
```diff
1a2
> bootargs=console=ttyMSM0,115200n8
4c5
< bootdelay=5
---
> bootdelay=2
10a12,16
> ethact=eth0
> machid=1260
> stderr=serial
> stdin=serial
> stdout=serial
```

your mtd21.fw-env seems to be ok. your MAC addresses and other stuff there all seem valid.

this is weird:
```
br-lan    Link encap:Ethernet  HWaddr[mac stripped]
          inet addr:192.168.2.1  Bcast:192.168.2.255  Mask:255.255.255.0
eth2      Link encap:Ethernet  HWaddr[mac stripped]
          inet addr:192.168.1.104  Bcast:192.168.1.255  Mask:255.255.255.0
```
this should not be the reset state afaik.

jeremywh7, 2025-01-24:
>> [jeremywh7:] I installed rc6 w/o saving configs/settings - same issue for me.  An interesting thing happened after changing the eth2/wan IP and getting wireless working again.  I changed a couple 'Interfaces' -> 'lan' settings, replugged my Ethernet, and I got a dhcp address!  Unfortunately, I still couldn't get internet, as before.  The port does actually show '1GbE' now instead of 'no link' even now that dhcp isn't working again (after trying to replicate this scenario).  I have both Ethernet and wireless enabled/connected, and the 'lan4' port does show traffic (as before), but a tiny bit compared to the 'wan'.

> please always write text in the open, unless it is sensitive.

Gotcha, added my email above for continuity.  ;-)

> there are a few missing entries in your uboot env, idk why:

To be clear, are you comparing my OpenWrt `fw_printenv` to your (or my) uboot `printenv` or something else?  My uboot does have those entries, that are missing from OpenWrt [1].  So, should I change entries in OpenWrt, uboot, and/or do something else...?  I wonder if @4q1 has tried 24.10, since they have the same variant (with OEM firmware on Chaos Calmer)...maybe there's some difference with how it was setup for uboot, etc.?  Actually, I can compare that from the wiki; there are definitely differences with the network eth#/MAC/ports [2].

> this is weird:

Are you referring to the IP address?  That isn't the reset IP address, as stated above: "after **changing the eth2/wan IP** and getting wireless [internet] working again".  The attached log at the top has the defaults.  Since I'm connecting to my normal gateway, I needed to change the 'wan' (eth2) IP to connect.  Note:  This is the same setup used when testing 23.05.5 (with working internet and everything).

Thanks!

[1] Also for completeness, to compare my OpenWrt:
```
root@OpenWrt:/# fw_printenv
bootdelay=5
baudrate=115200
boot_stock=bootipq
boot_custom=test -n "$boot_pre" && run boot_pre; run boot_main; run boot_recovery; run boot_tftp
boot_main=set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:155M@0x6500000(mtd_ubi) && ubi part mtd_ubi && ubi read 0x44000000 kernel && bootm
boot_recovery=run boot_ubi_recovery; run boot_extra_recovery
boot_ubi_recovery=set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:155M@0x6500000(mtd_ubi) && ubi part mtd_ubi && ubi read 0x44000000 recovery && bootm
boot_extra_recovery=set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:64M@0x1340000(mtd_extra) && ubi part mtd_extra && ubi read 0x44000000 recovery && bootm
boot_tftp=test -n "$ipaddr" || set ipaddr 192.168.1.1; test -n "$serverip" || set serverip 192.168.1.2; tftpboot recovery.bin && bootm
bootcmd=run boot_custom
```
...versus uboot:
```
(IPQ) # printenv
baudrate=115200
boot_custom=test -n "$boot_pre" && run boot_pre; run boot_main; run boot_recovery; run boot_tftp
boot_extra_recovery=set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:64M@0x1340000(mtd_extra) && ubi part mtd_extra && ubi read 0x44000000 recovery && bootm
boot_main=set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:155M@0x6500000(mtd_ubi) && ubi part mtd_ubi && ubi read 0x44000000 kernel && bootm
boot_recovery=run boot_ubi_recovery; run boot_extra_recovery
boot_stock=bootipq
boot_tftp=test -n "$ipaddr" || set ipaddr 192.168.1.1; test -n "$serverip" || set serverip 192.168.1.2; tftpboot recovery.bin && bootm
boot_ubi_recovery=set mtdids nand0=nand0 && set mtdparts mtdparts=nand0:155M@0x6500000(mtd_ubi) && ubi part mtd_ubi && ubi read 0x44000000 recovery && bootm
bootcmd=run boot_custom
bootdelay=5
ethact=eth0
machid=1260
stderr=serial
stdin=serial
stdout=serial
```

[2] Here are the diffs before "Hit any key to stop autoboot:" (your device also shows the 26 ubi partition details; mine does not):
[image]

[3] Note my 'DRAM' value, and also your OEM firmware bootup:
```
Memory: 473568K/485376K available (4627K kernel code, 376K rwdata, 1592K rodata, 184K init, 605K bss, 11808K reserved, 0K highmem)
vmalloc : 0xde800000 - 0xff000000   ( 520 MB)
lowmem  : 0xc0000000 - 0xde000000   ( 480 MB)
.text : 0xc0208000 - 0xc081b050   (6221 kB)
.init : 0xc081c000 - 0xc084a000   ( 184 kB)
.data : 0xc084a000 - 0xc08a803c   ( 377 kB)
.bss : 0xc08a803c - 0xc093f654   ( 606 kB)
RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
```
...compared my OEM firmware bootup: "1009664K available" !
```
Memory: 993584K/1009664K available (4482K kernel code, 370K rwdata, 1540K rodata, 167K init, 601K bss, 16080K reserved, 237568K highmem)
vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
.text : 0xc0208000 - 0xc07e9c58   (6024 kB)
.init : 0xc07ea000 - 0xc0813d80   ( 168 kB)
.data : 0xc0814000 - 0xc0870a48   ( 371 kB)
.bss : 0xc0870a48 - 0xc0906fd0   ( 602 kB)
RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
```
Unfortunately, running normal OpenWrt shows, "Memory: 475392K/491520K available".
Is that limited by a kernel parameter (or other)?  Anyway, thanks again for taking a look!

Lanchon, 2025-01-24:
> are you comparing my OpenWrt `fw_printenv` to your (or my) uboot `printenv`

yes i am.

i probably promoted those values by using saveenv while i was originally hacking the device.
anyway i just documented the diff; i don't see anything of importance.

> That isn't the reset IP address

well if it's not the reset state, then i personally have no use for that info.

since i can't repro, i can't help you further.
the only thing left to do was to compare the reset state of our routers.

maybe someone else can help you.

```
oot@OpenWrt:~# free
              total        used        free      shared  buff/cache   available
Mem:         476416       46668      402284         324       27464      391280
Swap:             0           0           0

root@OpenWrt:~# dmesg | grep -i memory
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Early memory node ranges
[    0.000000] Memory: 475392K/491520K available (7726K kernel code, 648K rwdata, 1208K rodata, 1024K init, 250K bss, 16128K reserved, 0K cma-reserved, 0K highmem)
[    4.209143] Freeing unused kernel image (initmem) memory: 1024K

```

jeremywh7, 2025-01-24:
Can you email me your 24.10-rc6 boot log?  I'd like to compare that to mine.

> well if it's not the reset state, then i personally have no use for that info.
> the only thing left to do was to compare the reset state of our routers.

Oh, you didn't ask for that, and stated it was in the first post boot log.  The 'reset state' `ifconfig` is:
```
br-lan    Link encap:Ethernet  HWaddr [mac stripped]:F4
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: [mac stripped]:b017::1/60 Scope:Global
          inet6 addr: [mac stripped]:49f4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:653 errors:0 dropped:0 overruns:0 frame:0
          TX packets:167 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:30861 (30.1 KiB)  TX bytes:8478 (8.2 KiB)
eth0      Link encap:Ethernet  HWaddr [mac stripped]:F4
          inet6 addr: [mac stripped]:49f4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1502  Metric:1
          RX packets:698 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2736 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:44328 (43.2 KiB)  TX bytes:165756 (161.8 KiB)
          Interrupt:35
eth2      Link encap:Ethernet  HWaddr [mac stripped]:F1
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:38
lan1      Link encap:Ethernet  HWaddr [mac stripped]:F4
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:698 errors:0 dropped:4 overruns:0 frame:0
          TX packets:181 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:42932 (41.9 KiB)  TX bytes:11332 (11.0 KiB)
lan2      Link encap:Ethernet  HWaddr [mac stripped]:F4
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
lan3      Link encap:Ethernet  HWaddr [mac stripped]:F4
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
lan4      Link encap:Ethernet  HWaddr [mac stripped]:F4
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:232 errors:0 dropped:0 overruns:0 frame:0
          TX packets:232 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:18096 (17.6 KiB)  TX bytes:18096 (17.6 KiB)
```

Lanchon, 2025-01-24:
> Is that limited by a kernel parameter (or other)?

the ram is defined here: 

https://github.com/openwrt/openwrt/blob/9ada8578fd2768d077607b189987972ad600a804/target/linux/ipq806x/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq8065-tr4400-v2.dts#L12

format: `reg = <base-address size>;`

this defines 480 MB of ram.

it is typical for the physical ram to start at offset 0x40000000. here you see that there are 32 MB of memory that linux is not touching. why? this was copied from the DTB of the original firmware. idk, but i'd guess that those 32 MB of ram is where the trustzone mini-OS is running. in that case, accessing that ram would trigger a fault. you can easily try reading a value in that range from uboot and see.

if your device has 1 GB ram, your line should be:
```
reg = <0x42000000 0x3e000000>;
```
resulting in 992 MB.

but this would mean that your device is a different hardware version! did you even check the label?

Lanchon, 2025-01-24:
> The 'reset state' `ifconfig` is:

looks the same as mine.

> Can you email me your 24.10-rc6 boot log?

what log? dmesg? i don't have serial now.

please check the label first.

jeremywh7, 2025-01-24:
> but this would mean that your device is a different hardware version! did you even check the label?

Yes it is the same FCCID (UIDTR4400) and model name (RAC2V1A).  So this 'SAC2V1A' variant (in small print on the QR code app label) had upgraded RAM, along with newer software.  I recall reading before modding mine about 512MB and 1024MB variants.  That doesn't (necessarily) require a new PCB nor FCC re-cert.  Who knows what other components might have changed... but why does 23.05 work while 24.10 doesn't?  That's why I'd like to see your boot log. ;-)

Lanchon, 2025-01-24:
> please check the label first.

mine says:

FCC ID: UIDTR4400

and there is no other mention of any hw versrion.

jeremywh7, 2025-01-24:
> what log? dmesg? i don't have serial now.

dmesg is fine since the uboot log is in the wiki 👍🏼

Lanchon, 2025-01-24:
> dmesg is fine

[dmesg.txt.zip](https://github.com/user-attachments/files/18542299/dmesg.txt.zip)

jeremywh7, 2025-01-25:
@Lanchon  One thing I saw comparing to your dmesg output; my non-working 24.10 and snapshot boot logs have:
> Cannot parse config file '/etc/fw_env.config': No such file or directory
Failed to find NVMEM device

While my 23.05 and your 24.10 logs (both working) do not have that.  In my 23.05 backup, that file references the uboot env:
> /dev/mtd9 0x0 0x40000 0x20000

Might this be related to the issue?

Lanchon, 2025-01-25:
i suppose the file exist:

```
root@OpenWrt:/# cat /etc/fw_env.config  
/dev/mtd9 0x0 0x40000 0x20000 
```

this would mean that something is using the file before it is created, i don't know why.
but it might not be important.

---

this file *was* needed for commands such as `fw_printenv` to work.

but apparently [not anymore](https://github.com/openwrt/openwrt/commit/be37ab6e51b7526d386fce035b6d668d64598248):
> Since uboot-envtools 2024.01, there is a way to make these utils work
> without /etc/fw_env.config if the DT has an env partition with the prop.:
> 
> compatible = "u-boot,env";

but your device does not have that tag:
https://github.com/openwrt/openwrt/blob/7322d3266d84ec4062046254792d78ebca3ea26d/target/linux/ipq806x/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq8065-tr4400-v2.dts#L180-L184

jeremywh7, 2025-01-26:
Thanks - I will give that a try (along with the memory change); but that line goes in the partition section with "label = "0:APPSBLENV";" though, right?

Is your /proc/mtd different (below is from 24.10-rc5) - I think it is?  Would different addresses and/or sizes cause an issue, or is all that lower level?  [i.e. doesn't matter to OpenWrt bootup - so long as the partition is big enough of course]
```
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00020000 "0:SBL1"
mtd1: 00140000 00020000 "0:MIBIB"
mtd2: 00140000 00020000 "0:SBL2"
mtd3: 00280000 00020000 "0:SBL3"
mtd4: 00120000 00020000 "0:DDRCONFIG"
mtd5: 00120000 00020000 "0:SSD"
mtd6: 00280000 00020000 "0:TZ"
mtd7: 00280000 00020000 "0:RPM"
mtd8: 00500000 00020000 "0:APPSBL"
mtd9: 00080000 00020000 "0:APPSBLENV"
mtd10: 00140000 00020000 "0:ART"
mtd11: 04000000 00020000 "extra"
mtd12: 00060000 00020000 "0:BOOTCONFIG"
mtd13: 00140000 00020000 "0:SBL2_1"
mtd14: 00280000 00020000 "0:SBL3_1"
mtd15: 00120000 00020000 "0:DDRCONFIG_1"
mtd16: 00120000 00020000 "0:SSD_1"
mtd17: 00280000 00020000 "0:TZ_1"
mtd18: 00280000 00020000 "0:RPM_1"
mtd19: 00060000 00020000 "0:BOOTCONFIG1"
mtd20: 00500000 00020000 "0:APPSBL_1"
mtd21: 00100000 00020000 "fw_env"
mtd22: 09b00000 00020000 "ubi"
mtd23: 00008000 00008000 "spi0.0"
```

> [me:] I recall reading before modding mine about 512MB and 1024MB variants.

I noticed that the [techinfodepot page](https://en.techinfodepot.shoutwiki.com/wiki/Arris_RAC2V1A) shows RAC2V1A as 1 GB RAM (2*512MB chips - NT5CC256M16DP-DI, DDR3L-1600, CL11-11-11).  Fwiw the [WiFi Alliance cert](https://www.wi-fi.org/product-finder-results?keywords=WFA71730) confirms the 'two variants' for that FCCID:

[image]
[image]

Lanchon, 2025-01-26:
> Is your /proc/mtd different

that is only a runtime reflection of the dts. i wrote the dts, it is part of the fw image, and it is the same for both of us.

> Fwiw the [WiFi Alliance cert](https://www.wi-fi.org/product-finder-results?keywords=WFA71730) confirms the 'two variants' for that FCCID

makes sense. i wouldn't have mistaken my ram size, i gather, when i did the port.

but none of these details matter to your issue. and i *really* can't debug without being able to repro. (and even then, there is no guarantee that i'd be able to fix the issue.)

Lanchon, 2025-01-26:
> Thanks - I will give that a try

well you can, but i'm not asking for that. idk the root cause of your issues.

jeremywh7, 2025-01-30:
>if your device has 1 GB ram, your line should be: `reg = <0x42000000 0x3e000000>;`
resulting in 992 MB.

Right on, that was easy; testing 24.10-snapshot that I built last night:
```
Kernel command line: rootfstype=squashfs noinitrd
Bootloader command line not present
Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Built 1 zonelists, mobility grouping on.  Total pages: 252224
mem auto-init: stack:off, heap alloc:off, heap free:off
Memory: 993672K/1015808K available (7496K kernel code, 646K rwdata, 2028K rodata, 1024K init, 239K bss, 22136K reserved, 0K cma-reserved, 229376K highmem)
```
👍🏼 
I tried adding the 'compatible...' line but the kernel crashed, then I noticed [this change](https://github.com/openwrt/openwrt/commit/fe17cc1bfd1) last month, so I merged that in and it built:
```
                        partition@c80000 {
                                label = "0:APPSBL";
                                reg = <0x0c80000 0x0500000>;
                                read-only;
                        };
                        partition@1180000 {
                                label = "0:APPSBLENV";
                                reg = <0x1180000 0x0080000>;
                                nvmem-layout {
                                        compatible = "u-boot,env";
                                };
                        };
```
It does go in the APPSBLENV section like the other commits, right (you orig mentioned APPSBL) ?

Unfortunately, I still get the same boot message.
```
[    5.459856] init: - preinit -
Cannot parse config file '/etc/fw_env.config': No such file or directory
Failed to find NVMEM device
[    6.247609] ipq806x-gmac-dwmac 37000000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
```
Fwiw, that file does exist after boot (as in my 23.05.5).  No ethernet of course, but I guess I'll fix my memory size for 23.05-snapshot instead.  \o/

jeremywh7, 2025-01-30:
Done, after an [issue with building ninja](https://github.com/openwrt/openwrt/issues/16675#issuecomment-2623367616) 1.11.1...
[image]

Lanchon, 2025-01-31:
> Thanks - I will give that a try (along with the memory change); but that line goes in the partition section with "label = "0:APPSBLENV";" though, right?

yes, jeez, sorry!

jeremywh7, 2025-02-01:
> does this also happen in official rcN (whatever rc you tested)?

Yes, it in the logs for 24.10-rc5 and -rc6, my 24.10-snapshot (with _nvmem,layout_ patch), and the /main snapshot I tried a week or two ago.  Does not show up in 25.05.5 nor my 25.05-snapshot.

Lanchon, 2025-02-01:
hey @robimarko 

we have a strange case here: while booting a device, something is trying to access `/etc/fw_env.config` before the file is created. (later in the boot process the file is created correctly.)

this happens 1) only on @jeremywh7 's device (never on mine), and 2) only if he runs SNAPSHOT or 24.10 (not on 23.05). take a look [here](https://github.com/openwrt/openwrt/issues/17631#issuecomment-2613734476), [here](https://github.com/openwrt/openwrt/issues/17631#issuecomment-2627996836), and [here](https://github.com/openwrt/openwrt/issues/17631#issuecomment-2628647327).

any ideas on why this puzzling behavior? @jeremywh7 's device seems to be a new HW revision (has twice the ram).

also, when the issue above happens, the ethernet ports of the device become non-fucntional. (again this happens only on SNAPSHOT or 24.10, but not on 23.05.)

jeremywh7, 2025-02-01:
> the ethernet ports of the device become non-fucntional. (again this happens only on SNAPSHOT or 24.10, but not on 23.05.)

Interesting thing that just happened, in case this is any clue to the issue.  I've had 23.05 running on ethernet (port 1) for 2.5 days; it was working fine 8 hours ago.  But this morning, it's acting just like 24.10 - can't get an IP address.  Tried replugging the WAN and ethernet (at the PC), no help.  So I connected to wireless and the GUI still showed a lease with almost 12 hrs left, but an unknown mac[1] for my PC name (I've only been connecting my main PC to test this device).

`dmesg` didn't show anything except when I replugged the WAN.  So I restarted odhcpd from GUI (System -> Startup), moved the ethernet cable on the router to port 4, and now everything works fine again, so I turned off wireless again.  [and yes, I had tried restarting services on 24.10: dnsmasq, firewall, network, odhcpd]

[1] The GUI still shows the unknown mac but the PC name is now on the working entry (.158 is the ethernet):
[image]

If this happens again I'll see if just changing the port brings it back...

jeremywh7, 2025-02-02:
In case anyone is curious, here's an html diff of Lanchon's RAC2V1A 24.10-rc6 dmesg vs my SAC2V1A 24.10-rc5 serial boot.  There's also a [uboot diff](https://github.com/openwrt/openwrt/issues/17631#issuecomment-2613359139) b/t them in note [2].

[RAC2V1A dmesg vs SAC2V1A serial log diff.html.zip](https://github.com/user-attachments/files/18630775/RAC2V1A.dmesg.vs.SAC2V1A.serial.log.diff.html.zip)

jeremywh7, 2025-02-04:
Could this [ipq806x DSA WAN issue](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=cd747cb3aa) be a similar cause of my DSA LAN issue?  E.g. `ifconfig` on 23.05 shows my LAN with eth1 (no eth0 present) and mac :F0 (same for br-lan):
```
eth1      Link encap:Ethernet  HWaddr 8C:[removed]:F0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1088 errors:0 dropped:2 overruns:0 frame:0
          TX packets:928 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:169980 (165.9 KiB)  TX bytes:782256 (763.9 KiB)
          Interrupt:39
```
...but on 24.10 and SNAPSHOT it comes up with eth0 (no eth1 present) and mac :F4 (same for br-lan and lan[0-4]):
```
eth0      Link encap:Ethernet  HWaddr 8C:[removed]:F4
          inet6 addr: fe80::8e5a:25ff:fe29:49f4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1502  Metric:1
          RX packets:698 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2736 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:44328 (43.2 KiB)  TX bytes:165756 (161.8 KiB)
          Interrupt:35
```
So the mac is definitely being handled differently.  @Lanchon Does your LAN mac also change b/t 23.05 and 24.10?  My WAN is eth2 and the same mac ( :F1 ) for both though.