openwrt/openwrt #17632: ASIA RF AW7916-NPD (MT7916) no 6GHz.

FerrumMaster, 2025-01-16:
### Describe the bug

No WIFI 6E capabilities in 24.x.x builds using this card.

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

Use AW7916-NPD card on any openwrt 24.x.x built. Try to set up 6GHz AP. There will be no options to do it.

Tested BPI-R3 and BPI-R4.

### Actual behaviour

iw info actually does not detect 6GHz capability.

`root@BPI-R4:~# iw phy0 info
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2200 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
[... 1066 lines cut ...]
[  835.083722] br-lan: port 6(phy1-ap0) entered disabled state
[  835.089348] mt7915e 0001:01:00.0 phy1-ap0: entered allmulticast mode
[  835.095865] mt7915e 0001:01:00.0 phy1-ap0: entered promiscuous mode
[  835.102279] br-lan: port 6(phy1-ap0) entered blocking state
[  835.107871] br-lan: port 6(phy1-ap0) entered forwarding state
[  835.537730] br-lan: port 6(phy1-ap0) entered disabled state
[  899.267362] br-lan: port 6(phy1-ap0) entered blocking state
[  899.272953] br-lan: port 6(phy1-ap0) entered forwarding state
`

### Expected behaviour

The ASIA RF card is good and working, it is tested on older BPI-R3 using older 23.x.x builds, iw list exposes all supported frequencies there. 24.x.x base also has the same problem on BPI-R3. BPI-R4 has no official 23.x.x release. I have tried using older MT7916 firmware, no change.

### Additional info

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Kiwimannen, 2025-01-16:
I can confirm this bug with other hardware such as the Nanopi R5S and Raspberry pi 5.

FerrumMaster, 2025-01-24:
Same on RC6

ndoo, 2025-02-04:
Same here on Turris Omnia + AsiaRF AW7916-NPD.

```
OpenWrt 24.10.0-rc7, r28417-daef29c75d
```

```
# opkg list_installed|grep mt7
kmod-mt76-connac - 6.6.73.2025.01.14~8e4f72b6-r1
kmod-mt76-core - 6.6.73.2025.01.14~8e4f72b6-r1
kmod-mt7915e - 6.6.73.2025.01.14~8e4f72b6-r1
kmod-mt7916-firmware - 6.6.73.2025.01.14~8e4f72b6-r1
```

Manually setting `/etc/config/wireless` phy1 `wifi-device` with `option band '6g'` & ensuring `wifi-iface` are `option encryption 'sae'`, hostapd refuses to start, both on `option channel 'auto'` and `option channel '21'`, with the following error in `logread`: `IEEE 802.11 Hardware does not support configured mode (2)`.

```
# dmesg|grep mt7915
[   15.372981] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20240823172725a
[   15.522672] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20240823172741
[   15.624152] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20240823172837
[   15.790869] mt7915e 0000:01:00.0: registering led 'mt76-phy0'
[   15.833727] mt7915e 0000:01:00.0: registering led 'mt76-phy1'
[   20.571629] mt7915e 0000:01:00.0 phy0-ap0: entered allmulticast mode
[   20.578187] mt7915e 0000:01:00.0 phy0-ap0: entered promiscuous mode
[   21.505412] mt7915e 0000:01:00.0 phy1-ap0: entered allmulticast mode
[   21.511938] mt7915e 0000:01:00.0 phy1-ap0: entered promiscuous mode
```

```
# iw phy1 info
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 4
        max scan IEs length: 2190 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x1c RX 0x1c
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 2:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 2 usec (0x04)
                HT TX/RX MCS rate indexes supported: 0-15
                VHT Capabilities (0x339a59f6):
                        Max MPDU length: 11454
[... 287 lines cut ...]
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ STA_TX_PWR ]: TX power control per station
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ FILS_DISCOVERY ]: FILS discovery frame transmission support
                * [ UNSOL_BCAST_PROBE_RESP ]: unsolicated broadcast probe response transmission support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
                * [ BSS_COLOR ]: BSS coloring support
                * [ RADAR_BACKGROUND ]: Radar background support
                * [ POWERED_ADDR_CHANGE ]: can change MAC address while up
```

[image]

ndoo, 2025-02-05:
Issue persists on release:

```
OpenWrt 24.10.0, r28427-6df0e3d02a
```

```
# opkg list_installed|grep mt7
kmod-mt76-connac - 6.6.73.2025.01.14~8e4f72b6-r1
kmod-mt76-core - 6.6.73.2025.01.14~8e4f72b6-r1
kmod-mt7915e - 6.6.73.2025.01.14~8e4f72b6-r1
kmod-mt7916-firmware - 6.6.73.2025.01.14~8e4f72b6-r1
```

ndoo, 2025-02-05:
Downgrading back to 23.05.5 solves the issue so it is indeed a regression.

FerrumMaster, 2025-02-11:
Still nothing on latest builds.

ndoo, 2025-02-12:
Why isn't this issue acknowledged in the release notes in spite of the multiple reports across different targets and the correct tags against the release?

s09289728096, 2025-02-13:
[image]
(All tests has been done on snapshot version of openwrt with BPi-R4 and AW7916-NPD)

TL;DR
You need to select the band for either 5 Ghz or 6 GHz by changing parameter "enable_6ghz"

Detailed explain:
The value in efuse has been proven that supports both 5 Ghz and 6 Ghz.
[wifi: mt76: mt7915: add module param to select 5 GHz or 6 GHz on MT7916](https://github.com/openwrt/mt76/commit/5f6051569639460ba0d88154d8b7cd8a96e1cbad)
However, mt7915e driver has no longer supported both 5Ghz and 6Ghz after this commit. Unfortunately, it seems that drvier doesn't work fine by just changing this value in debugfs, which means reloading driver is required.
To test it, try the command on your device:

```
echo 1 > /sys/kernel/debug/ieee80211/<phy_for_band2>/mt76/enable_6ghz
```

To solve the issue, choose either way at follow:
1. Change the default value of 'enable_6ghz' to 1 and recompile driver.
You can add this patch into openwrt/package/kernel/mt76/patches (you may need to create a patches folder) and the mt7915e driver will enable 6G band in default, which means 5G band will be disabled, 

```diff --git a/mt7915/eeprom.c b/mt7915/eeprom.c
index 928e0b07..af09d5df 100644
--- a/mt7915/eeprom.c
+++ b/mt7915/eeprom.c
@@ -6,7 +6,7 @@
 #include "mt7915.h"
 #include "eeprom.h"

-static bool enable_6ghz;
+static bool enable_6ghz = 1;
 module_param(enable_6ghz, bool, 0644);
 MODULE_PARM_DESC(enable_6ghz, "Enable 6 GHz instead of 5 GHz on hardware that supports both");
```

2. Transfer parameter into driver.
On your device, do either:
```
#execute whenever booting
rmmod mt7915e; insmod mt7915e enable_6ghz=1

#or just do once
echo "mt7915e enable_6ghz=1" > /etc/modules.d/mt7915e
```

btw, the default country code 00 doesn't support 6G band. Remember to switch to the country which supports it.

[image]

If there are any questions about this, feel free to tag me.

ndoo, 2025-02-13:
This change definitely needs some form of documentation since it changes the behavior from LUCI/UCI perspective between major releases.

As an aside, implementing a flag in mt76 that allows simultaneous 5+6 instead of 2+6 would be welcome, too, if the issue all along was about memory limiting which cal data could be loaded simultaneously.

s09289728096, 2025-02-13:
Tbh I didn't even know this issue until this morning (lmao)
I'm pretty curious why they made this change too, which takes me half a day to check eeprom values (lmao again). 
However, I'm not the maintainer of mt76 nor the employee of Mediatek, so if there are related topics about driver, please suggest [**here**](https://github.com/openwrt/mt76).

> This change definitely needs some form of documentation since it changes the behavior from LUCI/UCI perspective between major releases.
> As an aside, implementing a flag in mt76 that allows [...]

MT7916 can only enable DBDC on either 2+5Ghz or 2+6Ghz. This chip isn't designed to 5+6, and it's extremely rare that a wifi chip has DBDC for 5+6 Ghz even in all industries. The common way is intergrate multiple chips into a single module to archive 2+5+6 or 5+6.

> From consumer perspective it is weird, it renders all AsiaRF products crippled, notifications about this would be welcome too in change log. Who thought it out like that? Why simply you do not split mt7915 and create proper mt7916 driver with 6GHz enabled?

We'll definitely have a notification about this issue after we figuring out the reasons.
For mt76, mediatek develops drivers by chip series and I don't think they will split driver by this reason.

ndoo, 2025-02-13:
I think the criticism was mostly to the maintainers because this is a breaking change for anyone who had their bands set to 2g and 6g respectively, then found 6g phy not starting after the upgrade, with no mention in the release notes about the regression, despite it being reported several RCs ago.

FerrumMaster, 2025-02-14:
> For mt76, mediatek develops drivers by chip series and I don't think they will split driver by this reason.

I was thinking openwrt repo having same driver compiled with each use case option ie 2+5 or 2+6 and naming the 2+6 MT7916 for better clarity.

I only took the card because of 6GHz as my 5GHz is crowded, it could have been communicated better for sure.

DanaGoyette, 2025-04-06:
Another question on top of this: if it's a module parameter, then how is one ever supposed to set one card to be 2.4GHz + 5GHz, and another to be 2.4 GHz (or disabled) + 6GHz?

Switzerg95, 2025-04-11:
Is 802.11s workable on the 6Ghz radio? Has anyone tried?

FerrumMaster, 2025-05-19:
Situation has worsened on latest snapshot after doing clean BPI-5 install on emmc. After adding enable_6ghz=1, luci has no GUI options to enable or change anything.

You can add lines to /etc/config/wireless and then the add in card starts to work.

Seriously what's going on?

bubobih, 2025-07-19:
Any solutions till today?

bubobih, 2025-07-22:
[image]

I did that but after reboot its disabled and i cant change inside channel or any other except wifi pass ssid name and minor things

Delitants, 2026-01-16:
LuCI Master (26.006.47974~6ebcfe6) / OpenWrt SNAPSHOT (r32562-f948f71300)

Same, no changes. Only 5 and 2.4 Ghz.
What takes so long?

Delitants, 2026-04-05:
> > Set country code to GB to make 6 Ghz work.
> 
> And you are replying to who?

Not to you, dummy.

Shulyaka, 2026-05-08:
I think I figured it out. Once you configure a valid channel and ht mode (and wpa3 of course) for the first time, then it all starts to work.

sandreas, 2026-09-03:
So to sum it up:

- Wifi 6Ghz and 5Ghz are not supported simultaneously by default?
- This is broken after 23.05 (where everything worked) - probably mt76 driver change caused this
- You can [patch the source or use a config file to re-enable 6Ghz](https://github.com/openwrt/openwrt/issues/17632#issuecomment-2655973661) (`echo "mt7915e enable_6ghz=1" > /etc/modules.d/mt7915e`)
- Luci / GUI changes will NOT work, you have to use `uci` or config changes via SSH
- Changing the ` /etc/modules.d/mt7915e` will make OpenWRT operate 5GHz and 6GHz simultaneously again?

Does that sound right?

Delitants, 2026-09-03:
> So to sum it up:
>
> * Wifi 6Ghz and 5Ghz are not supported simultaneously by default?
> * This is broken after 23.05 (where everything worked) - probably mt76 driver change caused this
> * You can [patch [...]

1. No supported entirely, only 6+2.4 / 5+2.4
2. Not broken, works fine
3. Just execute that in the terminal, no need patching source, reboot. To switch back to 5G, remove the line.
4. Luci works just fine with 6G
5. No it will not, one or another.

Th reason is probably because mPCIE is not able to provide enough current for 5+6. And still yet I had to rewire power for my mPCIE slot to be able to run AW7916-NPD.

DanaGoyette, 2026-09-03:
The toggle really ought to be per-device via sysfs, not via a module option.  
Or at the very least, it should be an array-type module option, like some of the options on the `e1000e` wired ethernet driver are. 
Though the order of which card is 0 and which is 1 might not be stable, so going by sysfs is better.